apache/nuttx #12686: STM32H745I-DISCO Userleds Error

NonpawitE, 2024-07-13:
```FAILED: boards/CMakeFiles/board.dir/arm/stm32h7/stm32h745i-disco/src/stm32_userleds.c.obj 
/usr/bin/arm-none-eabi-gcc -D__KERNEL__ -D__NuttX__ -I../arch/arm/src/stm32h7 -I../arch/arm/src/common -I../arch/arm/src/armv7-m -I../sched -isystem ../include -isystem include -isystem include_arch -mtune=cortex-m7 -march=armv7e-m -mfpu=fpv5-d16 -mfloat-abi=hard -fno-common -Wall -Wshadow -Wundef -nostdlib -mthumb -Wa,-mthumb -Wa,-mimplicit-it=always -fomit-frame-pointer -ffunction-sections -fdata-sections -g -Wno-attributes -Wno-unknown-pragmas -Wstrict-prototypes -fdiagnostics-color=always -DNDEBUG -MD -MT boards/CMakeFiles/board.dir/arm/stm32h7/stm32h745i-disco/src/stm32_userleds.c.obj -MF boards/CMakeFiles/board.dir/arm/stm32h7/stm32h745i-disco/src/stm32_userleds.c.obj.d -o boards/CMakeFiles/board.dir/arm/stm32h7/stm32h745i-disco/src/stm32_userleds.c.obj   -c ../boards/arm/stm32h7/stm32h745i-disco/src/stm32_userleds.c
../boards/arm/stm32h7/stm32h745i-disco/src/stm32_userleds.c:56:3: error: 'GPIO_LED_BLUE' undeclared here (not in a function); did you mean 'GPIO_LCD_BL'?
   56 |   GPIO_LED_BLUE,
      |   ^~~~~~~~~~~~~
      |   GPIO_LCD_BL
```

I faced this error and looked through the code for STM32H745I-DISCO board, should the line 56 in stm32_userleds.c be `GPIO_LED_ORANGE` instead of `GPIO_LED_BLUE`?

https://github.com/apache/nuttx/blob/c86ef1cbc43016a019b6098343ff8dac6760cd25/boards/arm/stm32h7/stm32h745i-disco/src/stm32_userleds.c#L56

acassis, 2024-07-13:
Hi @NonpawitE thank you for reporting this issue. Strange, the GPIO_LED_BLUE seems to be defined:

```
alan@dev:~/nuttxspace/nuttx/boards/arm/stm32h7/stm32h747i-disco$ git grep GPIO_LED_BLUE
src/stm32_autoleds.c:  GPIO_LED_BLUE,
src/stm32_userleds.c:  GPIO_LED_BLUE,
src/stm32h747i-disco.h:#define GPIO_LED_BLUE   GPIO_LD4
alan@dev:~/nuttxspace/nuttx/boards/arm/stm32h7/stm32h747i-disco$ git grep GPIO_LD4
src/stm32h747i-disco.h:#define GPIO_LD4        (GPIO_OUTPUT | GPIO_PUSHPULL | GPIO_SPEED_50MHz | GPIO_OUTPUT_SET | \
src/stm32h747i-disco.h:#define GPIO_LED_BLUE   GPIO_LD4
alan@dev:~/nuttxspace/nuttx/boards/arm/stm32h7/stm32h747i-disco$
```

acassis, 2024-07-13:
Could you please share your .config file?

NonpawitE, 2024-07-13:
Here's my .config file [.config.txt](https://github.com/user-attachments/files/16203592/default.config.txt)

acassis, 2024-07-13:
Thank you @NonpawitE I run:
```
$ ./tools/configure.sh stm32h745i-disco:nsh
```

And used meld to compare your .config with the default nsh .config and move all your modifications related to USERLED to and run "make -j", it compiled fine.

Please find it attached. Just run that configure.sh above and copy the config.txt from here to over your .config and run "make -j". Please let me know if it worked.
[config_stm32h745i-disco_userled_nsh.txt](https://github.com/user-attachments/files/16203675/config_stm32h745i-disco_userled_nsh.txt)

NonpawitE, 2024-07-13:
Hi @acassis, I copied your .config to my .config file and run `make -j`, it compile with no errors, but when I ran `ls dev` in nsh the "userleds" won't show up. I have disabled `Board Selection > Board LED Status Support` and enabled `Application Configuration > NSH Library > Have architecture-specific initialization`, and after I compile with make I still got the same error

```
f24w@f24w-VirtualBox:~/nuttxspace/nuttx$ make -j
Create version.h
LN: platform/board to /home/f24w/nuttxspace/apps/platform/dummy
Register: leds
Register: nsh
Register: sh
CC:  stm32_appinitialize.c stm32_bringup.c: In function 'stm32_bringup':
stm32_bringup.c:140:9: warning: implicit declaration of function 'userled_lower_initialize' [-Wimplicit-function-declaration]
  140 |   ret = userled_lower_initialize("/dev/userleds");
      |         ^~~~~~~~~~~~~~~~~~~~~~~~
stm32_userleds.c:56:3: error: 'GPIO_LED_BLUE' undeclared here (not in a function); did you mean 'GPIO_LCD_BL'?
   56 |   GPIO_LED_BLUE,
      |   ^~~~~~~~~~~~~
      |   GPIO_LCD_BL
make[2]: *** [/home/f24w/nuttxspace/nuttx/boards/Board.mk:83: stm32_userleds.o] Error 1
make[1]: *** [Makefile:181: board/libboard.a] Error 2
make: *** [tools/Unix.mk:546: nuttx] Error 2
```

NonpawitE, 2024-07-13:
Is it possible that in [stm32h745i_disco.h](https://github.com/apache/nuttx/blob/c86ef1cbc43016a019b6098343ff8dac6760cd25/boards/arm/stm32h7/stm32h745i-disco/src/stm32h745i_disco.h#L85) only have the definition for
```
#define GPIO_LED_GREEN  GPIO_LD1
#define GPIO_LED_ORANGE GPIO_LD2
#define GPIO_LED_RED    GPIO_LD3
```

acassis, 2024-07-13:
Hi @NonpawitE 

You are right, I got same error now:
```
$ make -j
Create version.h
LN: platform/board to /home/alan/nuttxspace/apps/platform/dummy
Register: nsh
Register: sh
Register: leds
CPP:  /home/alan/nuttxspace/nuttx/boards/arm/stm32h7/stm32h745i-disco/scripts/flash.ld-> /home/stm32_userleds.c:56:3: error: 'GPIO_LED_BLUE' undeclared here (not in a function); did you mean 'GPIO_LCD_BL'?
   56 |   GPIO_LED_BLUE,
      |   ^~~~~~~~~~~~~
      |   GPIO_LCD_BL
make[2]: *** [/home/alan/nuttxspace/nuttx/boards/Board.mk:83: stm32_userleds.o] Error 1
make[2]: *** Waiting for unfinished jobs....
stm32_bringup.c: In function 'stm32_bringup':
stm32_bringup.c:140:9: warning: implicit declaration of function 'userled_lower_initialize' [-Wimplicit-function-declaration]
  140 |   ret = userled_lower_initialize("/dev/userleds");
      |         ^~~~~~~~~~~~~~~~~~~~~~~~
make[1]: *** [Makefile:181: board/libboard.a] Error 2
make: *** [tools/Unix.mk:546: nuttx] Error 2
```

acassis, 2024-07-13:
@NonpawitE actually I looked the schematic and this board doesn't have LEDs directly controlled by user:

https://www.google.com/url?sa=t&source=web&rct=j&opi=89978449&url=https://www.st.com/resource/en/schematic_pack/mb1381-h750xb-b01-schematic.pdf&ved=2ahUKEwjZu6eZkaWHAxXrqJUCHQPOBCIQFnoECBgQAQ&usg=AOvVaw1d__DAxuX_ThawnLixkt8x

Even LD1 that is connected to PA9 cannot be used as output because you can damage it when OTG 5V output is enabled at STMPS2151 chip.

So, you use USERLED on this board you need to use some accessible pins and connect LEDs to it.

NonpawitE, 2024-07-14:
@acassis Thank you very much. Do I need the same autoled bringup when I use external LED, or do I just use normal GPIO command?

acassis, 2024-07-14:
@NonpawitE to use the pin for USERLED you just need to define your pin as GPIO output (look as GPIO_LD1 was defined for other boards and change the Port and Pin number to the P(x)N you want to use: PA0, etc.)

Then include the symbol at g_ledcfg[BOARD_NLEDS]. You need to change BOARD_NLEDS to reflect the amount to LEDs you are controlling.

NonpawitE, 2024-07-14:
Got you @acassis thank you very much

closed 2024-07-14 by NonpawitE