apache/nuttx #12687: unwind_frame has issues on arm cortext M7

gneworld, 2024-07-14:
[image]

During the backtrace process,  the system freezes, and the anomaly shown in the image above. 

[image]

When I add this conditional check,  the problem disappeared，so is this a real issue or how should I do to avoid this issue？

acassis, 2024-07-14:
@gneworld probably it was caused by some function returning -1 to LR, what explain this 0xFFFFFFFF value.

Hi @anchao could you please help? I think you added support for it, right?

anjiahao1, 2024-07-15:
@gneworld 
[image]
try it

anjiahao1, 2024-07-15:
Perhaps the above method is not a good method. The reason for this problem is that the idle thread needs to clear lr before calling nxstart, otherwise the unwind backtrace is likely to go wrong because the value of `lr` is uncertain before `__start`.

gneworld, 2024-07-15:
> Perhaps the above method is not a good method. The reason for this problem is that the idle thread needs to clear lr before calling nxstart, otherwise the unwind backtrace is likely to go wrong because the value of `lr` is uncertain before `__start`.

@anjiahao1  but why does -O0 work correctly, while -O3 does not?

anchao, 2024-07-15:
> Hi @anchao could you please help? I think you added support for it, right?

@acassis @anjiahao1 is more familiar with the details of unwind table backtrace than me.  @anjiahao1 @gneworld Aren't you guys on the same floor? why not confirm the issue offline?

anjiahao1, 2024-07-15:
The root cause is that at reset, the value of the general register is usually not fixed, and lr needs to be set to 0 before calling nxstart, which requires modifying a lot of arch/chips code

anchao, 2024-07-15:
1. Seems as expected, you need to confirm with vendor the behavior of **RAR(Reset all registers)** when lockstep is disabled, which is fixed on design phase.
2. Zephyr does something similar, I think initializing the registers is necessary
https://github.com/zephyrproject-rtos/zephyr/pull/20473

[ARM_ECM_0690721_Cortex_M33_DCLS.pdf](https://github.com/user-attachments/files/16233802/ARM_ECM_0690721_Cortex_M33_DCLS.pdf)

[image]

anchao, 2024-07-15:
https://developer.arm.com/documentation/101773/0001/Functional-Description/CPU?lang=en

[image]

acassis, 2024-07-15:
> > Hi @anchao could you please help? I think you added support for it, right?
> 
> @acassis @anjiahao1 is more familiar with the details of unwind table backtrace than me. @anjiahao1 @gneworld Aren't you guys on the same floor? why not confirm the issue offline?

@anchao I think if they can do it at same room is fine, but please don't report the details here to let more people see what was the issue; how the root causes was discovered and why "that new commit" is the right solution :-)

lywind, 2024-07-16:
> 1. Seems as expected, you need to confirm with vendor the behavior of **RAR(Reset all registers)** when lockstep is disabled, which is fixed on design phase.
> 2. Zephyr does something similar, I think initializing the registers is necessary
>    [arch: arm: Rewrite Cortex-R reset vector function. zephyrproject-rtos/zephyr#20473](https://github.com/zephyrproject-rtos/zephyr/pull/20473)
> 
> [ARM_ECM_0690721_Cortex_M33_DCLS.pdf](https://github.com/user-attachments/files/16233802/ARM_ECM_0690721_Cortex_M33_DCLS.pdf)
> 
> ![20240715-194023](https://private-user-images.githubusercontent.com/758493/348691216-8b59d6e7-9ba5-44f6-8ce9-4769b3d310d1.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjExMDY1MTksIm5iZiI6MTcyMTEwNjIxOSwicGF0aCI6Ii83NTg0OTMvMzQ4NjkxMjE2LThiNTlkNmU3LTliYTUtNDRmNi04Y2U5LTQ3NjliM2QzMTBkMS5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNzE2JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDcxNlQwNTAzMzlaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0xMTI5N2E1MzUzMjEzN2NmODE1OTE0YjE5ZmY5Yzc3OGY0MDg0ZWNmOTU3N2QwNzM3OWU5Zjk3YmM2OTZkNjVhJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.o1cNZ0dyetb09TjLSEA19NCvTi7HnAhCNtMSIwrYNo8)

In fact, this is a Cortex-M7 MCU which is the ARM v7M-E architecture. And it does not have LOCKSTEP or RAR configurations. So I think it might not be the DCLS problem.

anchao, 2024-07-16:
> In fact, this is a Cortex-M7 MCU which is the ARM v7M-E architecture. And it does not have LOCKSTEP or RAR configurations. So I think it might not be the DCLS problem.

DCLS is configurable on Cortex-M7, I just suspect that the case they are facing is a issue on the lock-step core.

https://developer.arm.com/Processors/Cortex-M7

[image]

lywind, 2024-07-16:
> > In fact, this is a Cortex-M7 MCU which is the ARM v7M-E architecture. And it does not have LOCKSTEP or RAR configurations. So I think it might not be the DCLS problem.
> 
> DCLS is configurable on Cortex-M7, I just suspect that the case they are facing is a issue on the lock-step core.
> 
> https://developer.arm.com/Processors/Cortex-M7
> 
> ![20240716-160632](https://private-user-images.githubusercontent.com/758493/348996832-af875480-2af4-42f0-842c-4ac59047749c.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjExMjAwNDQsIm5iZiI6MTcyMTExOTc0NCwicGF0aCI6Ii83NTg0OTMvMzQ4OTk2ODMyLWFmODc1NDgwLTJhZjQtNDJmMC04NDJjLTRhYzU5MDQ3NzQ5Yy5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNzE2JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDcxNlQwODQ5MDRaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1jM2QyYmVkN2E5ZTk5ZmM2MzM5OGI2ODFjOTg3OGIxMjVjZWE2MDllODhkZGE5YjY3MWIxNjRmMWZjZmQ3ZjYwJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.Lghuob-mBYL9rlpiELOk3eRSZrQH3zzyyH6v7yJBxQ8)

Double checked and it's comfirmed that the LOCKSTEP and RAR are both enabled. And I'm wondering why the function unwind_find_entry(frame->pc) does not return NULL when frame->pc == 0xFFFFFFFC? It's obvious that 0xFFFFFFFC exceeds __exidx_end.

lywind, 2024-07-16:
> @gneworld probably it was caused by some function returning -1 to LR, what explain this 0xFFFFFFFF value.
> 
> Hi @anchao could you please help? I think you added support for it, right?

The reason of the 0xFFFFFFFF being on the stack is that the compiler considers the __start as a normal function and pushes LR in the stack。
[image]
And when the core boots from reset, the core will set LR to 0xFFFFFFFF.
[image]
B1.5.5 of Arm®v7-M Architecture Reference Manual

anchao, 2024-07-17:
I'm not sure if the `naked` attribute could avoid this issue, which ensures that the unwind `extab` does not contain any push content:

```
diff --git a/include/nuttx/init.h b/include/nuttx/init.h
index af3dce335f..98b9ba68f8 100644
--- a/include/nuttx/init.h
+++ b/include/nuttx/init.h
@@ -98,7 +98,7 @@ EXTERN uint8_t g_nx_initstate;  /* See enum nx_initstate_e */
 
 /* OS entry point called by boot logic */
 
-void nx_start(void);
+void nx_start(void) noreturn_function naked_function;
 
 #undef EXTERN
 #ifdef __cplusplus

```

gneworld, 2024-07-19:
@anchao  Unfortunately, this change cannot solve the problem.

anjiahao1, 2024-07-29:
hello all, i ceate a pr fix it https://github.com/apache/nuttx/pull/12787

closed 2024-07-29 by xiaoxiang781216