apache/nuttx #12710: esp32s3 boards with ESP32-S3-WROOM-1U can't boot into nsh?

chirping78, 2024-07-16:
Got two `esp32-s3-devkitc-1` boards recently. One is mounted with ESP32-S3-WROOM-1-N8R8,
Another is mounted with ESP32-S3-WROOM-1U-N8R8.

After install these tool:
```
curl -s -L "https://github.com/espressif/crosstool-NG/releases/download/esp-12.2.0_20230208/xtensa-esp32s3-elf-12.2.0_20230208-x86_64-linux-gnu.tar.xz" | tar -C xtensa-esp32s3-elf-gcc --strip-components 1 -xJ

pip3 install esptool==4.8.dev4
```

And use these commands to build and flash:
```
./tools/configure.sh -E esp32s3-devkit:nsh
make -j8

make flash ESPTOOL_PORT=/dev/ttyUSB0
```

One (with ESP32-S3-WROOM-1-N8R8) can boot into nsh:
```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fc8ac40,len:0xdbc
load:0x40374000,len:0x4a54
SHA-256 comparison failed:
Calculated: 89806fd28d864acd62041e463d876296efdab8615d97ec5002d7c405cb83d17a
Expected: 00000000b0a70000000000000000000000000000000000000000000000000000
Attempting to boot anyway...
entry 0x40374d24
*** Booting NuttX ***
dram: lma 0x00000020 vma 0x3fc8ac40 len 0xdbc    (3516)
iram: lma 0x00000de4 vma 0x40374000 len 0x4a54   (19028)
padd: lma 0x00005848 vma 0x00000000 len 0xa7b0   (42928)
imap: lma 0x00010000 vma 0x42010000 len 0x1233c  (74556)
padd: lma 0x00022344 vma 0x00000000 len 0xdcd4   (56532)
dmap: lma 0x00030020 vma 0x3c000020 len 0x2478   (9336)
total segments stored 6

NuttShell (NSH) NuttX-12.5.1
nsh> 
```

Another one (ESP32-S3-WROOM-1U-N8R8) can't boot into nsh:
```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fc8ac40,len:0xdbc
load:0x40374000,len:0x4a54
SHA-256 comparison failed:
Calculated: 89806fd28d864acd62041e463d876296efdab8615d97ec5002d7c405cb83d17a
Expected: 00000000b0a70000000000000000000000000000000000000000000000000000
Attempting to boot anyway...
entry 0x40374d24
*** Booting NuttX ***
dram: lma 0x00000020 vma 0x3fc8ac40 len 0xdbc    (3516)
iram: lma 0x00000de4 vma 0x40374000 len 0x4a54   (19028)
padd: lma 0x00005848 vma 0x00000000 len 0xa7b0   (42928)
imap: lma 0x00010000 vma 0x42010000 len 0x1233c  (74556)
padd: lma 0x00022344 vma 0x00000000 len 0xdcd4   (56532)
dmap: lma 0x00030020 vma 0x3c000020 len 0x2478   (9336)
total segments stored 6
```
So the question is: is the 1U module supported? If it's supported, what changes are needed for it?

acassis, 2024-07-16:
@tmedicci any idea?

eren-terzioglu, 2024-07-16:
We will look, thanks for reporting.

acassis, 2024-07-16:
Thank you @eren-terzioglu

tmedicci, 2024-07-26:
Hi!

We could not find any issues regarding the software. It's probably related to some hardware configuration.

Please:
1. Ensure that you provide proper power to the module ESP32-S3-WROOM-1U-N8R8;
2. Ensure that the external antenna is properly attached (because this could generate power issues and noise);
3. Please, use openocd + gdb to properly check where it is failing;
4. Enable, for the working configuration, the following configs  and run `nxdiag --all` (and share the results, please):
```
CONFIG_SYSTEM_NXDIAG=y
CONFIG_SYSTEM_NXDIAG_COMP_FLAGS=y
CONFIG_SYSTEM_NXDIAG_CONF=y
CONFIG_SYSTEM_NXDIAG_ESPRESSIF=y
CONFIG_SYSTEM_NXDIAG_HOST_PATH=y
```

tmedicci, 2024-08-22:
As we didn't get any answer, I'm closing. Please reopen if needed.

closed 2024-08-22 by tmedicci