apache/nuttx #12712: ESP32C3 usbnsh hang on ps command

no1wudi, 2024-07-16:
When using usbnsh, I find that it frequently hangs:
```
NuttShell (NSH) NuttX-12.5.1
nsh> 
nsh> ps
  PID GROUP PRI POLICY   TYPE    NPX STATE    EVENT     SIGMASK           STACK COMMAND
    0     0   0 FIFO     Kthread   - Ready              0000000000000000 003056 Idle_Task
    1     1 100 RR       Task      - Running            0000000000000000 002000 nsh_main
nsh> ps
  PID GROUP PRI POLICY   TYPE    NPX STATE    EVENT     SIGMASK           STACK COMMAND
    0     0   0 FIFO     Kthread   - Ready              0000000000000000 003056 Idle_Task
    1     1 100 RR       Task      - Running            0000000000000000 002000 nsh_main
nsh> ps
  PID GROUP PRI POLICY   TYPE    NPX STATE    EVENT     SIGMASK           STACK COMMAND
    0     0   0 FIFO     Kthread   - Ready              0000000000000000 003056 Idle_Task
    1     1 100 RR       Task      - Running            0000000000000000 002000 nsh_main
nsh> ps
  PID GROUP PRI POLICY   TYPE    NPX STATE    EVENT     SIGMASK           STACK COMMAND
    0     0   0 FIFO     Kthread   - Ready              0000000000000000 003056 Idle_Task
    1     1 100 RR       Task      - Running            0000000000000000 002000 nsh_main
nsh> ps
  PID GROUP PRI POLICY   TYPE    NPX STATE    EVENT     SIGMASK           STACK COMMAND
    0     0   0 FIFO     Kthread   - Ready              0000000000000000 003056 Idle_Task
    1     1 100 RR       Task      - Running            0000000000000000 002000 nsh_main
nsh> ps
  PID GROUP PRI POLICY   TYPE    NPX STATE    EVENT     SIGMASK           STACK COMMAND
    0     0   0 FIFO     Kthread   - Ready              0000000000000000 003056 Idle_Task
    1     1 100 RR       Task      - Running            0000000000000000 002000 nsh_main
nsh> ps
  PID GROUP PRI POLICY   TYPE    NPX STATE    EVENT     SIGMASK           STACK COMMAND
    0     0   0 FIFO     Kthread   - Ready              0000000000000000 003056 Idle_Task
    1     1 100 RR       Task      - Running            0000000000000000 002000 nsh_main
nsh> ps
  PID GROUP PRI POLICY   TYPE    NPX STATE    EVENT     SIGMASK           STACK COMMAND
    0     0   0 FIFO     Kthread   - Ready              0000000000000000 003056 Idle_Task
    1     1 100 RR       Task      - Running            0000000000000000 002000 nsh_main
nsh> ps
  PID GROUP PRI POLICY   TYPE    NPX STATE    EVENT     SIGMASK           STACK COMMAND
    0     0   0 FIFO     Kthread   - Ready              0000000000000000 003056 Idle_Task
    1     1 100 RR       Task      - Running            0000000000000000 002000 nsh_main
nsh> ps
  PID GROUP PRI POLICY   TYPE    NPX STATE    EVENT     SIGMASK           STACK COMMAND
    0     0   0 FIFO     Kthread   - Ready              0000000000000000 003056 Idle_Task
    1     1 100 RR       Task      - Running            0000000000000000 002000 nsh_main
nsh> ps
  PID GROUP PRI POLICY   TYPE    NPX STATE    EVENT     SIGMASK           STACK COMMAND
    0     0   0 FIFO     Kthread   - Ready              0000000000000000 003056 Idle_Task
    1     1 100 RR       Task      - Running            0000000000000000 002000 nsh_main
nsh> ps
  PID GROUP PRI POLICY   TYPE    NPX STATE    EVENT     SIGMASK           STACK COMMAND
    0     0   0 FIFO     Kthread   - Ready              0000000000000000 003056 Idle_Task
    1     1 100 RR       Task      - Running            0000000000000000 002000 nsh_main
nsh> ps
  PID GROUP PRI POLICY   TYPE    NPX STATE    EVENT     SIGMASK           STACK COMMAND
    0     0   0 FIFO     Kthread   - Ready              0000000000000000 003056 
```

I'm tested with boards/xtensa/esp32s3/esp32s3-devkit/configs/usbnsh, this issue is easy to reproduce by type `ps` multiple times.

acassis, 2024-07-16:
@tmedicci @eren-terzioglu @fdcavalcanti please take a look

w2016561536, 2024-07-16:
I met this kind of bug too. This bug could be triggered after some heavy load such as coremark.

pkarashchenko, 2024-07-16:
I met similar some time ago. In my case the default stack was configured to too small value. Setting default task stack to 4K fixed the issue

pkarashchenko, 2024-07-16:
Ups. Just checked the code and was confused by `002000` interpreting this as hex value, but found `%06lu` is code. So still can be default stack size issue

pkarashchenko, 2024-07-16:
Maybe worth of changing `%06lu` to `%6lu` that will remove prefixed zeroes for decimal value...

no1wudi, 2024-07-17:
> I met similar some time ago. In my case the default stack was configured to too small value. Setting default task stack to 4K fixed the issue. Here I see the stack is set to 8K, so that should not be an issue. Would be good to enable stack coloration to see the % of stack consumption.

Thanks, it's the same issue:
```
NuttShell (NSH) NuttX-12.5.1
nsh> ps
  PID GROUP PRI POLICY   TYPE    NPX STATE    EVENT     SIGMASK           STACK   USED  FILLED COMMAND
    0     0   0 FIFO     Kthread   - Ready              0000000000000000 003056 000592  19.3%  Idle_Task
    1     1 100 RR       Task      - Running            0000000000000000 004048 001744  43.0%  nsh_main
nsh> ps
  PID GROUP PRI POLICY   TYPE    NPX STATE    EVENT     SIGMASK           STACK   USED  FILLED COMMAND
    0     0   0 FIFO     Kthread   - Ready              0000000000000000 003056 000592  19.3%  Idle_Task
    1     1 100 RR       Task      - Running            0000000000000000 004048 001888  46.6%  nsh_main
nsh> ps
  PID GROUP PRI POLICY   TYPE    NPX STATE    EVENT     SIGMASK           STACK   USED  FILLED COMMAND
    0     0   0 FIFO     Kthread   - Ready              0000000000000000 003056 000592  19.3%  Idle_Task
    1     1 100 RR       Task      - Running            0000000000000000 004048 002000  49.4%  nsh_main
nsh> ps
  PID GROUP PRI POLICY   TYPE    NPX STATE    EVENT     SIGMASK           STACK   USED  FILLED COMMAND
    0     0   0 FIFO     Kthread   - Ready              0000000000000000 003056 000592  19.3%  Idle_Task
    1     1 100 RR       Task      - Running            0000000000000000 004048 002000  49.4%  nsh_main
nsh> ps
  PID GROUP PRI POLICY   TYPE    NPX STATE    EVENT     SIGMASK           STACK   USED  FILLED COMMAND
    0     0   0 FIFO     Kthread   - Ready              0000000000000000 003056 000592  19.3%  Idle_Task
    1     1 100 RR       Task      - Running            0000000000000000 004048 002032  50.1%  nsh_main
nsh> ps
  PID GROUP PRI POLICY   TYPE    NPX STATE    EVENT     SIGMASK           STACK   USED  FILLED COMMAND
    0     0   0 FIFO     Kthread   - Ready              0000000000000000 003056 000592  19.3%  Idle_Task
    1     1 100 RR       Task      - Running            0000000000000000 004048 002128  52.5%  nsh_main
nsh> ps
  PID GROUP PRI POLICY   TYPE    NPX STATE    EVENT     SIGMASK           STACK   USED  FILLED COMMAND
    0     0   0 FIFO     Kthread   - Ready              0000000000000000 003056 000592  19.3%  Idle_Task
    1     1 100 RR       Task      - Running            0000000000000000 004048 002128  52.5%  nsh_main
nsh> ps
  PID GROUP PRI POLICY   TYPE    NPX STATE    EVENT     SIGMASK           STACK   USED  FILLED COMMAND
    0     0   0 FIFO     Kthread   - Ready              0000000000000000 003056 000592  19.3%  Idle_Task
    1     1 100 RR       Task      - Running            0000000000000000 004048 002128  52.5%  nsh_main
nsh> ps
  PID GROUP PRI POLICY   TYPE    NPX STATE    EVENT     SIGMASK           STACK   USED  FILLED COMMAND
    0     0   0 FIFO     Kthread   - Ready              0000000000000000 003056 000592  19.3%  Idle_Task
    1     1 100 RR       Task      - Running            0000000000000000 004048 002128  52.5%  nsh_main
nsh> ps
  PID GROUP PRI POLICY   TYPE    NPX STATE    EVENT     SIGMASK           STACK   USED  FILLED COMMAND
    0     0   0 FIFO     Kthread   - Ready              0000000000000000 003056 000592  19.3%  Idle_Task
    1     1 100 RR       Task      - Running            0000000000000000 004048 002128  52.5%  nsh_main
nsh> ps
  PID GROUP PRI POLICY   TYPE    NPX STATE    EVENT     SIGMASK           STACK   USED  FILLED COMMAND
    0     0   0 FIFO     Kthread   - Ready              0000000000000000 003056 000592  19.3%  Idle_Task
    1     1 100 RR       Task      - Running            0000000000000000 004048 002128  52.5%  nsh_main
nsh> ps
  PID GROUP PRI POLICY   TYPE    NPX STATE    EVENT     SIGMASK           STACK   USED  FILLED COMMAND
    0     0   0 FIFO     Kthread   - Ready              0000000000000000 003056 000592  19.3%  Idle_Task
    1     1 100 RR       Task      - Running            0000000000000000 004048 002128  52.5%  nsh_main
nsh> ps
  PID GROUP PRI POLICY   TYPE    NPX STATE    EVENT     SIGMASK           STACK   USED  FILLED COMMAND
    0     0   0 FIFO     Kthread   - Ready              0000000000000000 003056 000592  19.3%  Idle_Task
    1     1 100 RR       Task      - Running            0000000000000000 004048 002128  52.5%  nsh_main
nsh> ps
  PID GROUP PRI POLICY   TYPE    NPX STATE    EVENT     SIGMASK           STACK   USED  FILLED COMMAND
    0     0   0 FIFO     Kthread   - Ready              0000000000000000 003056 000592  19.3%  Idle_Task
    1     1 100 RR       Task      - Running            0000000000000000 004048 002128  52.5%  nsh_main
nsh> ps
  PID GROUP PRI POLICY   TYPE    NPX STATE    EVENT     SIGMASK           STACK   USED  FILLED COMMAND
    0     0   0 FIFO     Kthread   - Ready              0000000000000000 003056 000592  19.3%  Idle_Task
    1     1 100 RR       Task      - Running            0000000000000000 004048 002128  52.5%  nsh_main
nsh> ps
  PID GROUP PRI POLICY   TYPE    NPX STATE    EVENT     SIGMASK           STACK   USED  FILLED COMMAND
    0     0   0 FIFO     Kthread   - Ready              0000000000000000 003056 000592  19.3%  Idle_Task
    1     1 100 RR       Task      - Running            0000000000000000 004048 002128  52.5%  nsh_main
nsh> ps
[... cut ...]
    0     0   0 FIFO     Kthread   - Ready              0000000000000000 003056 000592  19.3%  Idle_Task
    1     1 100 RR       Task      - Running            0000000000000000 004048 002208  54.5%  nsh_main
nsh> ps
  PID GROUP PRI POLICY   TYPE    NPX STATE    EVENT     SIGMASK           STACK   USED  FILLED COMMAND
    0     0   0 FIFO     Kthread   - Ready              0000000000000000 003056 000592  19.3%  Idle_Task
    1     1 100 RR       Task      - Running            0000000000000000 004048 002208  54.5%  nsh_main
nsh> 
```
It won't hang if I increase the stack size for nsh.

closed 2024-07-17 by xiaoxiang781216