apache/nuttx #12748: Create an Issue template to automatically assigned the label tags when someone open an Issue

alancassis, 2024-07-22:
Refer to documentation:
https://docs.github.com/en/communities/using-templates-to-encourage-useful-issues-and-pull-requests/configuring-issue-templates-for-your-repository

When opening the issue the user will be asked about what this issue refers to, i.e.:

**This issue is a(n):** [ ] BUG, [ ] Crash, [ ] Feature Request, [ ] Improvement, [ ] Logic Error,  etc

**This issue apply to the architecture(s):** [ ] All, [ ] ARM, [ ] ARM64, [ ] AVR, [ ] CEVA [ ] HC, [ ] MIPS, etc

**What areas are this issue related to:** [ ] Audio, [ ] Clock Control, [ ] Ethernet, [ ] GPIO, [ ] Memory, etc

**Which vendors are affected:** [ ] AMD, [ ] Atmel/Microchip [ ] Broadcom, [ ] Espressif, [ ] Renesas, [ ] STMicroelectronics, etc

acassis, 2024-07-24:
Really cool @simbit18 !!! Thank you very much for implementing it.

I think for affected architecture we cannot use Combo-box, because an issue can affect two or three architecture.

Also it is important that "Related area" area also be an automatic label/tag.

simbit18, 2024-08-02:
hi @acassis  added an action for automatically labelling issues

PR #12828

closed 2024-08-02 by acassis