apache/nuttx #12767: CONFIG_CONSOLE_SYSLOG and CONFIG_SYSLOG_CONSOLE are misleading

raiden00pl, 2024-07-25:
These options are misleading. 

All other console enable options have the format like `CONFIG_{device_name}_CONSOLE`, but for SYSLOG it is `CONFIG_CONSOLE_SYSLOG`. Moreover there is `CONFIG_SYSLOG_CONSOLE` option but it has different purpose than other options with format `CONFIG_{device_name}_CONSOLE`.

```
config SYSLOG_CONSOLE
	bool "Log to /dev/console"
	default !ARCH_LOWPUTC && !SYSLOG_CHAR && !RAMLOG_SYSLOG && !SYSLOG_RPMSG && !SYSLOG_RTT
	depends on DEV_CONSOLE
	---help---
		Use the system console as a SYSLOG output device.
```

```
config CONSOLE_SYSLOG
	bool "Use SYSLOG for /dev/console"
	default n
	depends on DEV_CONSOLE && !SYSLOG_CONSOLE
	---help---
		Use the syslog logging device as a system console. If this feature is
		enabled (along with DEV_CONSOLE), then all console output will be
		re-directed to syslog output (syslog_putc). This is useful, for
		example, if the only console is a Telnet console.  Then in that case,
		console output from non-Telnet threads will go to the syslog output.
```

acassis, 2024-07-25:
I think CONFIG_CONSOLE_SYSLOG actually should be called CONFIG_SYSLOG_CONSOLE and actual CONFIG_SYSLOG_CONSOLE should be rename to CONFIG_SYSLOG_CONSOLE_FOR_LOG.

I think it is a good ideal to keep the consistence "CONFIG_{device_name}_CONSOLE" because we could create a CI check to detect two or more "*\_CONSOLE " in defconfig and avoid include it into mainline.

raiden00pl, 2024-07-26:
I agree that consistency is important, but on the other hand this change will break many user configs, so it will be important to document it well