apache/nuttx #12768: nrf9160/nrf9160-dk:modem_ns crash on start

maxikrie, 2024-07-25:
Hello,

I am trying to run the nrf9160-dk:modem_ns example, however it seems to crash very early in the startup routine. Specifically, it crashes in nrf91_clockconfig.c:75 when accessing a register. I have started looking into this and I feel it is related to the configuration `CONFIG_ARCH_TRUSTZONE_NONSECURE=y`. The nrf9160-dk:nsh example works fine, however when I change the one setting to `CONFIG_ARCH_TRUSTZONE_NONSECURE=y`it also fails at the first register access (through `putreg32`).

I am using the CMake build since the make build has an issue resolving the NRXFLIB URL. I am on nuttx-12.5.1 and I briefly tried nuttx-12.3.0, the release shortly after nrf9160-dk:modem_ns was added.

Tagging @raiden00pl, since you added this config.

I will continue looking into this, but maybe somebody can provide some insights?

Thanks!
Max

acassis, 2024-07-25:
@raiden00pl please take a look

raiden00pl, 2024-07-26:
It looks like I broke modem example with dd611b9c5ba8aca0d6d6901fb7ac51d0fce35e84

maxikrie, 2024-07-26:
Thanks for looking into this! Can you provide me a brief tutorial on the relation between the miniboot_s and modem_ns configuration? I am not familiar with the secure bootloader.

I compiled the miniboot_s configuration and flashed it to the board, but I don't know how to proceed. Also, I am on nuttx-12.5.1 or do I need to be on nuttx-12.5.0?

maxikrie, 2024-07-26:
Thanks, @raiden00pl, the additional documentation in https://github.com/apache/nuttx/pull/12778#issue-2431829826 helped!

closed 2024-07-26 by maxikrie

maxikrie, 2024-07-26:
@raiden00pl do you potentially also have instructions on how to test the modem? I see that wget is configured, but I am not sure if there is additional steps required in hardware and configuring the modem?

The nrf9160-dk comes with a iBASIS SIM card. Do you have experience with it?

raiden00pl, 2024-07-26:
You have to register your SIM card on https://nrfcloud.com/ and activate it. 

After chip reset, modem is in power off state, so you have to enable modem and if necessary configure it. At this moment, this must be done manually either with lapi from apps or with AT commands on `/dev/modem` (you can use `cu` program enabled by default for modem configuration). 
This should be done automatically by the LTE link-control daemon, but NuttX doesn't have anything like that yet, and I haven't had a chance to finish my implementation.

To start with, I recommend playing around with Nordic tools to understand how LTE-M works ([nRF Connect Cellular Monitor](https://infocenter.nordicsemi.com/topic/ug_cellular_monitor/UG/cellular_monitor/intro.html)

acassis, 2024-07-26:
@maxikrie if you get it working, please submit a documentation. Also creating blog posts helps to let more people to discover about NuttX.

maxikrie, 2024-08-06:
@acassis I got it working with a 1nce (https://1nce.com/) SIM card, which I was using with an Arduino board. I can also try with iBASIS SIM card, which comes with the nrf9160-dk and maybe more convenient. Where should I provide the documentation?

acassis, 2024-08-06:
@maxikrie I think at Modem Support here: https://nuttx.apache.org/docs/latest/platforms/arm/nrf91/index.html

maxikrie, 2024-08-06:
@raiden00pl Thank you again for your amazing work, this helped me to get started quite easily. Very satisfying to be able to just open a socket()! I have a few follow-up questions:

1. I tried break-point debugging with Ozone, but I only can get it working with the nsh example. The different memory layout with the combination of miniboot_s and modem_ns seems to mess up things and I likely have it not configured correctly. Did you try this?
2. I am still quite new to usrsock in Nuttx. I looked at the gs220m and alt1250 implementation, which employ an additional daemon and also employ '/dev/usrsock' while for the nrf9160 it doesn't have that. Instead NET_USRSOCK_CUSTOM is set. Is this what you mean by missing the 'LTE link-control daemon' or what is the difference between NET_USRSOCK_DEVICE and NET_USRSOCK_CUSTOM?

raiden00pl, 2024-08-08:
> I tried break-point debugging with Ozone, but I only can get it working with the nsh example. The different memory layout with the combination of miniboot_s and modem_ns seems to mess up things and I likely have it not configured correctly. Did you try this?

I've worked with gdb only so I can't help.

> I am still quite new to usrsock in Nuttx. I looked at the gs220m and alt1250 implementation, which employ an additional daemon and also employ '/dev/usrsock' while for the nrf9160 it doesn't have [...]

You can look at alt1250 architecture here: https://github.com/apache/nuttx-apps/blob/master/lte/alt1250/README.txt
alt1250 implements user socket API on apps side and NET_USRSOCK_DEVICE provides interface to communicate with kernel side part of alt1250 implementation (like SPI data transfer).
For nrf91 all user socket API is implemented on kernel side, so we don't need `/dev/usrsock`. 
When I write 'LTE link-control daemon', I mean something like https://docs.nordicsemi.com/bundle/ncs-latest/page/nrf/libraries/modem/lte_lc.html so we can automate the modem configuration and control.

> I often have the issue that when I only want to flash changes to the modem_ns build, the board bricks and I have to first reflash the miniboot_s build (chiperasing everything during the operation) and then again flashing the modem_s. Do you also encounter this issue?

Sounds like AP-protect mechanism in Nordic chips: https://docs.nordicsemi.com/bundle/ncs-2.6.0/page/nrf/security/ap_protect.html 
It blocks the debugger and the only option to unlock the system is to erase the chip.
AP-protect mechanism is broken for NRF52 (famous NRF52 fault injection bug) so for new chips Nordic enables AP-protect at default in HW. But I don't remember if it's enabled by default for NRF91, probably it probably is. 
The missing logic is here:
https://github.com/apache/nuttx/blob/3862b717286ddd6ad8375639daa294844b817ade/arch/arm/src/nrf91/nrf91_start.c#L79-L88