apache/nuttx #12774: Adding Board Support in Nuttx: Steps and Requirements

i-tesetd-it-no-problem, 2024-07-26:
Hello, I'm planning to start working with Nuttx recently. However, I couldn't find documentation on porting Nuttx to a board that isn't supported in the source code. What steps do I need to take to add support for my own board? For example, should I copy files from arch/arm and board/ directories?

i-tesetd-it-no-problem, 2024-07-26:
I copied the gd32f4 directory from arm/arm to gd32f3 and also copied the gd32f4 directory from the boards folder to gd32f3. Additionally, I made some changes to the Kconfig file.

However, I encountered the following error when generating the configuration.

```bash
wenshuyu@wenshuyu:~/my_nuttx/nuttx$ ./tools/configure.sh -L | grep "gd32"
  gd32f303zg-eval:nsh
  gd32f303zg-eval:sdionsh
  gd32f303zg-eval:netnsh
  gd32f303zg-eval:fmclfsnsh
  gd32f303zg-eval:i2c0testnsh
  gd32f303zg-eval:spilfsnsh
  gd32f470zk-eval:nsh
  gd32f470zk-eval:sdionsh
  gd32f470zk-eval:netnsh
  gd32f470zk-eval:fmclfsnsh
  gd32f470zk-eval:i2c0testnsh
  gd32f470zk-eval:spilfsnsh
  gd32f470zk-aiotbox:aiotboxnsh
  gd32f450zk-aiotbox:aiotboxnsh
  gd32f470ik-eval:nsh
  gd32f470ik-eval:netnsh
  gd32f470ik-eval:fmclfsnsh
  gd32f470ik-eval:i2c0testnsh
  gd32f450zk-eval:nsh
  gd32f450zk-eval:sdionsh
  gd32f450zk-eval:netnsh
  gd32f450zk-eval:fmclfsnsh
  gd32f450zk-eval:i2c0testnsh
  gd32f450zk-eval:spilfsnsh
  gd32f450zk-eval:btngpionsh
  gd32f450zk-eval:spifsnsh
  gd32f450zk-eval:fmcfsnsh
wenshuyu@wenshuyu:~/my_nuttx/nuttx$ ./tools/configure.sh -l gd32f303zg-eval:nsh
  Copy files
  Select CONFIG_HOST_LINUX=y
  Refreshing...
CP: arch/dummy/Kconfig to /home/wenshuyu/my_nuttx/nuttx/arch/dummy/dummy_kconfig
CP: boards/dummy/Kconfig to /home/wenshuyu/my_nuttx/nuttx/boards/dummy/dummy_kconfig
LN: platform/board to /home/wenshuyu/my_nuttx/apps/platform/dummy
LN: include/arch to arch/arm/include
LN: include/arch/board to /home/wenshuyu/my_nuttx/nuttx/boards/arm/gd32f3/gd32f303zg-eval/include
LN: drivers/platform to /home/wenshuyu/my_nuttx/nuttx/drivers/dummy
LN: include/arch/chip to /home/wenshuyu/my_nuttx/nuttx/arch/arm/include/gd32f3
LN: arch/arm/src/chip to /home/wenshuyu/my_nuttx/nuttx/arch/arm/src/gd32f3
LN: arch/arm/src/board to /home/wenshuyu/my_nuttx/nuttx/boards/arm/gd32f3/gd32f303zg-eval/src
mkkconfig in /home/wenshuyu/my_nuttx/apps/audioutils
mkkconfig in /home/wenshuyu/my_nuttx/apps/benchmarks
mkkconfig in /home/wenshuyu/my_nuttx/apps/boot
mkkconfig in /home/wenshuyu/my_nuttx/apps/canutils
mkkconfig in /home/wenshuyu/my_nuttx/apps/crypto
mkkconfig in /home/wenshuyu/my_nuttx/apps/database
mkkconfig in /home/wenshuyu/my_nuttx/apps/examples/mcuboot
mkkconfig in /home/wenshuyu/my_nuttx/apps/examples
mkkconfig in /home/wenshuyu/my_nuttx/apps/fsutils
mkkconfig in /home/wenshuyu/my_nuttx/apps/games
mkkconfig in /home/wenshuyu/my_nuttx/apps/graphics
mkkconfig in /home/wenshuyu/my_nuttx/apps/industry
mkkconfig in /home/wenshuyu/my_nuttx/apps/inertial
mkkconfig in /home/wenshuyu/my_nuttx/apps/interpreters/luamodules
mkkconfig in /home/wenshuyu/my_nuttx/apps/interpreters
mkkconfig in /home/wenshuyu/my_nuttx/apps/logging
mkkconfig in /home/wenshuyu/my_nuttx/apps/lte
mkkconfig in /home/wenshuyu/my_nuttx/apps/math
mkkconfig in /home/wenshuyu/my_nuttx/apps/mlearning
mkkconfig in /home/wenshuyu/my_nuttx/apps/netutils
mkkconfig in /home/wenshuyu/my_nuttx/apps/sdr
mkkconfig in /home/wenshuyu/my_nuttx/apps/system
mkkconfig in /home/wenshuyu/my_nuttx/apps/testing
mkkconfig in /home/wenshuyu/my_nuttx/apps/videoutils
[... 43 lines cut ...]
arch/arm/src/gd32f4/Kconfig:2149:warning: choice value used outside its choice group
arch/arm/src/gd32f4/Kconfig:2154:warning: choice value used outside its choice group
arch/arm/src/gd32f4/Kconfig:2159:warning: choice value used outside its choice group
arch/arm/src/gd32f4/Kconfig:2380:warning: choice value used outside its choice group
arch/arm/src/gd32f4/Kconfig:2385:warning: choice value used outside its choice group
arch/arm/src/gd32f4/Kconfig:2390:warning: choice value used outside its choice group
arch/arm/src/gd32f4/Kconfig:2523:warning: choice value used outside its choice group
arch/arm/src/gd32f4/Kconfig:2528:warning: choice value used outside its choice group
arch/arm/src/gd32f4/Kconfig:2533:warning: choice value used outside its choice group
boards/arm/gd32f4/gd32f470zk-eval/Kconfig:36:warning: choice value used outside its choice group
boards/arm/gd32f4/gd32f470zk-eval/Kconfig:41:warning: choice value used outside its choice group
boards/arm/gd32f4/gd32f470zk-eval/Kconfig:46:warning: choice value used outside its choice group
boards/arm/gd32f4/gd32f470zk-eval/Kconfig:56:warning: choice value used outside its choice group
boards/arm/gd32f4/gd32f470zk-eval/Kconfig:60:warning: choice value used outside its choice group
boards/arm/gd32f4/gd32f470zk-eval/Kconfig:76:warning: choice value used outside its choice group
boards/arm/gd32f4/gd32f470zk-eval/Kconfig:79:warning: choice value used outside its choice group
boards/arm/gd32f4/gd32f470zk-eval/Kconfig:82:warning: choice value used outside its choice group
boards/arm/gd32f4/gd32f470zk-eval/Kconfig:85:warning: choice value used outside its choice group
boards/arm/gd32f4/gd32f470zk-eval/Kconfig:103:warning: choice value used outside its choice group
boards/arm/gd32f4/gd32f470zk-eval/Kconfig:111:warning: choice value used outside its choice group
boards/arm/gd32f4/gd32f470zk-eval/Kconfig:119:warning: choice value used outside its choice group
#
# configuration written to .config
#
```

acassis, 2024-07-26:
@i-tesetd-it-no-problem there is an old Porting Guide here: https://cwiki.apache.org/confluence/display/NUTTX/Porting+Guide

Are you in NuttX mailing list? @linguini1 asked it one or two days ago!

You can copy all the files like you did, however you could bring many things that doesn't pertains to gd32f3 family. A better approach is creating the arch/arm/src/gd32f3, then add the create a hardware/ inside to to include the header files to the chip you want to add. Then add Kconfig and Makefile/CMakefile and the start.c, heapalloc, etc.

Adding on demand will avoid making mistakes.

i-tesetd-it-no-problem, 2024-07-27:
@acassis 
Thank you, I'm trying to add my own board using the GD32F3 series chips. I copied the 'gd32f4' directory under 'arch' to 'gd32f3'. When I tried to modify the source code in 'gd32f3/src', I found it very cumbersome and difficult. I constantly need to refer to the chip manual to adjust corresponding offsets and so on. Can I directly replace the files in 'src' with the firmware library provided by the chip manufacturer? This way, I only need to follow the driver's registration rules to implement it, which can save a lot of work.

acassis, 2024-07-27:
@i-tesetd-it-no-problem NuttX needs these definitions, there is not shortcuts for it. Actually, there are some exceptions, like Espressif found an way to integrate their HAL into arch. But Greg Nutt (@patacongo) always was against doing this way. Why? Because normally semiconductor companies are interested to release their chip in the market ASAP, not to create the best code of the world.

When each manufacturer use their own HAL the final code is very different between each architecture, as you probably saw on some poor quality RTOSes.

 There is a python script to convert CMSIS files to NuttX header files created by Robert Lipe: https://github.com/robertlipe/svdtoheaders

You can try to use it if you don't want to update the header files manually