apache/nuttx #12785: mkrd fail in rc.sysinit

geniusdo, 2024-07-27:
I can use command mkrd -m 0 -s 512 512 correctly in NSH, but when I put it in rc.sysinit it gives nsh: mkfatfs: mkfatfs failed: 13. I am wondering if there is anything I missed?

geniusdo, 2024-07-27:
Btw, I put this command in rcS script, it worked perfectly. Any differences between rc.sysinit and rcS file.

acassis, 2024-07-27:
Hi @geniusdo are you following this Guide: https://nuttx.apache.org/docs/latest/guides/etcromfs.html ?

Look: "rcS is the start-up script, where rc.sysinit is the system-init script." 

The error -13 means "Permission denied"

Both rc.sysinit and rcS are very similar, but there are some small details in the order they are called:

Look at apps/nshlib/nsh_init.c and you will see:

rc.sysinit is called early:

```
#if defined(CONFIG_ETC_ROMFS) && !defined(CONFIG_NSH_DISABLESCRIPT)
  pstate = nsh_newconsole(false);

  /* Execute the system init script */

  nsh_sysinitscript(&pstate->cn_vtbl);
#endif
```

rcS is called later:

```
#if defined(CONFIG_ETC_ROMFS) && !defined(CONFIG_NSH_DISABLESCRIPT)
  /* Execute the start-up script */

  nsh_initscript(&pstate->cn_vtbl);
  nsh_release(&pstate->cn_vtbl);
#endif
```

Maybe we should comment that rcS is better place to call user code

geniusdo, 2024-07-27:
Thanks for ur reply.@acassis.  Yes, I followed the guide to migrate the custom start up script from an earlier version of nuttx. I am curious why I got the permission deny error. I noticed that in board/sim, there are password related files. Do I need to login as root or sth to execute this command?

acassis, 2024-07-27:
@geniusdo the default nsh in sim request login (user/pass = admin:Administrator), you can modify to disable NSH_LOGIN. I think the rc script is executed before the login, read the comment above nsh_initialize(void) function.

There is also a nsh_loginscript that is executed everytime the user log in the nsh.

geniusdo, 2024-07-28:
@acassis thx, I will check the code. Appreciate the detailed explanation. Closing this issue.

closed 2024-07-28 by geniusdo