apache/nuttx #12798: Cannot Run ESP32C6-devkitm

FelipeMdeO, 2024-07-30:
Hello all, 

I am trying use Nuttx in ESP32C6-devkitm.
I followed the tutorial: https://nuttx.apache.org/docs/latest/platforms/risc-v/esp32c6/index.html
Below you can se my outputs:
`~/nuttxspace/nuttx$ make bootloader
Using direct bootloader to boot NuttX.
`
`riscv-none-elf-ld: warning: /home/felipe-moura/nuttxspace/nuttx/nuttx has a LOAD segment with RWX permissions
CP: nuttx.hex
MKIMAGE: NuttX binary
esptool.py -c esp32c6 elf2image --ram-only-header -fs 4MB -fm dio -ff 80m -o nuttx.bin nuttx
esptool.py v4.8.dev4
Creating esp32c6 image...
Image has only RAM segments visible. ROM segments are hidden and SHA256 digest is not appended.
Merged 1 ELF section
Successfully created esp32c6 image.
Generated: nuttx.bin
esptool.py -c esp32c6 -p /dev/ttyACM0 -b 921600  write_flash -fs 4MB -fm dio -ff 80m 0x0000 nuttx.bin
esptool.py v4.8.dev4
Serial port /dev/ttyACM0
Connecting...
Chip is ESP32-C6FH4 (QFN32) (revision v0.0)
Features: WiFi 6, BT 5, IEEE802.15.4
Crystal is 40MHz
MAC: 54:32:04:ff:fe:07:14:94
BASE MAC: 54:32:04:07:14:94
MAC_EXT: ff:fe
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Flash will be erased from 0x00000000 to 0x0003cfff...
Compressed 246716 bytes to 104167...
Wrote 246716 bytes (104167 compressed) at 0x00000000 in 2.0 seconds (effective 978.2 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...

`
But after i connect in the board I am getting bootloader issue.
I am not sure if my bootloader process is right, can you help me, please?

`
Terminal ready
load:0x408042f0,len:0x884
SHA-256 comparison failed:
Calculated: 113a530452b06587e1697284c537b530939f3b6c0afde0ea002b4170b843b604
Expected: 0000000050b40000000000000000000000000000000000000000000000000000
Attempting to boot anyway...
entry 0x40803f96
ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0x7 (TG0_WDT_HPSYS),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x40800b10
SPIWP:0xee
mode:DIO, clock div:2
load:0x40800000,len:0x42e8
load:0x408042f0,len:0x884
SHA-256 comparison failed:
Calculated: 113a530452b06587e1697284c537b530939f3b6c0afde0ea002b4170b843b604
Expected: 0000000050b40000000000000000000000000000000000000000000000000000
Attempting to boot anyway...
entry 0x40803f96
ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0x7 (TG0_WDT_HPSYS),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x40800b10
SPIWP:0xee
mode:DIO, clock div:2
load:0x40800000,len:0x42e8
load:0x408042f0,len:0x884
SHA-256 comparison failed:
Calculated: 113a530452b06587e1697284c537b530939f3b6c0afde0ea002b4170b843b604
Expected: 0000000050b40000000000000000000000000000000000000000000000000000
Attempting to boot anyway...
entry 0x40803f96
ESP-ROM:esp32c6-20220919
Build:Sep 19 2022

`
steps:
./tools/configure.sh esp32c6-devkitc:nsh
make
make bootloader 
make flash ESPTOOL_PORT=<port> ESPTOOL_BINDIR=./
picocom...

eren-terzioglu, 2024-07-30:
Hi,

Seems you are using direct bootloader, you don't need to use `make bootloader` command. Please apply this steps:

```
make distclean
./tools/configure.sh esp32c6-devkitm:nsh
make
# ESPTOOL_PORT value might change
make download ESPTOOL_PORT=/dev/ttyUSB0
# picocom ...
```

If problem still happens, please inform and please inlcude compiler and versions (nuttx and nuttx-apps) you are using.

FelipeMdeO, 2024-07-30:
Hello @eren-terzioglu , thank you for help me. Issue still happens:
Please, look file attached, there I put the terminal output
[term_out.txt](https://github.com/user-attachments/files/16428318/term_out.txt)

About my configs:
I am working inside nuttx docker: https://github.com/apache/nuttx/blob/master/tools/ci/docker/linux/Dockerfile
I clonned nuttx using:
`https://developer.espressif.com/blog/nuttx-getting-started/`

Below you can se my tools version:
`riscv-none-elf-gcc --version
riscv-none-elf-gcc (xPack GNU RISC-V Embedded GCC x86_64) 13.2.0
Copyright (C) 2023 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
`
root@1079d9310790:/nuttxspace/nuttx# cat .version 
#!/bin/bash

CONFIG_VERSION_STRING="12.6.0-RC1"
CONFIG_VERSION_MAJOR=12
CONFIG_VERSION_MINOR=6
CONFIG_VERSION_PATCH=0
CONFIG_VERSION_BUILD="cd4fdf27c5"`
`

eren-terzioglu, 2024-07-30:
I tested with an esp32c6 with this specs and it is working well (only difference is revision 0.0 -> 0.1):

```
esptool.py v4.8.dev4
Found 2 serial ports
Serial port /dev/ttyUSB0
Connecting....
Detecting chip type... ESP32-C6
Chip is ESP32-C6FH4 (QFN32) (revision v0.1)
Features: WiFi 6, BT 5, IEEE802.15.4
Crystal is 40MHz
```
Are you using usb port or uart port? Seems you are USB port, please change it to uart port.

FelipeMdeO, 2024-07-30:
You are right, I was using wrong usb connector. After change for UART port it is working properly. Thank you.

closed 2024-07-30 by FelipeMdeO