apache/nuttx #12815: Add esp32c6 power manager feature

FelipeMdeO, 2024-07-31:
I want apply power manager to esp32c6.
Try activate it in the menu config I'm facing build issue:

`riscv-none-elf-ld: warning: /home/felipe-moura/nuttxspace/nuttx/nuttx has a LOAD segment with RWX permissions
riscv-none-elf-ld: /home/felipe-moura/nuttxspace/nuttx/staging/libarch.a(riscv_initialize.o): in function `up_initialize':
riscv_initialize.c:(.text.up_initialize+0x4): undefined reference to `riscv_pminitialize'
make[1]: *** [Makefile:189: nuttx] Error 1
make: *** [tools/Unix.mk:546: nuttx] Error 2`

Looking in the code I was able to see that the esp32c3 can build. It's happen because esp32c3 has a file with "function implementation":
[image]

I want enable this feature to esp32c6, but I am confusing about the code development process. Esp32c3 has a folder called esp32c3-legacy, inside it be function `void riscv_pminitialize(void)`, this the right way to organize thinks, I mean, this function is common to esp32c3 and c6, so it can be available in an same file, for example `esp32_pminitialize.c` link in xtensa arquitecture.

[image]

eren-terzioglu, 2024-08-01:
Hello,

`esp32c3-legacy` approach is in maintenence mode. Currently we are adding driver support for our risc-v devices under `arch/risc-v/src/common/espressif` folder. This is a common folder for esp32c3, esp32c6 and esp32h2. We don't have any support for PM for our common approach so there is no way to use it right now.

FelipeMdeO, 2024-08-01:
Hello @eren-terzioglu ,

Can I help to add driver support for risc-v devices? Do you have an specific topic or email thread about it ?

eren-terzioglu, 2024-08-01:
NuttX is an open source project. If you want to add a support for our devices, of course you can. Just you need to do is open a pull request to this repository.

closed 2024-08-01 by FelipeMdeO