apache/nuttx #12875: [HELP] ESP32 chip revision ERROR

TarikHamedovic, 2024-08-09:
### Description

Hello

I ran intro trouble trying to run Nuttx on a ESP32 board, I tried to configure it to run on espdevkitc and after all of the steps in the documentation and running the terminal through Putty and clicking the `EN` pin I get the following message:

```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip:0, SPIWP:0xee
clk_drv:0x00, q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO,clock_div:2
load:0x3ffb1f50,len:1372
ho 0 tail 12 room 4
load:0x40080000, len:19796
entry 0x40082504
*** Booting Nuttx ***
dram:  lma 0x00001020 vma 0x3ffb1f50   len 0x55c          (1372)
iram:   lma 0x00001584 vma 0x40080000 len 0x4d54       (19796)
padd:  lma 0x000062e8 vma 0x00000000 len 0x9d10       (40208)
imap:  lma 0x00010000 vma 0x400e0000 len 0x11788     (71560)
padd:  lma 0x00021790 vma 0x00000000 len 0xe888       (59528)
dmap: lma 0x00030020 vma 0x4f400020 len 0x2b9c        (11164)
total segmets stored 6
ERROR: Nuttx supports ESP32 chip revision >= v3.0(chip revision is v1.0)
```

### Verification

- [X] I have verified before submitting the report.

eren-terzioglu, 2024-08-10:
Hi,

This feature is for some functional problems on old revisions of ESP32 chips. To bypass this revision check (some stuff might not work as intended) you can use `-DESP32_IGNORE_CHIP_REVISION_CHECK` flag during building (e.g. make EXTRAFLAGS="-DESP32_IGNORE_CHIP_REVISION_CHECK").

closed 2024-08-10 by TarikHamedovic

tmedicci, 2024-08-12:
> ### Description
> Hello
>
> I ran intro trouble trying to run Nuttx on a ESP32 board, I tried to configure it to run on espdevkitc and after all of the steps in the documentation and running the terminal [...]

Hi @TarikHamedovic , please note the error message: your chip revision is quite old and contains a lot of unsuported features that may affect your experience in NuttX. Please, don't use `-DESP32_IGNORE_CHIP_REVISION_CHECK` for anything other than testing (nor consider your tests results in case of failures as valid using this chip revision).