apache/nuttx #12897: [BUG] ESP326 DevKitM spi issue

FelipeMdeO, 2024-08-12:
### Description / Steps to reproduce the issue

Hello All,

I am trying use spi in esp32c6 DevKit M but I am facing issues.
I tried use default spi example, it fails some time, please look image below:
[image]
So I used logic analyzer to try debug whats happen and spi app has strange behavior. I cannot get clocks and cannot see MISO/MOSI bits.
Also, I implemented a app by myself and I had same behavior, I cannot see clocks and data using logic analyzer and oscilloscope.
Follow my app example:

```
#include <nuttx/config.h>
#include <nuttx/spi/spi_transfer.h>
#include <fcntl.h>
#include <unistd.h>
#include <stdio.h>
#include <stdint.h>

#include "spitool.h"

#define SPI_BUS 2  // SPI bus number
#define SPI_DEVICE 0  // SPI device (chip select)
#define SPI_MODE 0  // SPI mode (0, 1, 2, 3)
#define SPI_FREQUENCY 1000000  // SPI frequency (1 MHz)
#define SPI_BITS 8  // Number of bits per SPI word
#define SPI_DELAY 0  // Delay between transfers (in microseconds)

void send_spi_data(int fd, uint8_t *txbuffer, int length)
{
    struct spi_sequence_s seq;
    struct spi_trans_s trans;

    // Configure the transfer sequence
    seq.dev = SPIDEV_ID(SPIDEVTYPE_USER, SPI_DEVICE);
    seq.mode = SPI_MODE;
    seq.nbits = SPI_BITS;
    seq.frequency = SPI_FREQUENCY;
    seq.ntrans = 1;
    seq.trans = &trans;

    // Configure the transfer
    trans.deselect = true;
    trans.delay = SPI_DELAY;
    trans.nwords = length;
    trans.txbuffer = txbuffer;
    trans.rxbuffer = NULL;  // We are not receiving data

    // Perform the SPI transfer
    if (spidev_transfer(fd, &seq) < 0) {
        perror("SPI transfer failed");
    } else {
        printf("SPI transfer succeeded\n");
    }
}

int main(void)
{
    int fd;
    uint8_t txbuffer[] = {0xDE, 0xAD, 0xBE, 0xEF};  // Known data buffer

    // Open the SPI device
    fd = spidev_open(SPI_BUS);
    if (fd < 0) {
        perror("Failed to open SPI device");
        return -1;
    }

    while (1) {
        // Send data via SPI
        send_spi_data(fd, txbuffer, sizeof(txbuffer));

        // Wait for 1 second
        sleep(1);
    }

    // Close the SPI device
    close(fd);
    return 0;
}

```

Below, follow logic analyzer and oscilloscope captures:
[image]
[image]
[image]
[image]
[image]

Please, anyone can check it in your side?

### On which OS does this issue occur?

[Linux]

### What is the version of your OS?

Ubuntu 23.1

### NuttX Version

master

### Issue Architecture

[risc-v]

### Issue Area

[Applications], [Drivers]

### Verification

- [X] I have verified before submitting the report.

tmedicci, 2024-08-12:
Hi @FelipeMdeO , 

This feature is constantly tested by our internal CI and we found no related issues regarding SPI master. Let's try to debug to find the root cause of the issue you reported.

First of all, please try the attached firmware. You can flash it with:
```
esptool.py -c esp32c6 -p /dev/ttyUSB0 -b 921600 write_flash -fs detect -fm dio 0x0 nuttx.merged.bin
```
[build_test_spi_master_esp32c6.zip](https://github.com/user-attachments/files/16582907/build_test_spi_master_esp32c6.zip)

You don't need to make any external connection (`CONFIG_ESPRESSIF_SPI_TEST_MODE` is enabled). Just run the same commands and you should get the expected result.

That being said, some questions:
- Which `defconfig` did you use?
- Which pins did you connect externally?

Finally, can you please enable the following configs:
```
CONFIG_SYSTEM_NXDIAG=y
CONFIG_SYSTEM_NXDIAG_ESPRESSIF=y
```

and, on NuttX, run `nxdiag --all` and share your results, please?

FelipeMdeO, 2024-08-12:
Hello @tmedicci , 

**Which defconfig did you use?**
    ./tools/configure.sh esp32c6-devkitm:spi
    
**Which pins did you connect externally?**
[image]
[image]

Brown - CS PIN 19
Red - FSPIQ 2
Yellow - CLK 6
Orange - FSPID 7

FelipeMdeO, 2024-08-12:
About nxdiag output:
Follow it:

```
nsh> nxdiag --all
Nxdiag Report:

NuttX RTOS info:
	Hostname: 
	Release: 12.6.0-RC1
	Build: 3c05da536a-dirty Aug 12 2024 15:01:44
	Arch: risc-v
	Defconfig: esp32c6-devkitm:spi-dirty

Host system OS:
	Ubuntu 24.04 LTS Linux felipe-moura 6.8.0-40-generic #40-Ubuntu SMP PREEMPT_DYNAMIC Fri Jul  5 10:34:03 UTC 2024 x86_64 x86_64

Espressif specific information:

Bootloader version:
	esp32: Bootloader image not found
	esp32s2: Bootloader image not found
	esp32s3: Bootloader image not found
	esp32c2: Bootloader image not found
	esp32c3: Bootloader image not found
	esp32c6: Bootloader image not found
	esp32h2: Bootloader image not found

Toolchain version:
	clang: Not found
	gcc: gcc (Ubuntu 13.2.0-23ubuntu4) 13.2.0
	xtensa-esp32-elf-gcc: xtensa-esp32-elf-gcc (crosstool-NG esp-12.2.0_20230208) 12.2.0
	xtensa-esp32s2-elf-gcc: Not found
	xtensa-esp32s3-elf-gcc: xtensa-esp32s3-elf-gcc (crosstool-NG esp-12.2.0_20230208) 12.2.0
	riscv32-esp-elf-gcc: Not found
	riscv64-unknown-elf-gcc: Not found

Esptool version: 4.8.dev4

HAL version: sync/release_v5.1.c-nuttx-20230814-65-g8e8e62cf6a
```

FelipeMdeO, 2024-08-12:
Hello @tmedicci, 

about test you firmware, It isn't work properly also, I cannot get spi signals properly, however miso and mosi are equal even it aren't connected one with other, I think your version is running in loopback situation. Please look video attached:

https://github.com/user-attachments/assets/046d4dad-d9b4-422d-a9b7-a4edb95a3429

tmedicci, 2024-08-12:
> nxdiag

Yes, it's a loopback. You will not be able to measure anything, the pins are connected internally and the result was as expected: same content sent is successfully read.

> Hello @tmedicci ,
>
> **Which defconfig did you use?** ./tools/configure.sh esp32c6-devkitm:spi
>
> **Which pins did you connect externally?** [...]

I didn't get: in this test aren't you connecting MOSI with MISO? (pins 2 and 7 connected together) to test the loopback? If not, what is the slave device?

FelipeMdeO, 2024-08-12:
Hello @tmedicci,

my early tests I keep MISO ando MOSI "floating" only connected to my logic analyzer, I believe I should at least view the clocks, right?

tmedicci, 2024-08-12:
Oh, I see... Can you confirm that the loopback test is working as expected? (connecting pins 2 and 7)

@eren-terzioglu , can you check about the clock line, please?

FelipeMdeO, 2024-08-12:
Hello @tmedicci, @eren-terzioglu .

I am facing issue using external loopback too. I connected directly, without "protoboard" and keep issue.
Loop my setup:
[image]

Look for result:
[image]

eren-terzioglu, 2024-08-13:
We reproduced the issue, investigating it. Thanks for reporting.

FelipeMdeO, 2024-08-13:
Hello @tmedicci  and @eren-terzioglu . First of all thank for help.

Just to organize my last posts:

1 - Difference between data sent and received by spi
I just realized that this problem only occurs when I change the CS pin from 10 to any other pin. I need to change this pin because pin 10 is not exposed on the esp32c6 devkit m header.

2 - I can't see spi clock signals using a logic analyzer.

FelipeMdeO, 2024-08-13:
Hello @eren-terzioglu 
The "issue" not happen when I check option "SPI software CS", maybe this is expected behavior.

[image]

acassis, 2024-08-14:
@eren-terzioglu @tmedicci in fact the documentation is confusing: https://nuttx.apache.org/docs/latest/applications/system/spi/index.html doesn't say we need to connect MOSI to MISO (it is obvious for us, but not for someone outside the project). Also the text is citing "I2C Build Configuration" that is not related to that matter. I'll submit a fix to the documentation.

acassis, 2024-08-14:
Done: https://github.com/apache/nuttx/pull/12913

FelipeMdeO, 2024-08-15:
Hello @xiaoxiang781216, this topic is about spi issue. It wasn't solved yet. I think this topic must be kept open.

eren-terzioglu, 2024-08-15:
> Hello @xiaoxiang781216, this topic is about spi issue. It wasn't solved yet. I think this topic must be kept open.

Created https://github.com/apache/nuttx/pull/12924

xiaoxiang781216, 2024-08-15:
Let me reopen this issue and link to https://github.com/apache/nuttx/pull/12924

closed 2024-08-17 by xiaoxiang781216