apache/nuttx #12905: [HELP] Cannot debug esp32c6 using open ocd

FelipeMdeO, 2024-08-13:
### Description

Description

Hello All,

I am trying debug esp32c6 using open ocd, I followed tutorial in link:
 [[https://nuttx.incubator.apache.org/docs/latest/platforms/risc-v/esp32c6/index.html#debugging-with-openocd](url)
 
`openocd -c 'set ESP_RTOS hwthread; set ESP_FLASH_SIZE 0' -f board/esp32c6-builtin.cfg`

```
Open On-Chip Debugger v0.12.0-esp32-20240726 (2024-08-13-15:00)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
0
Info : only one transport option; autoselecting 'jtag'
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Info : esp_usb_jtag: capabilities descriptor set to 0x2000
Info : Hardware thread awareness created
WARNING: ESP flash support is disabled!
force hard breakpoints
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Error: esp_usb_jtag: could not find or open device!
/usr/local/bin/../share/openocd/scripts/target/esp_common.cfg:9: Error: 
at file "/usr/local/bin/../share/openocd/scripts/target/esp_common.cfg", line 9
Error: [esp32c6] Unsupported DTM version: -1
Error: [esp32c6] Could not identify target type.
```
Do you have any idea what's happen?

### Verification

- [X] I have verified before submitting the report.

acassis, 2024-08-13:
@fdcavalcanti @tmedicci please take a look!

@FelipeMdeO seems similar to this issue: https://github.com/espressif/openocd-esp32/issues/315

tmedicci, 2024-08-13:
Hi @FelipeMdeO ,

Considering the message `Error: esp_usb_jtag: could not find or open device!`, it seems the USB/JTAG port is not connected. Can you please double-check if the 2nd USB port (the one in the right) is connected?

You should expected something like that on your `dmesg` when you attach the USB-C cable to the USB/JTAG port of the ESP32-C6:
```
[1214203.144071] usb 3-1.2.2: new full-speed USB device number 109 using xhci_hcd
[1214203.259848] usb 3-1.2.2: New USB device found, idVendor=303a, idProduct=1001, bcdDevice= 1.02
[1214203.259867] usb 3-1.2.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[1214203.259874] usb 3-1.2.2: Product: USB JTAG/serial debug unit
[1214203.259880] usb 3-1.2.2: Manufacturer: Espressif
[1214203.259884] usb 3-1.2.2: SerialNumber: 60:55:F9:F7:4C:D4
[1214203.274098] cdc_acm 3-1.2.2:1.0: ttyACM0: USB ACM device
```

FelipeMdeO, 2024-08-13:
Hello @tmedicci ,

Yes, USB/JTAG port is connected, please look image below:

[image]

acassis, 2024-08-13:
@FelipeMdeO see the LIBUSB_ERROR_ACCESS, please try the same command with "sudo"

fdcavalcanti, 2024-08-13:
@FelipeMdeO Hi Felipe, what OS are you using? Did you check udev rules?

FelipeMdeO, 2024-08-13:
Hello @fdcavalcant.

I am using Ubuntu 20 .04 noble.

@acassis Thank you very much, running with sudo it worked.

closed 2024-08-13 by FelipeMdeO

tmedicci, 2024-08-14:
Hi @FelipeMdeO , you can set the udev rules too to avoid this kind of error: https://docs.espressif.com/projects/esp-idf/en/latest/esp32c6/api-guides/jtag-debugging/configure-builtin-jtag.html#configure-usb-drivers

tmedicci, 2024-08-14:
Thanks for reporting. We will update the documentation to make it clearer...