apache/nuttx #12907: [HELP] <title>prompts  "undefined referenc to `_impure_prt`"

superxyxiao, 2024-08-14:
### Description

I have to use nuttx version 8.2. Using libm in newlibc, the link prompts an "undefined referenc to `_impure_prt`" error when using the lagmma function. How can I solve this? 

### Verification

- [X] I have verified before submitting the report.

acassis, 2024-08-14:
@superxyxiao it seems something used by newlibc reentrancy, see: https://embdev.net/topic/what-is-_impure_ptr
lagmma is not part of NuttX libc yet

xiaoxiang781216, 2024-08-14:
mainline already fix this problem: https://github.com/apache/nuttx/blob/master/libs/libc/misc/lib_impure.c
8.2 is very very old release, I would suggest you upgrade to the new release.

superxyxiao, 2024-08-15:
Thank you very much.

When calling log2() in protected build mode, the User.map file does not have log2() symbols and cannot compile the elf file. But log2() works fine in app.bin, How can I solve this？

gcc version 7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907] (GNU Tools for Arm Embedded Processors 7-2018-q2-update)

xiaoxiang781216, 2024-08-15:
which libm library are use using?

superxyxiao, 2024-08-15:
Using libm from newlib

xiaoxiang781216, 2024-08-15:
do you add libm to the ld command line when you build user space bin?

closed 2024-09-28 by xiaoxiang781216