apache/nuttx #12916: [HELP] <title> Booting the OS on ARM A53 FVP

lzc1001, 2024-08-15:
### Description

I'm porting nuttx to FVP, and have a question during the porting. (This is my first time to port OS, hhh)
I need to set the related GIC(gic v3) registers in EL3 for boot successfully. So, does the nuttx need ARM TF-A before running the OS kernel? 
Thanks!
 

### Verification

- [X] I have verified before submitting the report.

xiaoxiang781216, 2024-08-15:
> ### Description
> I'm porting nuttx to FVP, and have a question during the porting. (This is my first time to port OS, hhh) I need to set the related GIC(gic v3) registers in EL3 for boot successfully. So, does the nuttx need ARM TF-A before running the OS kernel? Thanks!

no, you can create a pseudo interrupt frame and downgrade to EL1 inside NuttX if CPU start from EL3 or EL2. The common arrm64 boot code already has this code block:
https://github.com/apache/nuttx/blob/master/arch/arm64/src/common/arm64_boot.c#L105-L200

lzc1001, 2024-08-15:
Thanks for your answer! @xiaoxiang781216
 If i don't add code to set registers, the ternimal would not input command. I think the UART issue from the register configuration. So, is this a bug?

closed 2024-08-22 by lzc1001