apache/nuttx #13014: [BUG] is it typo in esp_wifi_adapter.c ? (ESP32C3)

metarutaiga, 2024-08-19:
### Description / Steps to reproduce the issue

https://github.com/apache/nuttx/blob/fb37391ea6b44ce2b3814b0b0fe6e09deba99b6b/arch/risc-v/src/esp32c3/esp_wifi_adapter.c#L5122

```
-          memcpy(wifi_cfg.sta.password, pdata, len);
+          memcpy(wifi_cfg.ap.password, pdata, len);
```

### On which OS does this issue occur?

[Mac]

### What is the version of your OS?

macOS 14

### NuttX Version

12.6.0

### Issue Architecture

[risc-v]

### Issue Area

[Networking]

### Verification

- [X] I have verified before submitting the report.

fdcavalcanti, 2024-08-19:
Hello @metarutaiga, I'll take a look at it.
Did you have issues using SoftAP mode?

fdcavalcanti, 2024-08-19:
Thanks for the info. Just tested here and fixed on #13023.

acassis, 2024-08-19:
Different topic @metarutaiga : your xxGraphic  project seems very interesting! Do you think it is possible to get it working on NuttX? Sometime ago I was thinking about porting TinyGL (https://bellard.org/TinyGL/) to NuttX

closed 2024-08-19 by pkarashchenko

metarutaiga, 2024-08-20:
> Different topic @metarutaiga : your xxGraphic project seems very interesting! Do you think it is possible to get it working on NuttX? Sometime ago I was thinking about porting TinyGL (https://bellard.org/TinyGL/) to NuttX

I looked the TinyGL, it implemented many GL functions but old school.
It's not compatible with xxGraphic.