apache/nuttx #13062: [HELP] How to print MMU related information on ARM FVP?

lzc1001, 2024-08-20:
### Description

I want to print virtual address and physical address when initialize MMU in arm64_mmu.c. So, I set CONFIG_MMU_DEBUG = 1 and set other configuration as following picture.
[image]
But the related mmu configuration information do not print. 
Thanks for everyone!

### Verification

- [X] I have verified before submitting the report.

acassis, 2024-08-20:
@lzc1001 look inside arch/arm64/src/common/arm64_mmu.c the virtual and physical addresses are no printed by default, you need to add a new "sinfo()" with that values you want to print

lzc1001, 2024-08-21:
@acassis Thanks for your answer!😀 So, I need to write a new "sinfo()" by myself? Because I found that this function was already written in the code, but it was wrapped in the macro definition.
[image]

acassis, 2024-08-21:
@lzc1001 hmm, I think I discovered the issue: there is not CONFIG_MMU_DEBUG.
Just run: $ git grep MMU_DEBUG

There need to exist an MMU_DEBUG symbol defined inside a Kconfig or you can just comment out these lines:

```
/* MMU debug option
 * #define CONFIG_MMU_ASSERT 1
 * #define CONFIG_MMU_DEBUG 1
 * #define CONFIG_MMU_DUMP_PTE 1
 */
```

acassis, 2024-08-22:
@lzc1001 if you get it working, please close this Issue

lzc1001, 2024-08-22:
@acassis i add "#define CONFIG_MMU_DEBUG 1" to arch/arm64/src/common/arm64_mmu.c, then run on FVP. But the related mmu information still do not print.😫

acassis, 2024-08-22:
@lupyuen any idea?

lupyuen, 2024-08-23:
Hi @lzc1001 you mentioned FVP, could you verify that this is the Build Config that we used?

```bash
$ tools/configure.sh fvp-armv8r:nsh
```

`fvp-armv8r:nsh` doesn't seem to support MMU. Maybe that's why the MMU Logging wasn't working?

```bash
$ grep MMU .config
# CONFIG_ARCH_HAVE_MMU is not set
```

I tried `qemu-armv8a:nsh`, but it doesn't support MMU either:

```bash
$ ./tools/configure.sh qemu-armv8a:nsh
$ grep MMU .config
CONFIG_ARCH_HAVE_MMU=y
# CONFIG_ARCH_USE_MMU is not set
```

lzc1001, 2024-08-23:
@lupyuen Thanks for your answer! I ported NuttX to A53 based on FVP in the last few days and ARM A53 have MMU. Then, NuttX can boot and input command to terminal, like "ls", "hello", etc. So, I think MMU is enabled for NuttX on FVP. But I do't make sure that the porting code have some potential bug, hhh😀.
I add code to arch/arm64/src/common/arm64_mmu.c and enable the related congfiguration by menuconfig. But the MMU infnormation still do not print. Maybe porting code have other issues. I will check my added code. The NuttX commit ID is 7a865b247969a9dbc06eddcaf58c59a14dd05d6d, tag: nuttx-12.6.0-RC1
[image]
[image]

lupyuen, 2024-08-23:
@lzc1001 Can you verify that the NuttX Config supports MMU?

```bash
$ grep MMU .config
CONFIG_ARCH_HAVE_MMU=y
# CONFIG_ARCH_USE_MMU is not set
```

Is it possible that the MMU code is disabled due to other reasons? Perhaps we can add an Error Message in arch/arm64/src/common/arm64_mmu.c:

```c
int arm64_mmu_init(bool is_primary_core)
{
#error MMU is enabled
...
```

lzc1001, 2024-08-23:
@lupyuen Yes! The config support MMU
[image]

lupyuen, 2024-08-23:
@lzc1001 I wonder if `sinfo` is failing to log correctly. What if you rename all `sinfo` to `_info`?

raiden00pl, 2024-08-23:
`arm64_mmu_init()` is called early in `arm64_chip_boot()` so it's possible that serial console is not yet initialized. `arm64_earlyserialinit()` is called after `arm64_mmu_init()`. You can add simple debug message (`_info("xxxx")`) at the top of `arm64_chip_boot()` and after `arm64_mmu_init()` to see if serial port works at this boot stage.

acassis, 2024-08-23:
@raiden00pl @lupyuen @lzc1001 maybe RAMLOG could solve this issue.

raiden00pl, 2024-08-23:
@acassis RAMLOG is registered also late in boot process (in `drivers_initialize()`). One option if serial is not available since boot and requires some kind of initialization is to separate `arm64_mmu_init()` from MMU debug dump functionality like it's done for ACPI in x86:
1. ACPI is parsed in early boot stage in `__nxstart()` as it's required for other initialization tasks
2. ACPI dump is done in `up_initialize()` when some low-level initialization is done (like `x86_64_serialinit()`)

lzc1001, 2024-08-23:
@acassis @raiden00pl @lupyuen Thank you very much! 
I added simple debug message (_info("111111111111111\n")). Then, the MMU information still don't print. but I added "_info("1111111111111111\n");" before and after up_initialize() in nuttx/sched/init/nx_start.c, the debug information after up_initialize() do print!! hhh, you are very niubility!😀✌
I tried to use ARM TF-A to boot NuttX and added the related macro in arch/arm64/src/common/arm64_mmu.c, the terminal print the related MMU information. But the exact cause of this problem is still unknown, I will continue to check my poring code!
Thank you very much again!

closed 2024-08-27 by lzc1001