apache/nuttx #13065: [HELP] Why does libm not have nextafter?

sastel, 2024-08-20:
### Description

I'm working on getting `gtest` executables to compile for kernel mode, so that I can run my tests on the target hardware. The EXPECT_NEAR tests in `gtest` rely on `nextafter()`. So my tests fail to link, because `libm` doesn't have a definition for `nextafter()`.

Does anyone know why this function was not included in `libm`? I can add it myself, but I'm wondering if there is a fundamental reason that I need to be aware of for why it isn't already present.

Using `libmcs` isn't an option for me, because RISC-V RV64 uses 128 bit long double, and `libmcs` only supports 64 bit `long double` in order to comply with IEEE-754. I tried `newlib`, but the `cmath` header fails to compile due to references to the undeclared `long double` versions of the math functions, which is the same symptom I get when I compile `libmcs` without `long double`.

### Verification

- [X] I have verified before submitting the report.

acassis, 2024-08-20:
Hi @sastel I think it is not included because nobody used it until now. Also ANSI C doesn't require this function: https://docs.rtems.org/releases/4.5.1-pre3/toolsdoc/newlib-1.8.2-docs/libm/libm00036.html

sastel, 2024-08-20:
Thanks @acassis .

closed 2024-08-20 by sastel

acassis, 2024-08-20:
@sastel if you submit an implementation, please link to this Issue here for future reference.