apache/nuttx #13067: [BUG] ESP32C6 SPI Read Issue

FelipeMdeO, 2024-08-20:
### Description / Steps to reproduce the issue

Hello All,

I am porting MFRC522 rfid card read to esp32 c6. During port test I am facing issue related SPI message read.

1 - Add breakpoint in mfrc522_getfwversion() function, - I did minor change in code, not necessary:
file: mfrc522.c
line 1160

[image]

2 - Run the code until it hits the breakpoint 
[image]

3 - In my case I start logic analyze and hit "next" in the gdb.
Looking syslog msgs and results in logic Analyze We can see that the rfid IC answer 0xB2, but spi drive can't get this value.

Look images below:
[image]
[image]

### On which OS does this issue occur?

[Linux]

### What is the version of your OS?

Ubuntu 24

### NuttX Version

master

### Issue Architecture

[risc-v]

### Issue Area

[Drivers]

### Verification

- [X] I have verified before submitting the report.

acassis, 2024-08-20:
@fdcavalcanti PTAL

FelipeMdeO, 2024-08-20:
Hello @fdcavalcanti 

I added a breakpoint in:
esp_spi.c:759

Reading:
`trans.rcv_buffer` and `trans.send_buffer ` it always have same value, maybe this behavior explain how the example is working.

Steps:
1 - add breakpoint in `mfrc522_getfwversion` function.
2 - add brakpoint in esp_spi.c `esp_spi_poll_send` function

Running code and get trans.* value, you will see:
[image]

Note: 0xee is expected value: 
( 0x80 | (MFRC522_VERSION_REG & 0x7e)

fdcavalcanti, 2024-08-20:
Hello @FelipeMdeO, let's figure this out.

We'll start making sure there's no hardware issues.

- Is it an Espressif dev-kit or a custom board?
- Which pins are you using with SPI?
- Any pull-ups on the circuit?
- Which SPI mode are you working with?

FelipeMdeO, 2024-08-20:
Hello @fdcavalcanti 

1 -
I am using esp32c6 dev kit m and rfid-rc522 rf card read:
[image]

2 - 
SPI2 CS - 15
SPI2 CLK - 6
SPI2 MOSI - 7
SPI2 MISO - 2

3 - No, direct connection
[image]

fdcavalcanti, 2024-08-21:
I see no issue on our side, regarding SPI on C6.
Please use NuttX main branch and try again. We need to be on common ground regarding the source code.
Let me know if anything different happens.

FelipeMdeO, 2024-08-21:
Hello @fdcavalcanti, 

How do you test in your side?
 I will try do the same thing on my side.

FelipeMdeO, 2024-08-21:
Hello @fdcavalcanti.

I tested using spi example and know slave device, in this situation it worked properly. So the issue is not in the drive.
I will investigate it better, sorry for "false" bug.

fdcavalcanti, 2024-08-21:
No problem! Good luck.

FelipeMdeO, 2024-08-23:
Hello @fdcavalcanti ,

I am reopen this issue, I will explain.

My issue happens when I tried use MFRC522 driver, this driver uses `SPI_SEND` function:
[image]

Using `SPI_SEND` function I cannot get read data properly.

The ESP SPI example uses `SPI_EXCHANGE` function. When we request once a byte, it send using exchange with nwords=1.

To do a proof, I changed the spi driver to  `esp_spi_send` call `esp_spi_exchange` instead of `esp_spi_poll_send`

[image]

In this way I was able to read MFRF522 FW Version.

@fdcavalcanti can you check SPI_SEND in your side, please ?

FelipeMdeO, 2024-08-27:
Hello,

I was clear in my last explanation, do you need more information?

fdcavalcanti, 2024-08-27:
Hi Felipe. I don't have this hardware to test.
But could you please try this inside `esp_spi_poll_send`:

```
------------------ arch/risc-v/src/common/espressif/esp_spi.c ------------------
index e84b22b336..4988f17aa3 100644
@@ -750,8 +750,9 @@ static uint32_t esp_spi_poll_send(struct esp_spi_priv_s *priv, uint32_t wd)
   trans.line_mode.cmd_lines = 0;
   priv->ctx->trans_config = trans;
 
-  spi_ll_set_mosi_bitlen(priv->ctx->hw, priv->nbits);
+  // spi_ll_set_mosi_bitlen(priv->ctx->hw, priv->nbits);
   spi_hal_prepare_data(priv->ctx, priv->dev_cfg, &trans);
+  spi_hal_setup_trans(priv->ctx, priv->dev_cfg, &trans);
   spi_hal_user_start(priv->ctx);
```
Make sure the rest of the file is the same as the main branch.

FelipeMdeO, 2024-08-27:
Hello @fdcavalcanti,

Sorry, not yet.
I put side by side comparation between SPI Exchange and SPI Send result.
Note that the SPI Send return strange data, like memory address, this happens with current master code too.
Maybe you can try use any spi slave device, if it uses SPI Send in the driver.

[image]

acassis, 2024-08-27:
> Hi Felipe. I don't have this hardware to test. But could you please try this inside `esp_spi_poll_send`:
>
> ```
> ------------------ arch/risc-v/src/common/espressif/esp_spi.c ------------------
> index [...]

@FelipeMdeO I think you can use the spi tool to so this test. Just do the ordinary loop test, but using send instead of exchange command.

FelipeMdeO, 2024-08-28:
Hello @acassis ,

Thank you for you idea, very nice!

So, first I return all to master and run esp32c6 devkit m spi config. I Tested spi exch cmd and it worked properly as expected.

I commeted SPI_EXCHANGED call to known value to test my approach:
[image]
[image]

I changed spi_transfer function to use SPI_SEND instead SPI_EXCHANGE. 
[image]
[image]
So, using SPI_SEND return, data printed is wrong.

Last step is change SPI_SEND as @fdcavalcanti  request:
[image]
[image]

Issue still happen.

FelipeMdeO, 2024-08-28:
Thank you for you suggestion @acassis, very nice.

I did changes in spi_transfer.c to use SPI_SEND instead SPI_EXCHANGE in the spi tool. 
Transfer is ok, so I don't know whats happen when I am try send/read data to MFRC device.

I will take a look better during next days.

[image]
[image]

acassis, 2024-08-28:
Actually the message was to Filipe Cavalcanti that doesn't have the board:
@fdcavalcanti  (since you don't have the board) I think you can use the spi tool to so this test. Just do the ordinary loop test, but using send instead of exchange command.

fdcavalcanti, 2024-10-02:
Recommend checking if #13780 changed anything

fdcavalcanti, 2025-07-23:
@FelipeMdeO were you able to make this work? Some weeks ago we had some changes on SPI driver and was able to replicate a similar issue to yours (solved now).

closed 2025-07-25 by fdcavalcanti