apache/nuttx #13161: TCP retransmition is not correct

ChenChuangForLinux, 2024-08-26:
### Description / Steps to reproduce the issue

tcp retransmission has an exponential regression mechanism. For example, if no ack is received for 1.5 seconds, the first retransmission will be sent. If no ack is received, the second retransmission will be generated after 3s. If an ack is received, the exponential retreat timer resets. But timing doesn't work that way. After receiving an ack for the first retransmission, no reset was performed. The tcp interval becomes larger when the subsequent transmission becomes retransmission.

### On which OS does this issue occur?

[Linux]

### What is the version of your OS?

ubuntu22.04

### NuttX Version

unkown

### Issue Architecture

[arm]

### Issue Area

[Networking]

### Verification

- [X] I have verified before submitting the report.

acassis, 2024-08-27:
@wengzhe please take a look

wengzhe, 2024-09-02:
@Meissi-jian Do you have any idea about this? I saw you were modifying retransmission these days https://github.com/apache/nuttx/pull/13120

Meissi-jian, 2024-12-10:
@ChenChuang-23 You can have a try with https://github.com/apache/nuttx/pull/15118 . And I think it will resolve you probrem.

casaroli, 2026-11-08:
Possibly fixed by #15118.