apache/nuttx #13227: [BUG] /drivers/spi/spi_slave_driver.c broken - not working in master but working in old commit

FelipeMdeO, 2024-08-28:
### Description / Steps to reproduce the issue

Hello all.

I cannot use spi slave in esp32 c6 dev kit. 
I follower instructions in PR https://github.com/apache/nuttx/pull/3787 and using the same code in the PR I was able to reproduce PR results.
So I started track issue, and spi slave broken here, look image below:
[image]

I believe spi example cannot work using approach added in this commit because you need open spi to configure spi, what not happen when you use spi dev tool.
I tried develop app to read using open and read and it cannot work too.

### On which OS does this issue occur?

[Linux]

### What is the version of your OS?

Ubuntu 24

### NuttX Version

master

### Issue Architecture

[all]

### Issue Area

[Drivers]

### Verification

- [X] I have verified before submitting the report.

acassis, 2024-08-29:
@FelipeMdeO in fact we cannot use SPI Tool for SPI Slave, only for SPI Master.

Did you get the same example working with ESP32 using that old commit from @gustavonihei right?

@fdcavalcanti @tmedicci please take a look

FelipeMdeO, 2024-08-29:
> @FelipeMdeO in fact we cannot use SPI Tool for SPI Slave, only for SPI Master.
> 
> Did you get the same example working with ESP32 using that old commit from @gustavonihei right?
> 
> @fdcavalcanti @tmedicci please take a look

Right, but we need discuss if it is an issue or a feature that we lost when power manager was added. It isn't esp32 specific.
Actually we cannot use SPI Tool to test spi slave because we are doing spi slave bind in open function, in the past spi slave bind happen in spi slave register.

fdcavalcanti, 2024-08-29:
Hi Felipe. Not sure I get the issue.
Are you trying use two C6 devkits, one as SPI slave and another as master?
Then, you try to exchange data between them?

FelipeMdeO, 2024-08-29:
> Hi Felipe. Not sure I get the issue. Are you trying use two C6 devkits, one as SPI slave and another as master? Then, you try to exchange data between them?

Exactly @fdcavalcanti, I done by @gustavonihei here: PR https://github.com/apache/nuttx/pull/3787

fdcavalcanti, 2024-08-29:
Ok, but you said it is not specific for ESP32. So I assume the same issue happens on other boards, right?
And also that using the code from that PR you could get it to work.

So the issue is on SPI driver then? Are you proposing a fix?

FelipeMdeO, 2024-08-29:
@fdcavalcanti , yes, the behavior will be the same in all boards because changes were made in `spi_slave_driver.c ` file.

**So the issue is on SPI driver then?**
This is an issue only if we want to use spi tool to test, if we consider that we cannot use spi tool to test like PR https://github.com/apache/nuttx/pull/3787, we have no issue.
 
 **Are you proposing a fix?**
What do you think @acassis ? Maybe we can think in create spi slave example using 2 devices, in this way we don't need talk about change `spi_slave_driver.c `.

acassis, 2024-08-29:
@FelipeMdeO I think it should be nice to have a SPI example, but it not necessarily means we need to have the Master and Slave working with SPI Tool. Maybe we can use the SPI Tool with the master and the create a SPI Slave to receive the data.

What do you think?

FelipeMdeO, 2024-08-29:
Let’s do it @acassis. 
I will do it soon.

closed 2024-08-29 by FelipeMdeO