apache/nuttx #13230: [HELP] ESP32C3 sleep after software reset

metarutaiga, 2024-08-29:
### Description

I tested LUATOS ESP32C3 and ESP32-C3-mini

I tried these
1. boardctl(BOARDIOC_RESET, 0);
2. esp_restart();
3. esp_rom_software_reset_cpu(0);

But it sleep always.

I tried esp_restart from esp-idf project, it works normally.
So how to know where is the bug? NuttX or MCUboot ?

### Verification

- [X] I have verified before submitting the report.

tmedicci, 2024-08-29:
Hi,

Please check `esp32c3-generic:sta_softap` defconfig. It enables the `reboot` command (just type `reboot`). It uses the [`boardctl(BOARDIOC_RESET, 0)`](https://github.com/apache/nuttx-apps/blob/master/nshlib/nsh_syscmds.c#L479) interface for that.

If it still does not work, please enable the following configs in `menuconfig`:
```
CONFIG_SYSTEM_NXDIAG=y
CONFIG_SYSTEM_NXDIAG_COMP_FLAGS=y
CONFIG_SYSTEM_NXDIAG_CONF=y
CONFIG_SYSTEM_NXDIAG_ESPRESSIF=y
CONFIG_SYSTEM_NXDIAG_HOST_PATH=y
```

And run in NuttX `nxdiag --all` (sharing your results here).

Also, please double-check which hardware exactly you are using (Is it a module? a board? Which Espressif module/chip is being used?)

metarutaiga, 2024-08-29:
Thank you
I found the problem is CONFIG_ELF now.
I will try where the problem is.

metarutaiga, 2024-08-29:
Skip this, reboot works.
```
#ifdef CONFIG_ELF
  ret = elf_initialize();
  if (ret < 0)
    {
      berr("ERROR: elf_initialize failed: %d\n", ret);
    }
#endif
```
But skip here, it will sleep
```
int elf_initialize(void)
{
  int ret;

  /* Register ourselves as a binfmt loader */

  binfo("Registering ELF\n");

  ret = register_binfmt(&g_elfbinfmt);
  if (ret != 0)
    {
      berr("Failed to register binfmt: %d\n", ret);
    }

  return ret;
}
```
Very strange...

metarutaiga, 2024-08-30:
I close this issue, it's very strange, and now it works normally...

metarutaiga, 2024-10-08:
I guest the problem is mcuboot
the ESP32-C3 generic is supported mcuboot only.
And if firmware is over than 1MB
It stuck at mcuboot (after the esp32 rom bootloader)

tmedicci, 2024-10-16:
Hi @metarutaiga , 

esp-idf is not required to build MCUboot. Can you share exactly the steps you are doing? Also, can you please post your `defconfig` (run ` make savedefconfig` and a new `defconfig` will be generated)? (If not possible, please share a reduced version of it with the same error)

metarutaiga, 2024-10-16:
I used the mcuboot repository to build it directly.
Any MCUboot version is not affect the result.

I known it's hang not sleep now.
I am sure that if binary size is over than 1MB, it hangs when reboot.

tmedicci, 2024-10-16:
> I used the mcuboot repository to build it directly. Any MCUboot version is not affect the result.
> 
> I known it's hang not sleep now. I am sure that if binary size is over than 1MB, it hangs when reboot.

Right, can you provide the steps to reproduce and a failing branch of nuttx and nuttx-apps, please?

metarutaiga, 2024-10-18:
https://github.com/NyankoLab/esp32c3-nuttx

if it enabled wget and uiP web Client, it would be over 1MB.

tmedicci, 2024-10-21:
Hi @metarutaiga , I couldn't reproduce your issue. I need clearer and more detailed instructions.

I tried:

1. `esp32c6-devkitc:sta_softap`, enabling `wget`, `uiP web client` and `mbedTLS` (with example applications). Firmware is over 1MB and the device boots successfully and `reboot` commands reboots the device successfully.
2. `esp32c6-devkitc:sta_softap`, enabling `wget`, `uiP web client` and `mbedTLS` (with example applications) and MCUboot (`CONFIG_ESPRESSIF_BOOTLOADER_MCUBOOT=y`). I adjusted the slots accordingly (default slot size is 1MB): `CONFIG_ESPRESSIF_OTA_SCRATCH_OFFSET=0x410000`,  `CONFIG_ESPRESSIF_OTA_SECONDARY_SLOT_OFFSET=0x210000` and `CONFIG_ESPRESSIF_OTA_SLOT_SIZE=0x200000` to 2MB. Same result: device boots as expected and reboots successfully.

metarutaiga, 2024-10-22:
Hmmm, it's almost the same instructions.
I give up to find the problem.