apache/nuttx #13238: [BUG] ESP32C6 Slave Driver Issue

FelipeMdeO, 2024-08-30:
### Description / Steps to reproduce the issue

Hello All,

I am trying to use 2 esp32c6 devkit m, one as master and one as slave.
Slave spi cannot work. I tried solve it by myself but I can't. Please @tiagojbalmeida, @tmedicci , can you take a look?

I did custom app with open and read cmds, master doesn't receive data. I am using master branch. SPI 2, without DMA.
Please, look image below:
[image]

`spi_slave_hal_get_rcv_bitlen` should be called after `spi_slave_hal_storage_result`, we aren't doing it so `transfer_size` always will be 0.
 
 I changed `spi_slave_hal_get_rcv_bitlen` position,  but I faced other issues. I can give more details when you ask for.
Just to FYI, to advance in my solution I need change esp_hal (I know, it is not a good idea, I did it only for testing), I opened a thread in esp-idf forum to try understand better hal.
https://github.com/espressif/esp-idf/issues/14462

### On which OS does this issue occur?

[Linux]

### What is the version of your OS?

Ubuntu 24

### NuttX Version

master

### Issue Architecture

[risc-v]

### Issue Area

[Drivers]

### Verification

- [X] I have verified before submitting the report.

fdcavalcanti, 2024-08-30:
Hi Felipe. Can you share the app you've written to test it?

FelipeMdeO, 2024-08-30:
Sure @fdcavalcanti.

```
`int main(int argc, FAR char *argv[])
{

  int fd;
  int ret;
  uint8_t buffer[128];

  while(1)
  {
    fd = open("/dev/spislv2", O_RDONLY | O_WRONLY );
    if (fd < 0)
    {
        printf("Failed to open %s\n", "/dev/spislv2");
        return -1;
    }
    printf(" Trying to Read: \n");
    
    ret = read(fd, buffer, 30);
    for (int i = 0 ; i < ret; i++)
      {
        printf("%02X ", buffer[i]);
      }
    printf("\n");
    usleep(1000000);
    close(fd);
    usleep(2000000);
  }

  return 0;
}`
```

You can use my .config also:
[config.zip](https://github.com/user-attachments/files/16817268/config.zip)

fdcavalcanti, 2024-09-05:
@FelipeMdeO please do this on the slave device:
- Remove the `open` and `close` commands from the while loop. Just open once and let it print stuff
- Enable `DEBUG_FEATURES`, `DEBUG_ASSERTIONS_EXPRESSION` and `DEBUG_ASSERTIONS_FILENAME` on Debug Options menu
- Enable SPI Slave support on `SPI Driver Support` menu and also on `SPI Configuration`

Run that program and check if the system fails. If it does, send here the log.

FelipeMdeO, 2024-09-05:
Hello @fdcavalcanti, sorry for the delay.

I am with same behavior. I will put details here:
Master >
[image]

Slave >
[image]

```
nsh> hello
spi_slave_open: filep: 0x4080a980
spislave_bind: ctrlr=0x40804bf4 dev=0x40809b28 mode=0 nbits=8
spi_slave_select: sdev: 0x40809b28 CS: free
spi_slave_cmddata: sdev: 0x40809b28 CMD: command
spislave_initialize: ctrlr=0x40804bf4
spi_slave_select: sdev: 0x40809b28 CS: select
spislave_prepare_next_tx: TX buffer empty! Disabling TX for next transaction
spi_slave_select: sdev: 0x40809b28 CS: free
spi_slave_read: filep=0x4080a22c buffer=0x4080a9e0 buflen=30
 Trying to Read: 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
 Trying to Read: 
spi_slave_read: filep=0x4080a22c buffer=0x4080a9e0 buflen=30
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
spi_slave_select: sdev: 0x40809b28 CS: select
spislave_prepare_next_tx: TX buffer empty! Disabling TX for next transaction
spi_slave_select: sdev: 0x40809b28 CS: free
spi_slave_select: sdev: 0x40809b28 CS: select
spislave_prepare_next_tx: TX buffer empty! Disabling TX for next transaction
spi_slave_select: sdev: 0x40809b28 CS: free
spi_slave_select: sdev: 0x40809b28 CS: select
spislave_prepare_next_tx: TX buffer empty! Disabling TX for next transaction
spi_slave_select: sdev: 0x40809b28 CS: free
spi_slave_select: sdev: 0x40809b28 CS: select
spislave_prepare_next_tx: TX buffer empty! Disabling TX for next transaction
spi_slave_select: sdev: 0x40809b28 CS: free
 Trying to Read: 
spi_slave_read: filep=0x4080a22c buffer=0x4080a9e0 buflen=30
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
spi_slave_select: sdev: 0x40809b28 CS: select
spislave_prepare_next_tx: TX buffer empty! Disabling TX for next transaction
spi_slave_select: sdev: 0x40809b28 CS: free
spi_slave_select: sdev: 0x40809b28 CS: select
spislave_prepare_next_tx: TX buffer empty! Disabling TX for next transaction
spi_slave_select: sdev: 0x40809b28 CS: free
spi_slave_select: sdev: 0x40809b28 CS: select
spislave_prepare_next_tx: TX buffer empty! Disabling TX for next transaction
spi_slave_select: sdev: 0x40809b28 CS: free
spi_slave_select: sdev: 0x40809b28 CS: select
spislave_prepare_next_tx: TX buffer empty! Disabling TX for next transaction
spi_slave_select: sdev: 0x40809b28 CS: free
 Trying to Read: 
spi_slave_read: filep=0x4080a22c buffer=0x4080a9e0 buflen=30
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
```

Are you able to do it work in your side?

fdcavalcanti, 2024-09-06:
Ok, thanks.
I'll look into it.

fdcavalcanti, 2024-09-10:
@FelipeMdeO I've reviewed it and noticed two issues:
- There's an issue with the HAL item you mentioned, I also couldn't retrieve data from it.
- There's also an issue on the NuttX SPI Slave driver that was changed in commit dcb8188f07f, preventing the while loop to enter. I forced some values on the rx_buffer and could make it work. 

I can't dig deeper at the moment, but will get back to it when possible.

FelipeMdeO, 2024-09-19:
Hello @fdcavalcanti.

I am trying solve spi slave communication and I made some progress. However I am stucked in other issue, maybe you can help me.

The CS pin was configure to generate IRQ in Rising edge:

[image]

But I am getting IRQ in both edges. I solved this locally but I believe the final solution should be changing the HAL

I changed `spislave_cs_interrupt` to get more prints:

[image]

Look whats happen:

[image]

@tmedicci  Can you help us looking better about this IRQ?

tmedicci, 2024-09-19:
> Hello @fdcavalcanti.
>
> I am trying solve spi slave communication and I made some progress. However I am stucked in other issue, maybe you can help me.
>
> The CS pin was configure to generate IRQ in [...]

Hi @FelipeMdeO , I didn't get what is the local solution (and what would be the proposed HAL solution). Can you explaining how you made it work locally?

FelipeMdeO, 2024-09-19:
Hello @tmedicci,

I am fixing driver step by step in my local machine, it isn't finished and need be improved. 
If you want, I can share with you my repo.

closed 2024-10-21 by xiaoxiang781216