apache/nuttx #13303: [BUG] ESP32C6 WIFI stop work after enable spi 2 in slave mode

FelipeMdeO, 2024-09-05:
### Description / Steps to reproduce the issue

Hello All,

I am using esp32c6-devkitc. After enable SPI 2 in slave mode, WIFI feature stopped work.

Steps to reproduce:
1: Run esp32c6-devkitc wifi defconfig
2: Build and Flash, run in nuttx: `wapi scan wlan0` - I will get wifi available list
3: Enable SPI2 peripheral and configure it as slave.
4: Run step 2 again, now no one wifi will be available

This is my nxdiag:
```
nsh> nxdiag --all
Nxdiag Report:

NuttX RTOS info:
        Hostname: 
        Release: 12.6.0-RC1
        Build: da6ddea8d4 Sep  5 2024 13:03:53
        Arch: risc-v
        Defconfig: esp32c6-devkitc:wifi-dirty

Host system OS:
        Ubuntu 24.04 LTS Linux felipe-moura 6.8.0-41-generic #41-Ubuntu SMP PREEMPT_DYNAMIC Fri Aug  2 20:41:06 UTC 2024 x86_64 x86_64

Espressif specific information:

Bootloader version:
        esp32: Bootloader image not found
        esp32s2: Bootloader image not found
        esp32s3: Bootloader image not found
        esp32c2: Bootloader image not found
        esp32c3: Bootloader image not found
        esp32c6: Bootloader image not found
        esp32h2: Bootloader image not found

Toolchain version:
        clang: Not found
        gcc: gcc (Ubuntu 13.2.0-23ubuntu4) 13.2.0
        xtensa-esp32-elf-gcc: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r2) 8.2.0
        xtensa-esp32s2-elf-gcc: Not found
        xtensa-esp32s3-elf-gcc: Not found
        riscv32-esp-elf-gcc: Not found
        riscv64-unknown-elf-gcc: Not found

Esptool version: 4.8.dev4

HAL version: sync/release_v5.1.c-nuttx-20230814-1568-g51770f0c97
```

And my defconfig:
```
#
# This file is autogenerated: PLEASE DO NOT EDIT IT.
#
# You can use "make menuconfig" to make any modifications to the installed .config file.
# You can then do "make savedefconfig" to generate a new defconfig file that includes your
# modifications.
#
# CONFIG_NDEBUG is not set
# CONFIG_NSH_ARGCAT is not set
# CONFIG_NSH_CMDOPT_HEXDUMP is not set
CONFIG_ALLOW_BSD_COMPONENTS=y
CONFIG_ARCH="risc-v"
CONFIG_ARCH_BOARD="esp32c6-devkitc"
CONFIG_ARCH_BOARD_COMMON=y
CONFIG_ARCH_BOARD_ESP32C6_DEVKITC=y
CONFIG_ARCH_CHIP="esp32c6"
CONFIG_ARCH_CHIP_ESP32C6=y
CONFIG_ARCH_CHIP_ESP32C6WROOM1=y
CONFIG_ARCH_INTERRUPTSTACK=2048
CONFIG_ARCH_RISCV=y
CONFIG_ARCH_STACKDUMP=y
CONFIG_BOARD_LOOPSPERMSEC=15000
CONFIG_BUILTIN=y
CONFIG_DEV_ZERO=y
CONFIG_DRIVERS_IEEE80211=y
CONFIG_DRIVERS_WIRELESS=y
CONFIG_ESPRESSIF_ESP32C6=y
CONFIG_ESPRESSIF_SPI2=y
CONFIG_ESPRESSIF_SPI2_SLAVE=y
CONFIG_ESPRESSIF_SPIFLASH=y
CONFIG_ESPRESSIF_SPIFLASH_SPIFFS=y
CONFIG_ESPRESSIF_WIFI=y
CONFIG_EXAMPLES_RANDOM=y
CONFIG_FS_PROCFS=y
CONFIG_IDLETHREAD_STACKSIZE=2048
CONFIG_INIT_ENTRYPOINT="nsh_main"
CONFIG_INIT_STACKSIZE=8192
CONFIG_INTELHEX_BINARY=y
CONFIG_IOB_THROTTLE=24
CONFIG_LIBC_PERROR_STDOUT=y
CONFIG_LIBC_STRERROR=y
CONFIG_NETDB_DNSCLIENT=y
CONFIG_NETDEV_LATEINIT=y
CONFIG_NETDEV_PHY_IOCTL=y
CONFIG_NETDEV_WIRELESS_IOCTL=y
CONFIG_NETDEV_WORK_THREAD=y
CONFIG_NETUTILS_CJSON=y
CONFIG_NET_BROADCAST=y
CONFIG_NET_ICMP_SOCKET=y
CONFIG_NET_TCP=y
CONFIG_NET_TCP_DELAYED_ACK=y
CONFIG_NET_TCP_WRITE_BUFFERS=y
CONFIG_NET_UDP=y
CONFIG_NFILE_DESCRIPTORS_PER_BLOCK=6
CONFIG_NSH_ARCHINIT=y
CONFIG_NSH_BUILTIN_APPS=y
CONFIG_NSH_FILEIOSIZE=512
CONFIG_NSH_READLINE=y
CONFIG_NSH_STRERROR=y
CONFIG_PREALLOC_TIMERS=0
CONFIG_PTHREAD_MUTEX_TYPES=y
CONFIG_RR_INTERVAL=200
CONFIG_SCHED_BACKTRACE=y
CONFIG_SCHED_LPWORK=y
CONFIG_SCHED_WAITPID=y
CONFIG_SIG_DEFAULT=y
CONFIG_SPI_SLAVE=y
CONFIG_SPI_SLAVE_DRIVER=y
CONFIG_START_DAY=29
CONFIG_START_MONTH=11
CONFIG_START_YEAR=2019
CONFIG_SYSTEM_DHCPC_RENEW=y
CONFIG_SYSTEM_DUMPSTACK=y
CONFIG_SYSTEM_NSH=y
CONFIG_SYSTEM_NXDIAG=y
CONFIG_SYSTEM_NXDIAG_ESPRESSIF=y
CONFIG_SYSTEM_PING=y
CONFIG_TESTING_GETPRIME=y
CONFIG_TESTING_OSTEST=y
CONFIG_TLS_TASK_NELEM=4
CONFIG_UART0_SERIAL_CONSOLE=y
CONFIG_WIRELESS=y
CONFIG_WIRELESS_WAPI=y
CONFIG_WIRELESS_WAPI_CMDTOOL=y
CONFIG_WIRELESS_WAPI_INITCONF=y

```
@tmedicci  can you check this behavior in your side, please?

### On which OS does this issue occur?

[Linux]

### What is the version of your OS?

Ubuntu 24.0

### NuttX Version

master

### Issue Architecture

[risc-v]

### Issue Area

[Drivers]

### Verification

- [X] I have verified before submitting the report.

FelipeMdeO, 2024-09-05:
Hello All,

I changed the title because during investigation I was able to see that the issue is related enable GPIO interrupt when we are using wifi. 

So, another way to reproduce is:

1: ./tools/configure.sh esp32c6-devkitc:wifi
 2: make menuconfig
 3: System Type -> Peripheral Support -> enable GPIO pin interrupts.
 4: Test wapi scan wlan 0
 it won't work.

acassis, 2024-09-05:
@tmedicci did you see this issue before?

acassis, 2024-09-05:
> @tmedicci did you see this issue before?

acassis, 2024-09-05:
@FelipeMdeO @tmedicci I confirmed the issue:

```
nsh> ls /dev
/dev:
 console
 espflash
 gpio0
 gpio1
 gpio2
 null
 random
 rtc0
 ttyS0
 zero
nsh> wapi scan wlan0
nsh>
```

acassis, 2024-09-06:
@fdcavalcanti could you please help to investigate this issue?

tmedicci, 2024-09-06:
Hi @FelipeMdeO and @acassis , we will investigate it internally soon.

tmedicci, 2024-09-06:
Hi @FelipeMdeO,

Can you please test using another version of the HAL, just set the following env variables before running the `make` command. For instance:

```
ESP_HAL_3RDPARTY_VERSION=eff2dab80ce352018159d7fe67c437e505e12f0a ESP_HAL_3RDPARTY_URL="git@github.com:tmedicci/esp-hal-3rdparty.git" make flash ESPTOOL_PORT=/dev/ttyUSB0
```

Just to let you know: I just reserved some CPU interrupts that are specific for some peripherals and this probably is the reason for this issue. I was able to make it work with this new HAL. If it works for you too, I'll create the PR. Please let me know.

FelipeMdeO, 2024-09-07:
> Hi @FelipeMdeO,
>
> Can you please test using another version of the HAL, just set the following env variables before running the `make` command. For instance:
>
> ``` [...]

Let me test it properly and I update you as soon as possible.

FelipeMdeO, 2024-09-08:
Hello @tmedicci, 

Thank you for fix this behavior. I did some tests and all is good now. Let's open a PR.

tmedicci, 2024-09-09:
> Hello @tmedicci,
> 
> Thank you for fix this behavior. I did some tests and all is good now. Let's open a PR.

Alright, I just need to wait for the internal tests to pass and I will open it soon. Thanks for testing.

closed 2024-09-11 by xiaoxiang781216