apache/nuttx #13308: [BUG] esp32 "make bootloader" doesn't work

yamt, 2024-09-06:
### Description / Steps to reproduce the issue

i tried to build the patched bootloader for esp32-devkitc:knsh
it ended up with:
```
-- Building ESP-IDF components for target esp32
CMake Error at /Users/yamamoto/git/nuttx/nuttx/arch/xtensa/src/esp32/bootloader/esp-nuttx-bootloader/esp-idf/tools/cmake/build.cmake:542 (message):
  /Users/yamamoto/git/nuttx/nuttx/venv/bin/python: Error while finding module
  specification for 'idf_component_manager.prepare_components'
  (ModuleNotFoundError: No module named 'idf_component_manager')

Call Stack (most recent call first):
  /Users/yamamoto/git/nuttx/nuttx/arch/xtensa/src/esp32/bootloader/esp-nuttx-bootloader/esp-idf/tools/cmake/project.cmake:453 (idf_build_process)
  CMakeLists.txt:58 (project)

-- Configuring incomplete, errors occurred!
make[1]: *** [bootloader] Error 1
make: *** [bootloader] Error 2
```

### On which OS does this issue occur?

[Mac]

### What is the version of your OS?

12.7.6

### NuttX Version

master

### Issue Architecture

[xtensa]

### Issue Area

[Build System]

### Verification

- [X] I have verified before submitting the report.

tmedicci, 2024-09-06:
Hi @yamt ,

The kernel versions of the ESP32 are based on the legacy bootloader (a.k.a IDF bootloader), so you need to have the tools ESP-IDF requires to build it properly. That being said, the easiest way to enable ESP-IDF's tooling is exporting the associated environment var, just like described [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/linux-macos-setup.html#step-4-set-up-the-environment-variables).

yamt, 2024-09-08:
> Hi @yamt ,
>
> The kernel versions of the ESP32 are based on the legacy bootloader (a.k.a IDF bootloader), so you need to have the tools ESP-IDF requires to build it properly. That being said, the [...]

i was using export.sh of esp-idf.
which version of export.sh is expected to work?

tmedicci, 2024-09-09:
> i was using export.sh of esp-idf. which version of export.sh is expected to work?

Any version of the `release/v5.1` branch is expected to work. In particular, we have been using commit `1484f2d4f677e3fa41a310bf0952a3a76f69fb76`  of ESP-IDF. Don't forget to install the tools used by this version. We test this configuration in our internal CI, so we know that nothing broke about it recently.

tmedicci, 2024-09-09:
> Any version of the `release/v5.1` branch is expected to work. In particular, we have been using commit `1484f2d4f677e3fa41a310bf0952a3a76f69fb76` of ESP-IDF. Don't forget to install the tools used [...]

Hi @yamt , as I've said previously, we can build it without issues on our internal CI, but I have found that the complete firmware is not booting as expected (HW testing was disabled). I tested it with https://github.com/apache/nuttx/pull/13311 but it still halts during the boot-up.

tmedicci, 2024-09-16:
Yes, it was fixed!

closed 2024-09-16 by tmedicci