apache/nuttx #13370: [FEATURE] Support more perf events in Nuttx.

yorange1, 2024-09-11:
### Is your feature request related to a problem? Please describe.

Currently, Nuttx provides a basic perf support interface, as shown below:
``` c
/****************************************************************************
 * Name: up_perf_*
 *
 * Description:
 *   The first interface simply provides the current time value in unknown
 *   units.  NOTE:  This function may be called early before the timer has
 *   been initialized.  In that event, the function should just return a
 *   start time of zero.
 *
 *   Nothing is assumed about the units of this time value.  The following
 *   are assumed, however: (1) The time is an unsigned integer value, (2)
 *   the time is monotonically increasing, and (3) the elapsed time (also
 *   in unknown units) can be obtained by subtracting a start time from
 *   the current time.
 *
 *   The second interface simple converts an elapsed time into well known
 *   units.
 *
 ****************************************************************************/

void up_perf_init(FAR void *arg);
unsigned long up_perf_gettime(void);
unsigned long up_perf_getfreq(void);
void up_perf_convert(unsigned long elapsed, FAR struct timespec *ts);
```
It seems that only time-related events are monitored. Can we support more perf events in Nuttx, just like in Linux, so that we can monitor more hardware performance, such as cache or bpu performance?

### Describe the solution you'd like

More perf event support is added in Nuttx, providing an interface or framework so that platforms with PMU can be adapted according to the interface. 
It would be better to port perf to Nuttx app, so that we can use perf tool to monitor the hardware performance when running the application just like on Linux.

### Verification

- [X] I have verified before submitting the report.

xiaoxiang781216, 2024-09-11:
We are building coresight and perf framework, coresight is here https://github.com/apache/nuttx/pull/11605, perf need time to prepare upstream.

yorange1, 2024-09-18:
> We are building coresight and perf framework, coresight is here #11605, perf need time to prepare upstream.

Thanks for your reply, when will the perf framework be pushed upstream, or is there an open development branch? I am working on the PMU of the RISCV platform, and I want to align it with the perf framework as much as possible.

xiaoxiang781216, 2024-09-18:
@zyfeier please take a look.