apache/nuttx #13497: [BUG] Issue and regression with libcxx

g2gps, 2024-09-17:
### Description / Steps to reproduce the issue

I've identified a couple of issues when using LLVM libcxx and userspace applications. This is reproducible on `rv-virt:knsh` with the addition of:

```
CONFIG_HAVE_CXX=y
CONFIG_EXAMPLES_HELLOXX=m
CONFIG_HAVE_CXX=y
CONFIG_LIBCXX=y
CONFIG_TLS_NELEM=4
```

The first issue is that the static constructor isn't called. For example, when running the `helloxx` application:

```
nsh> helloxx
[68719476736000.001000] CHelloWorld: Constructor: mSecret=42
helloxx_main: Saying hello from the dynamically constructed instance
[77309411328000.001000] HelloWorld: HelloWorld: mSecret=42
CHelloWorld::HelloWorld: Hello, World!!
[77309411328000.001000] CHelloWorld: Constructor: mSecret=42
helloxx_main: Saying hello from the instance constructed on the stack
[77309411328000.001000] HelloWorld: HelloWorld: mSecret=42
CHelloWorld::HelloWorld: Hello, World!!
helloxx_main: Saying hello from the statically constructed instance
[81604378624000.001000] HelloWorld: HelloWorld: mSecret=0
CHelloWorld::HelloWorld: CONSTRUCTION FAILED!
[81604378624000.001000] ~CHelloWorld: Destructor
[85899345920000.001000] ~CHelloWorld: Destructor
```

However, if I change the cxx standard version to `CONFIG_CXX_STANDARD="c++2b"`, the static construct is called:

```
nsh> helloxx
[914828034048000.001000] CHelloWorld: Constructor: mSecret=42
[923417968640000.001000] CHelloWorld: Constructor: mSecret=42
helloxx_main: Saying hello from the dynamically constructed instance
[923417968640000.001000] HelloWorld: HelloWorld: mSecret=42
CHelloWorld::HelloWorld: Hello, World!!
[927712935936000.001000] CHelloWorld: Constructor: mSecret=42
helloxx_main: Saying hello from the instance constructed on the stack
[927712935936000.001000] HelloWorld: HelloWorld: mSecret=42
CHelloWorld::HelloWorld: Hello, World!!
helloxx_main: Saying hello from the statically constructed instance
[932007903232000.001000] HelloWorld: HelloWorld: mSecret=42
CHelloWorld::HelloWorld: Hello, World!!
[932007903232000.001000] ~CHelloWorld: Destructor
[936302870528000.001000] ~CHelloWorld: Destructor
```

The second issue seems to be a regression with #13110, in which exceptions are always enabled for libcxx, after this change, running any c++ application results in an IPC:

```
nsh> helloxx
[    2.271000] riscv_exception: EXCEPTION: Instruction page fault. MCAUSE: 0000000c, EPC: 018c0000, MTVAL: 018c0000
[    2.271000] riscv_exception: Segmentation fault in PID 3: helloxx
```

This is only an issue when static constructors are called, (so enabling c++2b in this example).  Reverting #13110 seems to fix this behaviour.

Running addr2line on the faulting instruction yields to results. My assumption is that there is an additional static constructor called for exception support, and this is what is causing the issue. However, I don't know enough about these layers to be sure.

Enabling `CONFIG_CXX_EXCEPTION` doesn't resolve the issue.

Possibly a reasonable solution to the second issue would be to revert the change which removed the `CONFIG_CXX_EXCEPTION` change from the libcxx build and let targets which need the library build with exception support add this option in the defconfigs?

### On which OS does this issue occur?

[Linux]

### What is the version of your OS?

Ubuntu 22.04 (Docker)

### NuttX Version

NuttX 12.6.0-RC1 f5b50be05e Sep 17 2024 05:07:45 risc-v rv-virt

### Issue Architecture

[risc-v]

### Issue Area

[OS Components]

### Verification

- [X] I have verified before submitting the report.

xiaoxiang781216, 2024-09-17:
@g2gps only kernel mode has this problem?

g2gps, 2024-09-19:
> @g2gps only kernel mode has this problem?

Yes, it seems to be just kernel mode. It works with ` rv-virt:libcxx` using `c++2b` and `gnu++20`.

xiaoxiang781216, 2024-10-09:
@xiaoxiang781216 could you try this patch: https://github.com/apache/nuttx/pull/13721 which restore c++ exception change.

g2gps, 2024-10-30:
@xiaoxiang781216 Apologies I was out of the NuttX space. 

I've update our local mirror to include #13721 and can confirm that it resolves this issue.

closed 2024-10-30 by g2gps