apache/nuttx #13552: [Feedback] PR Review Bot

lupyuen, 2024-09-20:
### Description

Please provide your feedback for the [NuttX PR Review Bot](https://github.com/apache/nuttx/pull/13494#issuecomment-2357371589). Thanks!

### Verification

- [X] I have verified before submitting the report.

lupyuen, 2024-09-20:
Earlier Discussion: https://github.com/apache/nuttx/pull/13494#issuecomment-2357371589

raiden00pl, 2024-09-20:
My comment on this message from bot: 

> Squash The Commits: This PR contains 2 Commits. Please Squash the Multiple Commits into a Single Commit.

A suggestion like this shouldn't be based solely on the number of commits in PR.
What should be squashed and what shouldn't require a greater understanding of PR content.
Having multiple commits in one PR isn't always bad. Sometimes it's even necessary.

lupyuen, 2024-09-20:
> A suggestion like this shouldn't be based solely on the number of commits in PR.
> What should be squashed and what shouldn't require a greater understanding of PR content.

Sorry @cederom and @acassis: What are your thoughts about this? Shall I remove the "Squash Commit" message, since our bot can't analyse the PR Content? Thanks!

raiden00pl, 2024-09-20:
maybe we can just change "Please squash commits" to "consider squashing commits if that makes sense for this PR" (or something similar)

lupyuen, 2024-09-20:
OK I removed the check for Squash Commits, I don't we're ready to handle it now :-)

acassis, 2024-09-20:
Thank you @lupyuen I think it is better not to include it to avoid inducing contributors to wrong direction.

It could be something that reviewers could do better

nuttxpr, 2024-09-23:
> Yes we need increase PR and git commit descriptions quality (both need to be self explanatory).. bot was created to help.. you can .. wait was there this nice copy-paste section where all sections were filled in already before

@cederom Yeah that's the problem with LLMs, we can't force it to say the same thing all the time :-) It seems to have a mind of its own / multiple personas?

closed 2024-10-04 by lupyuen

resyfer, 2024-10-22:
A suggestion to add a way to trigger the bot again. For example...if it gave some feedback, and I improved it, I don't have any way of checking it again without creating another PR (which would mean wasted GH checks as PR text and build runs are often unrelated).