apache/nuttx #13562: [BUG] <compile warning due to __CONCAT redefinition>

TimJTi, 2024-09-20:
### Description / Steps to reproduce the issue

Perhaps due to me upgrading my arm gcc cross compiler recently, I'm seeing a compiler warning because of the definition of __CONCAT - in my case this is in _sama5d2x_memorymap.h but it occurs in 12 other files too

The warning is:
```
arm-gnu-toolchain-13.2.1-arm-none-eabi/arm-none-eabi/include/sys/cdefs.h:166: note: this is the location of the previous definition
  166 | #define __CONCAT(x,y)   __CONCAT1(x,y)
```
I can fix this as follows but have no idea if a better, more global fix is needed and, if so, where/how?
```
#ifndef __CONCAT
#  define __CONCAT(a,b) a ## b
#endif
```

### On which OS does this issue occur?

[OS: Linux]

### What is the version of your OS?

Ubuntu 22.04.5 LTS

### NuttX Version

Master

### Issue Architecture

[Arch: arm]

### Issue Area

[Area: Board support]

### Verification

- [X] I have verified before submitting the report.

xiaoxiang781216, 2024-10-08:
where __CONCAT define in nuttx? @TimJTi

TimJTi, 2024-10-08:
> where __CONCAT define in nuttx? @TimJTi

@xiaoxiang781216 
```

/home/tim/designs/nuttx/nuttx/arch/arm/src/a1x/hardware/a10_memorymap.h
  48,9: #define __CONCAT(a,b) a ## b
  49,20: #define MKULONG(a) __CONCAT(a,ul)

/home/tim/designs/nuttx/nuttx/arch/arm/src/am335x/hardware/am3358_memorymap.h
  48,9: #define __CONCAT(a,b) a ## b
  49,20: #define MKULONG(a) __CONCAT(a,ul)

/home/tim/designs/nuttx/nuttx/arch/arm/src/chip/hardware/_sama5d2x_memorymap.h
  48,9: #define __CONCAT(a,b) a ## b
  49,20: #define MKULONG(a) __CONCAT(a,ul)

/home/tim/designs/nuttx/nuttx/arch/arm/src/chip/hardware/_sama5d3x_memorymap.h
  48,9: #define __CONCAT(a,b) a ## b
  49,20: #define MKULONG(a) __CONCAT(a,ul)

/home/tim/designs/nuttx/nuttx/arch/arm/src/chip/hardware/_sama5d4x_memorymap.h
  48,9: #define __CONCAT(a,b) a ## b
  49,20: #define MKULONG(a) __CONCAT(a,ul)

/home/tim/designs/nuttx/nuttx/arch/arm/src/imx6/hardware/imx_memorymap.h
  53,9: #define __CONCAT(a,b) a ## b
  54,20: #define MKULONG(a) __CONCAT(a,ul)

/home/tim/designs/nuttx/nuttx/arch/arm/src/sama5/hardware/_sama5d2x_memorymap.h
  48,9: #define __CONCAT(a,b) a ## b
  49,20: #define MKULONG(a) __CONCAT(a,ul)

/home/tim/designs/nuttx/nuttx/arch/arm/src/sama5/hardware/_sama5d3x_memorymap.h
  48,9: #define __CONCAT(a,b) a ## b
  49,20: #define MKULONG(a) __CONCAT(a,ul)

/home/tim/designs/nuttx/nuttx/arch/arm/src/sama5/hardware/_sama5d4x_memorymap.h
  48,9: #define __CONCAT(a,b) a ## b
  49,20: #define MKULONG(a) __CONCAT(a,ul)

/home/tim/designs/nuttx/nuttx/nuttx/arch/arm/src/a1x/hardware/a10_memorymap.h
  48,9: #define __CONCAT(a,b) a ## b
  49,20: #define MKULONG(a) __CONCAT(a,ul)

/home/tim/designs/nuttx/nuttx/nuttx/arch/arm/src/am335x/hardware/am3358_memorymap.h
  48,9: #define __CONCAT(a,b) a ## b
  49,20: #define MKULONG(a) __CONCAT(a,ul)

/home/tim/designs/nuttx/nuttx/nuttx/arch/arm/src/imx6/hardware/imx_memorymap.h
  53,9: #define __CONCAT(a,b) a ## b
  54,20: #define MKULONG(a) __CONCAT(a,ul)

/home/tim/designs/nuttx/nuttx/nuttx/arch/arm/src/sama5/hardware/_sama5d2x_memorymap.h
  48,9: #define __CONCAT(a,b) a ## b
  49,20: #define MKULONG(a) __CONCAT(a,ul)

/home/tim/designs/nuttx/nuttx/nuttx/arch/arm/src/sama5/hardware/_sama5d3x_memorymap.h
  48,9: #define __CONCAT(a,b) a ## b
  49,20: #define MKULONG(a) __CONCAT(a,ul)

/home/tim/designs/nuttx/nuttx/nuttx/arch/arm/src/sama5/hardware/_sama5d4x_memorymap.h
  48,9: #define __CONCAT(a,b) a ## b
  49,20: #define MKULONG(a) __CONCAT(a,ul)

```