apache/nuttx #13566: pico-sdk doesn't have picotool anymore

acassis, 2024-09-21:
### Description / Steps to reproduce the issue

$ export PICO_SDK_PATH=~/pico-sdk

$ ./tools/configure.sh raspberrypi-pico:usbnsh

$ make -j
picotool uf2 convert --quiet -t elf nuttx nuttx.uf2;
/usr/bin/bash: line 1: picotool: command not found

### On which OS does this issue occur?

[OS: Linux]

### What is the version of your OS?

Ubuntu 24.10

### NuttX Version

master

### Issue Architecture

[Arch: arm]

### Issue Area

[Area: Build System]

### Verification

- [X] I have verified before submitting the report.

acassis, 2024-09-21:
I found the solution here: https://forums.raspberrypi.com/viewtopic.php?t=358951

```
cd ~/pico
sudo apt install libusb-1.0-0-dev
git clone https://github.com/raspberrypi/picotool.git
cd picotool
mkdir build
cd build
cmake ..
make
sudo ln -s ~/pico/picotool/build/picotool /usr/local/bin/picotool
```

Note: using the pico-sdk branch 1.1.2 is failing on Ubuntu 24.10:

```
make[1]: *** No rule to make target '/home/alan/pico-sdk/src/rp2040/boot_stage2/boot2_w25q080.S', needed by 'rp2040_boot_stage2.elf'.  Stop.
make: *** [tools/Unix.mk:548: nuttx] Error 2
```

linguini1, 2024-09-23:
This is similar to the findings in #13483.

I opened a PR to change the docs for the Pico and Pico W but it appears other pages need updating too. I believe the Pico SDK back-ported changes from 2.0.0 to other versions which broke 1.1.2, but I can't find evidence of that in their git history.

acassis, 2024-09-23:
Thank you @linguini1 !!! Actually, compiling/installing the picotool and using version 2.0.0 works fine, but we need to check if WiFi still working for this version as well.

PR #14621 (2024-11-04): Pico SDK version bump + assimilate old READMEs into RST
## Summary

This PR changes all of the RP2040 documentation (chip and all RP2040-based boards) to use the 2.0.0 SDK version in their installation instructions. I have also moved all of the information from the boards' `README.txt` files into their `index.rst` files to help with transitioning the old READMEs into the new RST format.

Closes #13566, since now all pages document requirement to install picotool.

This PR comes from InSpace.

## Impact

The installation instructions will now be up to date with the latest changes in the NuttX k [... cut]
closed 2024-11-04 by xiaoxiang781216