apache/nuttx #13611: [HELP] Image size error after enabling wifi on esp32c6-devkitc:mcuboot_nsh

engdavidiogo, 2024-09-25:
### Description

I'm trying to run a demo of **MCUboot** and **NuttX** on my **ESP32-C6**. By configuring the project with _esp32c6-devkitc:mcuboot_nsh_, I can compile and burn the firmware on the ESP32 without any problems. However, when trying to enable Wi-Fi in my settings, I start facing issues related to image size.

I studied both projects and tried to configure them according to the information below. Please let me know what I'm doing wrong or what would be the best approach to tracking down the root cause of this issue.

[image]

[image]

[image]

My defconfig

CONFIG_ALLOW_BSD_COMPONENTS=y
CONFIG_ARCH="risc-v"
CONFIG_ARCH_BOARD="esp32c6-devkitc"
CONFIG_ARCH_BOARD_COMMON=y
CONFIG_ARCH_BOARD_ESP32C6_DEVKITC=y
CONFIG_ARCH_CHIP="esp32c6"
CONFIG_ARCH_CHIP_ESP32C6=y
CONFIG_ARCH_CHIP_ESP32C6WROOM1=y
CONFIG_ARCH_INTERRUPTSTACK=2048
CONFIG_ARCH_RISCV=y
CONFIG_ARCH_STACKDUMP=y
CONFIG_BOARDCTL_RESET=y
CONFIG_BOARD_LOOPSPERMSEC=15000
CONFIG_BUILTIN=y
CONFIG_DEV_ZERO=y
CONFIG_ESPRESSIF_BOOTLOADER_MCUBOOT=y
CONFIG_DRIVERS_IEEE80211=y
CONFIG_DRIVERS_WIRELESS=y
CONFIG_ESPRESSIF_ESP32C6=y
CONFIG_ESPRESSIF_SPIFLASH=y
CONFIG_ESPRESSIF_SPIFLASH_SPIFFS=y
CONFIG_ESPRESSIF_WIFI=y
CONFIG_EXAMPLES_RANDOM=y
CONFIG_FS_PROCFS=y
CONFIG_IDLETHREAD_STACKSIZE=2048
CONFIG_INIT_ENTRYPOINT="nsh_main"
CONFIG_INIT_STACKSIZE=8192
CONFIG_INTELHEX_BINARY=y
CONFIG_IOB_THROTTLE=24
CONFIG_LIBC_PERROR_STDOUT=y
CONFIG_LIBC_STRERROR=y
CONFIG_NETDB_DNSCLIENT=y
CONFIG_NETDEV_LATEINIT=y
CONFIG_NETDEV_PHY_IOCTL=y
CONFIG_NETDEV_WIRELESS_IOCTL=y
CONFIG_NETDEV_WORK_THREAD=y
CONFIG_NETUTILS_CJSON=y
CONFIG_NET_BROADCAST=y
CONFIG_NET_ICMP_SOCKET=y
CONFIG_NET_TCP=y
CONFIG_NET_TCP_DELAYED_ACK=y
CONFIG_NET_TCP_WRITE_BUFFERS=y
CONFIG_NET_UDP=y
CONFIG_NFILE_DESCRIPTORS_PER_BLOCK=6
CONFIG_NSH_ARCHINIT=y
CONFIG_NSH_BUILTIN_APPS=y
CONFIG_NSH_FILEIOSIZE=512
CONFIG_NSH_READLINE=y
CONFIG_NSH_STRERROR=y
CONFIG_PREALLOC_TIMERS=0
CONFIG_PTHREAD_MUTEX_TYPES=y
CONFIG_RR_INTERVAL=200
CONFIG_SCHED_BACKTRACE=y
CONFIG_SCHED_LPWORK=y
CONFIG_SCHED_WAITPID=y
CONFIG_SIG_DEFAULT=y
CONFIG_START_DAY=29
CONFIG_START_MONTH=11
CONFIG_START_YEAR=2019
CONFIG_SYSTEM_DHCPC_RENEW=y
CONFIG_SYSTEM_DUMPSTACK=y
CONFIG_SYSTEM_NSH=y
CONFIG_SYSTEM_PING=y
CONFIG_TESTING_GETPRIME=y
CONFIG_TESTING_OSTEST=y
CONFIG_TLS_TASK_NELEM=4
CONFIG_UART0_SERIAL_CONSOLE=y
CONFIG_WIRELESS=y
CONFIG_WIRELESS_WAPI=y
CONFIG_WIRELESS_WAPI_CMDTOOL=y
CONFIG_WIRELESS_WAPI_INITCONF=y

### Verification

- [X] I have verified before submitting the report.

FelipeMdeO, 2024-09-25:
Hello @fdcavalcanti can you help us to do it work, please ?

fdcavalcanti, 2024-09-25:
Hello, thanks for reporting. Will try and replicate it.

fdcavalcanti, 2024-09-26:
It seems like an issue with `imgtool`. The 0x50000004c image size seems excessive.
Can you check the .hex file size?

Some things to try:
- On bootloader configuration, try to modify the hash to the newest mcuboot version. 
- Update imgtool to newest version using pip.

engdavidiogo, 2024-09-26:
I just followed your instructions but I still have the same behavior.

[image]

[image]

[image]

fdcavalcanti, 2024-09-27:
Thanks for the info. Could be a linker issue, looking into it.

fdcavalcanti, 2024-10-03:
I saw that PR fixed the build issue but WiFi stopped. I jumped the gun on that PR, sorry about that.

FelipeMdeO, 2024-10-03:
Thank you for the effort.

fdcavalcanti, 2024-10-03:
Hello. So I noticed you guys still had issues on that linker fix.
There is a different issue going on regarding WiFi stability that was affected by a different merge request.
As soon as that is fixed, I will reopen that linker MR and it should be good. Just tested on a C6 using those fixes and its working as expected.

engdavidiogo, 2024-10-03:
Hi @fdcavalcanti, good news.
Are there any issues related to this fix?

Furthermore, I understand that you were able to track down the commit that caused the break. Could you share your successful test scenario

fdcavalcanti, 2024-10-03:
Yes. You guys got tangled between two issues.
One is the one you reported, which was fixed on #13795. I can't submit that until another issue gets fixed, which is #13814.

Here's the output using the defconfig you provided previously, connecting to my WiFi hotspot:
```
nsh> wapi psk wlan0 espubuntu123 3
nsh> wapi essid wlan0 espubuntu 1
nsh> renew wlan0
nsh> ifconfig
wlan0	Link encap:Ethernet HWaddr 60:55:f9:f6:06:04 at RUNNING mtu 576
	inet addr:10.42.0.199 DRaddr:10.42.0.1 Mask:255.255.255.0

nsh> 
```

xiaoxiang781216, 2024-10-07:
@FelipeMdeO could you try this patch https://github.com/apache/nuttx/pull/13841?

fdcavalcanti, 2024-10-07:
Looks good. Already reopened the PR for linker fix and this issue would be done.

closed 2024-10-07 by xiaoxiang781216