apache/nuttx #13624: [HELP] pl011_regs.CR set as 0 in arm64_earlyserialinit() then the _info() in nx_start() cannot print correctly

lzc1001, 2024-09-26:
### Description

Hi everyone,
In recent, I want to add some _info() in nx_start(), but the information don't print to ternimal on FVP. But I can use "dmesg" instruction to output the log information or set CONFIG_CONSOLE_SYSLOG=y. But the latter is set, the pl011 serial can't be initialized. Then, I find the arm64_earlyserialinit() in fvp_boot.c disable pl011_regs.cr and don't enbale until running DEBUGVERIFY(group_setupidlefiles()) in nx_start(). So, how do I print information before running DEBUGVERIFY(group_setupidlefiles()) and initialize pl011 serial currectly?
Thank you!
[image]
[image]
[image]

### Verification

- [X] I have verified before submitting the report.

acassis, 2024-09-26:
Hi @qinwei2004 since you are the author of fvp_boot.c, could you please help here?

@xiaoxiang781216 @anchao for awareness

anchao, 2024-09-27:
@lzc1001 I think you should enable `SYSLOG_DEFAULT` instead of `CONSOLE_SYSLOG`. The log can be printed during the startup phase because fifo is enabled in `pl011_earlyserialinit()` (`pl011_enable_fifo()`). Any parameter changes need to wait for user ioctl, so it is correct to call `pl011_enable()` after open.

```
xxx_boot
|
|->pl011_earlyserialinit
|  |
|   ->pl011_setup
|     |
|     |->pl011_disable
|     |->pl011_disable_fifo
|     |
|      ->pl011_enable_fifo  <-- enabled here
|
|->nx_start
|  |
|   ->group_setupidlefiles
|     |
|      ->open
|        |
|         ~>pl011_attach
|           |
|            ->pl011_enable
|
|~>ioctl ?

```

lzc1001, 2024-09-29:
Thank you very much! @acassis @anchao 
I set CONFIG_SYSLOG_DEFAULT=y, but the FVP terminal cann't print completely.
[image]
[image]
I remember the added _info() can be printed before lost configuration file😫(I forget configuration I added). I am trying to use different configuration and look the syslog code for solving the problem, hhh

anchao, 2024-09-29:
@lzc1001 It seems that you only printed out 16 bytes, which may be consistent with the hardware fifo size. Please confirm whether your code has the following PR. If up_putc() does not have flow control before, you will have this problem.

https://github.com/apache/nuttx/pull/13384

lzc1001, 2024-10-02:
@anchao Thank you very much! When I add "while (!pl011_irq_tx_ready(sport));" to pl011_send(), the code will pending. So, it mean that the fifo is full. and fifo don't still clean.
[image]

lzc1001, 2024-10-11:
@anchao @acassis Yesterday, I found this issue generated by IDE platform, not configuration file. Thank you very much!

closed 2024-10-11 by lzc1001