apache/nuttx #13648: [BUG] Regression when building with C++ toolchain enabled

leducp, 2024-09-26:
### Description / Steps to reproduce the issue

Since commit 4cec713dbf86c82e1f48abb0815dfd2f98016b68 I cannot build the stm32f411e-disco:nsh example with libcxxtoolchain enabled.

To reproduce:
git co 4cec713dbf86c82e1f48abb0815dfd2f98016b68
./tools/configure.sh -E -l stm32f411e-disco:nsh
make menuconfig <- enable libcxxtoolchain (LIBCXXTOOLCHAIN [=y])
make

Got this error:
```
P:➜  nuttx git:(4cec713dbf) LC_ALL=C make     
CC:  chip/stm32_gpio.c chip/stm32_gpio.c:41:11: note: '#pragma message: CONFIG_STM32_USE_LEGACY_PINMAP will be deprecated migrate board.h see tools/stm32_pinmap_tool.py'
   41 | #  pragma message "CONFIG_STM32_USE_LEGACY_PINMAP will be deprecated migrate board.h see tools/stm32_pinmap_tool.py"
      |           ^~~~~~~
CXX:  libcxxmini/libxx_new.cxx libcxxmini/libxx_new.cxx:72:11: error: redefinition of 'void* operator new(std::size_t, void*)'
   72 | FAR void *operator new(std::size_t nbytes, FAR void *ptr)
      |           ^~~~~~~~
In file included from /usr/include/newlib/c++/13.2.1/bits/atomic_base.h:36,
                 from /usr/include/newlib/c++/13.2.1/atomic:41,
                 from /home/leduc/wdc_workspace/src/autre/nuttx_dev/nuttx/include/nuttx/atomic.h:32,
                 from /home/leduc/wdc_workspace/src/autre/nuttx_dev/nuttx/include/nuttx/fs/fs.h:38,
                 from /home/leduc/wdc_workspace/src/autre/nuttx_dev/nuttx/include/nuttx/lib/lib.h:30,
                 from libcxxmini/libxx_new.cxx:29:
/usr/include/newlib/c++/13.2.1/new:174:33: note: 'void* operator new(std::size_t, void*)' previously defined here
  174 | _GLIBCXX_NODISCARD inline void* operator new(std::size_t, void* __p) _GLIBCXX_USE_NOEXCEPT
      |                                 ^~~~~~~~
make[1]: *** [Makefile:69: libxx_new.o] Error 1
make: *** [tools/LibTargets.mk:216: libs/libxx/libxx.a] Error 2
```

I'm not really sure to understand why the build system is targeting the libcxxmini in first place.

### On which OS does this issue occur?

[OS: Linux]

### What is the version of your OS?

Debian testing

### NuttX Version

master

### Issue Architecture

[Arch: arm]

### Issue Area

[Area: Build System]

### Verification

- [X] I have verified before submitting the report.

leducp, 2024-09-26:
The previous commit works fine, but since commit 4cec713dbf is unrelated to build system, it seems the bug was here before in atomic.h header.

xiaoxiang781216, 2024-09-26:
@crafcat7 please take a look. @leducp it's better to enable libcxxtoolchain in one of mainline defconfig to catch the similar issue in the future.

crafcat7, 2024-09-27:
This issue has been found in
https://github.com/apache/nuttx/blob/fd182dad6414bc614d6efb3f6deabb0d7e4543a7/libs/libxx/Makefile#L32-L41

When LIBCXXTOOLCHAIN ​​is enabled, LIBCXXMINI should not be included in the compilation because they are mutually exclusive

I submitted a PR to fix this issue：https://github.com/apache/nuttx/pull/13663

leducp, 2024-09-27:
> @crafcat7 please take a look. @leducp it's better to enable libcxxtoolchain in one of mainline defconfig to catch the similar issue in the future.

OK but... what is a "mainline" defconfig? :sweat_smile:

xiaoxiang781216, 2024-09-27:
you can add a new one, or reuse one of them.

closed 2024-09-27 by xiaoxiang781216