apache/nuttx #13704: [BUG] Cannot build telnet configuration for Raspberry Pi Pico W

linguini1, 2024-09-28:
### Description / Steps to reproduce the issue

I cannot build the telnet configuration for the Pi Pico W following the [documentation from the NuttX website](https://nuttx.apache.org/docs/latest/platforms/arm/rp2040/boards/raspberrypi-pico-w/index.html#readme-txt).

When I attempt to build the configuration, I get several errors about implicit definitions of a few functions, which you
can see in the attached error log. I have grepped the kernel source tree for `cpu_to_le16` as well as the Pico SDK and
there is no definition for it, only usages in the ieee80211 code.

Any ideas how to solve this?

```console
CC:  sched/sched_getfiles.c In file included from wireless/ieee80211/bcm43xxx/bcmf_driver.c:43:
/home/linguini/cuinspace/pico-nuttx/nuttx/include/nuttx/wireless/ieee80211/ieee80211.h: In function 'ieee80211_has_tods':
/home/linguini/cuinspace/pico-nuttx/nuttx/include/nuttx/wireless/ieee80211/ieee80211.h:412:16: error: implicit declaration of function 'cpu_to_le16' [-Wimplicit-function-declaration]
  412 |   return (fc & cpu_to_le16(IEEE80211_FCTL_TODS)) != 0;
      |                ^~~~~~~~~~~
CC:  sched/sched_addreadytorun.c /home/linguini/cuinspace/pico-nuttx/nuttx/include/nuttx/wireless/ieee80211/ieee80211.h: In function 'ieee80211_he_ppe_size':
/home/linguini/cuinspace/pico-nuttx/nuttx/include/nuttx/wireless/ieee80211/ieee80211.h:2532:7: error: implicit declaration of function 'hweight8' [-Wimplicit-function-declaration]
 2532 |   n = hweight8(ppe_thres_hdr &
      |       ^~~~~~~~
CC:  sched/sched_removereadytorun.c /home/linguini/cuinspace/pico-nuttx/nuttx/include/nuttx/wireless/ieee80211/ieee80211.h: In function 'ieee80211_he_oper_size':
/home/linguini/cuinspace/pico-nuttx/nuttx/include/nuttx/wireless/ieee80211/ieee80211.h:2608:20: error: implicit declaration of function 'le32_to_cpu' [-Wimplicit-function-declaration]
 2608 |   he_oper_params = le32_to_cpu(he_oper->he_oper_params);
      |                    ^~~~~~~~~~~
/home/linguini/cuinspace/pico-nuttx/nuttx/include/nuttx/wireless/ieee80211/ieee80211.h: In function 'ieee80211_he_spr_size':
/home/linguini/cuinspace/pico-nuttx/nuttx/include/nuttx/wireless/ieee80211/ieee80211.h:2682:23: error: implicit declaration of function 'BIT' [-Wimplicit-function-declaration]
 2682 |   if (he_spr_params & IEEE80211_HE_SPR_NON_SRG_OFFSET_PRESENT)
      |                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
CC:  sched/sched_mergepending.c make[1]: *** [Makefile:105: bcmf_driver.o] Error 1
make[1]: *** Waiting for unfinished jobs....
CC:  semaphore/sem_rw.c make: *** [tools/LibTargets.mk:101: drivers/libdrivers.a] Error 2
make: *** Waiting for unfinished jobs....
```

### On which OS does this issue occur?

[OS: Linux]

### What is the version of your OS?

Linux 6.10.10-arch1-1 x86_64 GNU/Linux

### NuttX Version

master

### Issue Architecture

[Arch: arm]

### Issue Area

[Area: Build System], [Area: Networking]

### Verification

- [X] I have verified before submitting the report.

linguini1, 2024-09-30:
@anchao Hey, sorry for the ping! I noticed using `git blame` that the addition of most of these missing definitions occurred within this PR which you authored: https://github.com/apache/nuttx/pull/5755

I was wondering if you might be able to shed some light on the issue I am having. Were these functions previously defined somewhere in the kernel source tree and they are now missing?

anchao, 2024-09-30:
There may be some internal implementations that have not been synchronized to the community. @zhhyu7 @xiaoxiang781216 Please provide some help with this issue

zhhyu7, 2024-09-30:
> There may be some internal implementations that have not been synchronized to the community. @zhhyu7 @xiaoxiang781216 Please provide some help with this issue

@anchao @linguini1 , It should be that no one has used the inline function in the `ieee80211.h` file before, so there is no similar error, have you added a call to the `ieee80211_has_tods` function locally? In this case, you can add these missing symbols at the same time.

linguini1, 2024-09-30:
> @anchao @linguini1 , It should be that no one has used the inline function in the `ieee80211.h` file before, so there is no similar error, have you added a call to the `ieee80211_has_tods` function locally? In this case, you can add these missing symbols at the same time.

I'm not quite sure what you mean, sorry. The header is pulled in when networking support is added for the Pi Pico W, which I assume has been tested before. I'm not sure if it's been tested with this header, but I would expect since it's a supported MCU that someone has tried WiFi on the Pico W before.

I have not added a call to `ieee80211_has_tods` locally, I am using a premade configuration for the Pico W. It appears this header is used as part of the driver for the WiFi chip on it. I'm just confused as to where these symbols are located, since they're nowhere to be found in the NuttX source tree.

zhhyu7, 2024-09-30:
> > @anchao @linguini1 , It should be that no one has used the inline function in the `ieee80211.h` file before, so there is no similar error, have you added a call to the `ieee80211_has_tods` function locally? In this case, you can add these missing symbols at the same time.
> 
> @zhhyu7 I'm not quite sure what you mean, sorry. The header is pulled in when networking support is added for the Pi Pico W, which I assume has been tested before. I'm not sure if it's been tested with this header, but I would expect since it's a supported MCU that someone has tried WiFi on the Pico W before.
> 
> I have not added a call to `ieee80211_has_tods` locally, I am using a premade configuration for the Pico W. It appears this header is used as part of the driver for the WiFi chip on it. I'm just confused as to where these symbols are located, since they're nowhere to be found in the NuttX source tree.

@linguini1 Can you show me the command you used when compiling, I'll try to reproduce it locally.
Is it similar to the following？
```
./tools/configure.sh raspberrypi-pico-w:telnet
make -j
```

linguini1, 2024-09-30:
@zhhyu7, yes that is the command I'm using!

linguini1, 2024-09-30:
For the record I tried doing this on Ubuntu in WSL and it worked. I am not sure what the difference is between my two systems, still investigating.

linguini1, 2024-10-16:
@zhhyu7 any luck reproducing the issue? I've not been able to figure out the difference between WSL Ubuntu and Arch Linux that would account for these definitions being missing.

zhhyu7, 2024-10-17:
@anchao There is really no relevant function implementation in nuttx, help to see? Is it the influence of the gcc version?

khang-874, 2024-11-06:
I also have the same problem

linguini1, 2024-11-06:
I managed to get the build working by adding the following lines above the included headers:
```c
#define cpu_to_le16 __builtin_bswap16
#define hweight8 __builtin_popcount
#define le32_to_cpu __builtin_bswap32
#define BIT(x) (1 << x)
```

I maybe these macros could be defined by NuttX in `compiler.h` or `endian.h` for systems where bswap functions exist? The gcc built-ins will do the job and this should solve the problem for Arch Linux while keeping builds working on Ubuntu. I did confirm after a successful build that running the IEEE802.11 driver worked (did this by using the Pico W `telnet` configuration and checking that I could connect to my home network and ping google.com).

@anchao @zhhyu7 I must admit that this is out of my element, what are your thoughts on this potential solution?

closed 2024-12-23 by xiaoxiang781216