apache/nuttx #13757: [BUG] is_power_of_2 of nuttx conflict

nicolas71640, 2024-09-30:
### Description / Steps to reproduce the issue

The compilation of botan with nuttx fails. It comes with the redefinition of **is_power_of_2** function : declared in botan AND in nuttx int math32.h.
 
Here's the error with the chain of includes. 

```
In file included from /build/nuttx/include/nuttx/clock.h:36,
                 from /build/nuttx/include/nuttx/semaphore.h:33,
                 from /build/nuttx/include/nuttx/mutex.h:31,
                 from /build/nuttx/include/nuttx/fs/fs.h:39,
                 from /build/nuttx/include/stdio.h:34,
                 from /build/nuttx/include/wchar.h:35,
                 from /build/nuttx/include/cxx/cwchar:30,
                 from /install/arm-none-eabi/include/c++/12.3.1/bits/postypes.h:40,
                 from /install/arm-none-eabi/include/c++/12.3.1/iosfwd:40,
                 from /install/arm-none-eabi/include/c++/12.3.1/ios:38,
                 from /install/arm-none-eabi/include/c++/12.3.1/ostream:38,
                 from /install/arm-none-eabi/include/c++/12.3.1/bits/unique_ptr.h:41,
                 from /install/arm-none-eabi/include/c++/12.3.1/memory:75,
                 from build/include/botan/types.h:17,
                 from build/include/botan/secmem.h:11,
                 from build/include/botan/symkey.h:11,
                 from build/include/botan/sym_algo.h:11,
                 from build/include/botan/block_cipher.h:11,
                 from build/include/botan/aes.h:11,
                 from src/lib/block/aes/aes.cpp:7:
build/include/botan/internal/bit_ops.h:43:39: error: expected ')' before 'arg'
   43 | inline constexpr bool is_power_of_2(T arg)

```

From what I understand, the nuttx libc shouldn't defined public functions outside of the one defines by the libc standard. 

Any ideas how can we fix that ? 

### On which OS does this issue occur?

[OS: Linux]

### What is the version of your OS?

Ubuntu 22

### NuttX Version

master

### Issue Architecture

[Arch: all]

### Issue Area

[Area: Other]

### Verification

- [X] I have verified before submitting the report.

TimJTi, 2024-09-30:
I had similar using LVGL (with the OS set to NuttX) also down to a conflict with a math32 function. I raised an issue on LVGL [here](https://github.com/lvgl/lvgl/issues/6949)

The proposal there was to rename but my feeling is that the use of `static` is a little too prevalent in LVGL - it's almost a habit - so could it be the same here? Perhaps an unhappy mix/match of C and C++-type libraries/functions? Not sure, but if this is twice this has arisen for different but similar reasons, perhaps there's a "root cause" that needs addressing somewhere?

xiaoxiang781216, 2024-09-30:
@GUIDINGLI look at this problem.

TimJTi, 2024-10-01:
Also, [[BUG] Compile warning due to __CONCAT redefinition #13562](https://github.com/apache/nuttx/issues/13562) is possibly related given the prevalence of #define functions/helpers in NuttX?

GUIDINGLI, 2024-10-07:
How about this:
https://github.com/apache/nuttx/pull/13864

closed 2024-10-07 by GUIDINGLI