apache/nuttx #13765: [HELP] <title> ARMv8r FVP (fvp-armv8r) does not shows boot up logs and also telenet to UART 1 port  does not prompt to nuttx shell

govindsi, 2024-09-30:
### Description

I am running fvp-armv8r Fixed Virtual Platforms (FVP) with nuttx.
**Reference:** https://nuttx.apache.org/docs/12.5.0/platforms/arm64/fvp-v8r/boards/fvp-armv8r/index.html

While running fvp, i don't see any boot logs and telnet to UART1 ( port 5001) does not prompts to nuttx shell.
Is the last working FVP model archived somewhere? 
ARM current release has only FVP_Base_AEMv8R_11.27_19_Linux64, i am wondering if some mismatch in load address or other configurations.

FVP_Base_AEMv8R_11.27_19_Linux64/AEMv8R_base_pkg/models/Linux64_GCC-9.3/FVP_BaseR_AEMv8R -f boards/arm64/fvp-v8r/fvp-armv8r/scripts/fvp_cfg.txt -a ./nuttx
terminal_0: Listening for serial connection on port 5000
terminal_1: Listening for serial connection on port 5001
terminal_2: Listening for serial connection on port 5002
terminal_3: Listening for serial connection on port 5003

### Verification

- [X] I have verified before submitting the report.

acassis, 2024-10-03:
@jkivilin @pussuw @lupyuen please take a look ^

pussuw, 2024-10-03:
Never used FVP so no help from here

xiaoxiang781216, 2024-10-07:
@GUIDINGLI please take a look

jerpelea, 2024-10-14:
@GUIDINGLI updates?

jerpelea, 2024-10-21:
@xiaoxiang781216 updates?

jerpelea, 2024-10-21:
@raiden00pl can you help?

jasonbu, 2024-10-21:
@jerpelea @govindsi  I tried get newest [FVP](https://developer.arm.com/-/cdn-downloads/permalink/Fixed-Virtual-Platforms/FM-11.27/FVP_Base_AEMv8R_11.27_19_Linux64.tgz), and master of nuttx, master of nuttx-apps. 
just work all right with my environment.

```bash
# nuttx dir
./tools/configure.sh -l fvp-armv8r:nsh
make -j
~/AEMv8R_base_pkg/models/Linux64_GCC-9.3/FVP_BaseR_AEMv8R -f boards/arm64/fvp-v8r/fvp-armv8r/scripts/fvp_cfg_smp.txt -a ./nuttx
# in new terminal
telnet 127.0.0.1 5001 
ostest
```

please let met know if the problem is already solved.

govindsi, 2024-10-21:
> @jerpelea @govindsi I tried get newest [FVP](https://developer.arm.com/-/cdn-downloads/permalink/Fixed-Virtual-Platforms/FM-11.27/FVP_Base_AEMv8R_11.27_19_Linux64.tgz), and master of nuttx, master of nuttx-apps. just work all right with my environment.
> 
> ```shell
> # nuttx dir
> ./tools/configure.sh -l fvp-armv8r:nsh
> make -j
> ~/AEMv8R_base_pkg/models/Linux64_GCC-9.3/FVP_BaseR_AEMv8R -f boards/arm64/fvp-v8r/fvp-armv8r/scripts/fvp_cfg_smp.txt -a ./nuttx
> # in new terminal
> telnet 127.0.0.1 5001 
> ostest
> ```
> 
> please let met know if the problem is already solved.

Thank you @jasonbu for trying at ur end. Seems the problem with my setup was invalid fvp-armv8r-aarch32:nsh configuration , i was trying with aarch32 configuration as mentioned in https://nuttx.apache.org/docs/latest/platforms/arm/fvp-v8r-aarch32/boards/fvp-armv8r-aarch32/index.html. fvp-armv8r:nsh target configuration is working fine with latest ARMv8r FVP.

jerpelea, 2024-10-22:
@jasonbu 

please try the same on releases/12.7 branch 

Thanks
Alin

jasonbu, 2024-10-22:
> @jasonbu
> 
> please try the same on releases/12.7 branch
> 
> Thanks Alin

it means you can confirm in your environment, the releases/12.7 can still re-produce the problem? if you can provide these information, we can speed up the local re-produce and code diff. or if you can provide from which commit problem happened or since which commit problem disappear, will be very appreciate.

jerpelea, 2024-10-22:
> it means you can confirm in your environment, the releases/12.7 can still re-produce the problem? if you can provide these information, we can speed up the local re-produce and code diff. or if you [...]

I can't test now and I was asking you to test the release branch 
Thanks
Alin

closed 2024-10-24 by xiaoxiang781216

jerpelea, 2024-10-24:
backport https://github.com/apache/nuttx/pull/14493