apache/nuttx #13776: [BUG] Compile options broken on GCC 9

stbenn, 2024-10-01:
### Description / Steps to reproduce the issue

Since commit 394a967 I can no longer compile Nuttx with make. The compile error appears to be caused by #13720 . The compile parameter that is causing the error, `min-pagesize`, was introduced in GCC version 12 (as far as I can tell). 

## Steps to reproduce
I have both nuttx and nuttx-apps (up to date with master) in a parent directory, then in nuttx directory: 
```
git checkout 394a967
make distclean
./tools/configure.sh -l nucleo-g474re:nsh
make
```
Produces the error. I have checked, the error does not occur with the previous commit (f2bdfa4). 

## Error
```
Create version.h
LN: platform/board to /home/username/nuttxspace/apps/platform/dummy
Register: nsh
Register: sh
CC:  clock/clock.c cc1: error: invalid --param name 'min-pagesize'
make[1]: *** [Makefile:64: clock.o] Error 1
make: *** [tools/LibTargets.mk:65: sched/libsched.a] Error 2
```

## System Info
* GCC version: 9.4.0
* arm-none-eabi-gcc version: 9.2.1
* Linux running in WSL2

### On which OS does this issue occur?

[OS: Linux]

### What is the version of your OS?

Ubuntu 20.04.6 LTS

### NuttX Version

master

### Issue Architecture

[Arch: arm]

### Issue Area

[Area: Build System]

### Verification

- [X] I have verified before submitting the report.

lupyuen, 2024-10-02:
RISC-V Build with GCC 10.2 is also failing: https://github.com/lupyuen/nuttx-ox64/actions/runs/11135472115/job/30945551434

```text
$ riscv64-unknown-elf-gcc -v
gcc version 10.2.0 (SiFive GCC-Metal 10.2.0-2020.12.8) 

$ tools/configure.sh ox64:nsh
$ make
Assembler messages:
Error: cannot find default versions of the ISA extension `zicsr'
Error: cannot find default versions of the ISA extension `zifencei'
/tmp/cchLCHdn.s:3: Error: cannot find default versions of the ISA extension `zicsr'
/tmp/cchLCHdn.s:3: Error: cannot find default versions of the ISA extension `zifencei'
make[1]: *** [Makefile:72: PROXY__assert.o] Error 1
make: *** [tools/LibTargets.mk:238: syscall/libproxies.a] Error 2
CC:  proxies/PROXY__assert.c 
```

But that's based on the older SiFive RISC-V Toolchain. We could probably ask devs to switch from SiFive Toolchain to the newer xPack RISC-V Toolchain (GCC 14.2), [which compiles OK](https://github.com/lupyuen/nuttx-sg2000/actions/runs/11135489983/job/30945604261).

pussuw, 2024-10-03:
> RISC-V Build with GCC 10.2 is also failing: https://github.com/lupyuen/nuttx-ox64/actions/runs/11135472115/job/30945551434
> 
> ```
> $ riscv64-unknown-elf-gcc -v
> gcc version 10.2.0 (SiFive GCC-Metal 10.2.0-2020.12.8) 
> 
> $ tools/configure.sh ox64:nsh
> $ make
> Assembler messages:
> Error: cannot find default versions of the ISA extension `zicsr'
> Error: cannot find default versions of the ISA extension `zifencei'
> /tmp/cchLCHdn.s:3: Error: cannot find default versions of the ISA extension `zicsr'
> /tmp/cchLCHdn.s:3: Error: cannot find default versions of the ISA extension `zifencei'
> make[1]: *** [Makefile:72: PROXY__assert.o] Error 1
> make: *** [tools/LibTargets.mk:238: syscall/libproxies.a] Error 2
> CC:  proxies/PROXY__assert.c 
> ```
> 
> But that's based on the older SiFive RISC-V Toolchain. We could probably ask devs to switch from SiFive Toolchain to the newer xPack RISC-V Toolchain (GCC 14.2), [which compiles OK for Ox64](https://github.com/lupyuen/nuttx-sg2000/actions/runs/11135489983/job/30945604261).
> 
> **Update:** BL602 won't boot when we switch SiFive Toolchain to xPack Toolchain. We might need to tweak the BL602 Makefiles to fix this: https://github.com/lupyuen/nuttx-bl602/releases/tag/upstream-2024-10-02
> 
> ```
> EXCEPTION: Illegal instruction. MCAUSE: 00000002, EPC: 2300f380, MTVAL: 00000000
> PANIC!!! Exception = 00000002
> Current Version: NuttX  12.6.0 b2e69b86ad Oct  2 2024 01:44:08 risc-v
> Assertion failed panic: at file: :0 task: Idle_Task process: Kernel 0x230024ea
> EPC: 2300f380
> ```

I thought I commented on this already..

I see the same issue with my toolchain:
```

riscv64-unknown-elf-gcc -v
Using built-in specs.
COLLECT_GCC=riscv64-unknown-elf-gcc
COLLECT_LTO_WRAPPER=/home/ville/wpc_toolchain/rv64_gcc/bin/../libexec/gcc/riscv64-unknown-elf/10.2.0/lto-wrapper
Target: riscv64-unknown-elf
Configured with: /scratch/jenkins/workspace/tpp-freedom-tools/tpp03--build-binary-packages--parameterized/obj/x86_64-linux-ubuntu14/build/riscv64-unknown-elf-gcc/riscv-gcc/configure --target=riscv64-unknown-elf --host=x86_64-linux-gnu --prefix=/scratch/jenkins/workspace/tpp-freedom-tools/tpp03--build-binary-packages--parameterized/obj/x86_64-linux-ubuntu14/install/riscv64-unknown-elf-gcc-10.2.0-2020.12.8-x86_64-linux-ubuntu14 --with-pkgversion='SiFive GCC-Metal 10.2.0-2020.12.8' --with-bugurl=https://github.com/sifive/freedom-tools/issues --disable-shared --disable-threads --enable-languages=c,c++ --enable-tls --with-newlib --with-sysroot=/scratch/jenkins/workspace/tpp-freedom-tools/tpp03--build-binary-packages--parameterized/obj/x86_64-linux-ubuntu14/install/riscv64-unknown-elf-gcc-10.2.0-2020.12.8-x86_64-linux-ubuntu14/riscv64-unknown-elf --with-native-system-header-dir=/include --disable-libmudflap --disable-libssp --disable-libquadmath --disable-libgomp --disable-nls --disable-tm-clone-registry --src=../riscv-gcc --with-system-zlib --enable-checking=yes --enable-multilib --with-abi=lp64d --with-arch=rv64imafdc CFLAGS=-O2 CXXFLAGS=-O2 'CFLAGS_FOR_TARGET=-Os -mcmodel=medany' 'CXXFLAGS_FOR_TARGET=-Os -mcmodel=medany'
Thread model: single
Supported LTO compression algorithms: zlib
gcc version 10.2.0 (SiFive GCC-Metal 10.2.0-2020.12.8) 

```

acassis, 2024-10-03:
@cuiziwei1 @xiaoxiang781216 for awareness!

xiaoxiang781216, 2024-10-04:
Sorry, we are in the national holiday until next Monday, the response will be a little bit later.

closed 2024-10-05 by xiaoxiang781216

yf13, 2024-10-10:
Ubuntu 22.04 stock RiscV toolchain (in APT package `gcc-riscv64-unknown-elf`) also breaks:

```shell
$ dpkg -S $(which riscv64-unknown-elf-gcc)
gcc-riscv64-unknown-elf: /usr/bin/riscv64-unknown-elf-gcc
$ riscv64-unknown-elf-gcc -v 2>&1 | fgrep -v Config
Using built-in specs.
COLLECT_GCC=riscv64-unknown-elf-gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/riscv64-unknown-elf/10.2.0/lto-wrapper
Target: riscv64-unknown-elf
Thread model: single
Supported LTO compression algorithms: zlib
gcc version 10.2.0 ()
```

xiaoxiang781216, 2024-10-11:
what's error message?

yf13, 2024-10-11:
> What's error message?

For `rv-virt:nsh` with Ubuntu 22.04 `gcc-riscv64-unknown-elf` toolchain:

```
$ ninja
[1/1256] Building C object arch/CMakeF...risc-v/src/qemu-rv/qemu_rv_start.c.obj
FAILED: arch/CMakeFiles/arch.dir/risc-v/src/qemu-rv/qemu_rv_start.c.obj
/usr/bin/riscv64-unknown-elf-gcc -D__KERNEL__ -D__NuttX__ -I/home/yf/Projects/Nuttx/nuttx/arch/risc-v/src/qemu-rv -I/home/yf/Projects/Nuttx/nuttx/arch/risc-v/src/common -I/home/yf/Projects/Nuttx/nuttx/sched -isystem /home/yf/Projects/Nuttx/nuttx/include -isystem /home/yf/tmp/rv64/aaa/include -Os -fno-strict-aliasing -fomit-frame-pointer -fno-common -Wall -Wshadow -Wundef -Wno-attributes -Wno-unknown-pragmas -Wstrict-prototypes -mcmodel=medany -ffunction-sections -fdata-sections -g -march=rv64imafdc_zicsr_zifencei -mabi=lp64d -DNDEBUG -fmacro-prefix-map=/home/yf/Projects/Nuttx/nuttx= -fmacro-prefix-map=/home/yf/Projects/Nuttx/apps= -fmacro-prefix-map=/home/yf/Projects/Nuttx/nuttx/boards/risc-v/qemu-rv/rv-virt= -fmacro-prefix-map=/home/yf/Projects/Nuttx/nuttx/arch/risc-v/src/qemu-rv= -MD -MT arch/CMakeFiles/arch.dir/risc-v/src/qemu-rv/qemu_rv_start.c.obj -MF arch/CMakeFiles/arch.dir/risc-v/src/qemu-rv/qemu_rv_start.c.obj.d -o arch/CMakeFiles/arch.dir/risc-v/src/qemu-rv/qemu_rv_start.c.obj -c /home/yf/Projects/Nuttx/nuttx/arch/risc-v/src/qemu-rv/qemu_rv_start.c
cc1: error: '-march=rv64imafdc_zicsr_zifencei': unsupported ISA subset 'z'
cc1: error: requested ABI requires '-march' to subsume the 'D' extension
cc1: error: ABI requires '-march=rv64'
```

xiaoxiang781216, 2024-10-11:
@cuiziweizw look at this problem.

cuiziwei1, 2024-10-11:
Sorry for the above problem. I have submitted a new patch https://github.com/apache/nuttx/pull/14130  to fix this problem.

anchao, 2024-10-12:
Any changes to the compiler need to take into account developers using different versions of the compiler. It is very strange why PR https://github.com/apache/nuttx/pull/13720 was merged. The impact of deleting it is to continuously apply meaningless patches. Please review the relevant commits carefully. @xiaoxiang781216