apache/nuttx #13828: [BUG] RaspberryPi-Pico-W is not getting WiFi with Pi Pico SDK 2.0

UAV-Pilot, 2024-10-05:
### Description / Steps to reproduce the issue

Except getting the latest Pi Pico SDK 2.0, the rest is the same as https://github.com/apache/nuttx/issues/7475, which is closed now.

Similar to the errors in https://github.com/apache/nuttx/issues/7475, the following errors occurred:

I set SSID and passphrase in menuconfig, and also tried to set them in nsh.

nsh> renew wlan0
ERROR: netlib_obtain_ipv4addr() failed
nsh> ifconfig wlan0
wlan0 Link encap:Ethernet HWaddr 00:00:00:00:00:00 at DOWN mtu 5nsh>
nsh>
nsh> ifup wlan0
ifup wlan0...Failed

ChatGPT suggested to set mac address:
ifconfig wlan0 hw ether XX:XX:XX:XX:XX:XX (e.g., 28:CD:C1:0A:38:9A)
but got error:
nsh: ifconfig: argument invalid

nsh> wapi show wlan0
wlan0 Configuration:
IP: 0.0.0.0
NetMask: 255.255.255.0
nsh> wapi scan wlan0
nsh> wapi scan_results wlan0
nsh> ifup wlan0
ifup wlan0...Failed

### On which OS does this issue occur?

[OS: Linux]

### What is the version of your OS?

Windows 11 WSL2 Ubuntu 22.04.2

### NuttX Version

NuttX  12.7.0-RC0 0976258299-dirty Oct  4 2024 22:58:36 arm raspn

### Issue Architecture

[Arch: arm]

### Issue Area

[Area: Networking]

### Verification

- [X] I have verified before submitting the report.

darwinbeing, 2024-10-06:
Download the BINARY firmware file from this link: [43439A0-7.95.49.00.combined](https://github.com/georgerobotics/cyw43-driver/blob/195dfcc10bb6f379e3dea45147590db2203d3c7b/firmware/43439A0-7.95.49.00.combined).

Make sure the configuration file variable points to the correct firmware file:
CONFIG_CYW43439_FIRMWARE_BIN_PATH="${PICO_SDK_PATH}/lib/cyw43-driver/firmware/43439A0-7.95.49.00.combined"
When compiling, the cyw43439.firmware.image file is created, and it’s the same as firmware/43439A0-7.95.49.00.combined. Ensure the paths and files match during the build, especially in:
boards/arm/rp2040/common/src/cyw43439.firmware.image
This should fix issue.

UAV-Pilot, 2024-10-06:
@darwinbeing Thanks a lot for your quick help! With that firmware, I got wifi working on pi pico W with SDK 2.0.

Besides this firmware issue during setup, along the way I encounter some other issues as shown by the following commands:
1) SSID and passphrase set in menuconfig were not used
2) SSID and passphrase entered manually in wapi commands disappeared after reboot
3) ifconfig did not output ip address, etc.
4) ifup wlan0 returned message OK which was misleading when wlan0 was not connected and not up running correctly

NuttShell (NSH) NuttX-12.7.0-RC0
nsh> ifconfig wlan0
wlan0   Link encap:Ethernet HWaddr 28:cd:c1:0a:38:9a at UP mtu 576nsh>
nsh> ifup wlan0
ifup wlan0...OK
nsh>
nsh> wapi show wlan0
wlan0 Configuration:
       IP: 0.0.0.0
  NetMask: 255.255.255.0
Frequency: 2412
     Flag: WAPI_FREQ_AUTO
  Channel: 1
Frequency: 2412
    ESSID:
     Flag: WAPI_ESSID_ON
     Mode: WAPI_MODE_MANAGED
  BitRate: 72000
     Flag: WAPI_BITRATE_FIXED
    Sense: 0
  Country: XX
nsh>
nsh> wapi psk wlan0 XXXXXXX WPA_ALG_CCMP
nsh> wapi essid wlan0 MySSID WAPI_ESSID_ON
nsh> renew wlan0
nsh> wapi show wlan0
wlan0 Configuration:
       IP: 192.168.1.160
  NetMask: 255.255.255.0
Frequency: 2462
     Flag: WAPI_FREQ_AUTO
  Channel: 11
Frequency: 2462
    ESSID: MySSID 
     Flag: WAPI_ESSID_ON
     Mode: WAPI_MODE_MANAGED
       AP: f8:9b:6e:e7:9e:44
  BitRate: 65000
     Flag: WAPI_BITRATE_FIXED
  TxPower: 31
     Flag: WAPI_TXPOWER_DBM
    Sense: -76
  Country: XX
 
nsh> ifconfig wlan0
wlan0   Link encap:Ethernet HWaddr 28:cd:c1:0a:38:9a at RUNNING mt5.255.0
<No IP address shown>

nsh>  ping 192.168.1.254
PING 192.168.1.254 56 bytes of data
56 bytes from 192.168.1.254: icmp_seq=0 time=0.0 ms
56 bytes from 192.168.1.254: icmp_seq=1 time=0.0 ms
56 bytes from 192.168.1.254: icmp_seq=2 time=0.0 ms
56 bytes from 192.168.1.254: icmp_seq=3 time=0.0 ms
rtt min/avg/max/mdev = 0.000/0.000/0.000/0.000 ms, time 3730 ms
nsh>

linguini1, 2024-10-07:
Is there a known reason why the firmware generated in the 2.0.0 SDK does not work with NuttX? I assume something has changed since then which needs to be accounted for?

linguini1, 2024-10-07:
It appears the [new firmware file](https://github.com/georgerobotics/cyw43-driver/blob/main/firmware/w43439A0_7_95_49_00_combined.h) actually allows the blob binary to be included as a C variable instead of requiring the assembly instruction to include the binary.

Maybe we could update the `boards/arm/rp2040/common/src/rp2040_firmware.c` file to do that? I'll look into it. It would also be good to add a warning for the missing firmware file instead of silently making a dummy file.

darwinbeing, 2024-10-26:
This is the patch to enable w43439 
[w43439.patch](https://github.com/user-attachments/files/17528747/w43439.patch)

make distclean && ./tools/configure.sh -l raspberrypi-pico-w:telnet && make -j16

linguini1, 2024-10-28:
This patch works in the sens that `ifup wlan0` is now returning okay, but it does not work when I try to renew `wlan0`.

```console
nsh> renew wlan0
ERROR: netlib_obtain_ipv4addr() failed
```