apache/nuttx #13830: [BUG] sim/smp is broken

raiden00pl, 2024-10-05:
### Description / Steps to reproduce the issue

on `tag: nuttx-12.7.0-RC0`:

```
./tools/configure.sh sim/smp
 make -j
 
 [raiden00:~/git/RTOS/nuttx/nuttx]$ ./nuttx                                                                                                                                                                                                                        
[1]    332900 segmentation fault (core dumped)  ./nuttx

```

On the current master (a613bcb7cb) we don't get segmentation fault and we can boot to NSH but **it's not functional**: interaction with NSH lags like hell. 

### On which OS does this issue occur?

[OS: Linux]

### What is the version of your OS?

Arch Linux

### NuttX Version

nuttx-12.7.0-RC0, master

### Issue Architecture

[Arch: simulator]

### Issue Area

[Area: Kernel]

### Verification

- [X] I have verified before submitting the report.

raiden00pl, 2024-10-05:
It looks like `sim/nsh` behaves the same. On the other hand `sim/foc` works fine.

xiaoxiang781216, 2024-10-07:
@hujun260 please take a look.

hujun260, 2024-10-08:
https://github.com/apache/nuttx/pull/13933
fix this issue

raiden00pl, 2024-10-12:
I'm reopening this because sim/smp is still broken. Now console output doesn't work properly.
Here are the details: https://github.com/apache/nuttx/pull/13933#issuecomment-2400097817

raiden00pl, 2024-10-12:
another problem is assertion in ostest:

```
signest_test: With task locking
  Total signalled 2480  Odd=1240 Even=1240
  Total handled   2480  Odd=1240 Even=1240
  Total nested    0    Odd=0   Even=0  
[CPU1] dump_assert_info: Current Version: NuttX  10.4.0 6e12b3d74b Oct 12 2024 10:57:16 sim
[CPU1] dump_assert_info: Assertion failed : at file: signest.c:228 task(CPU1): ostest process: ostest 0x441c23
ostest_main: Exiting with status 256
stdio_test: Standard I/O Check: fprintf to stderr

```

raiden00pl, 2024-10-12:
another issue in ostest is broken `wdog_test` that produce garbage output

```
wdog_test start...
[CPU2] wdtest_once 0 ns
[CPU2] wdtest_o[nCcPeU [31C] P nUsw0
d]te swtd_o[tnCecPesU 2t0] _ nwosdn
tceset _0on[ cCneP sU0
3 ]n sw
dtes[t_CoPncUe0 0]  nws[
dCPtUe2s] tw_dotnescte_o n[0cC ePn U0s3 
]n sw
dtest_[onCcPeU 00 ]n s[w
CdPtU2e]s twd_toensc[teC_ Po0Un 3cn]es  w
0d tness
t_[oCnPcUe0 ]0  nwsd
t[eCsPUt2_]o nwcde[tC Pe0Us 3tn]_ sow
dntcees t0_ o[nnCscP
eU 00 ]n sw
[dCtPeUs2t]_ owndctee s0t _nosn
ce 0 ns
[CPU1] wdtest_once 10 ns
[CPU1] wdtest_once 1 ns
[CP[[UCCP2U1P]] Uwd t0ews]td _owtncdeet s1e tsnst
__oon[nCPccU1e]e w  dte11st _ onnncess 1 

ns
[[CCPU1P] wUd[te2sCt]P_oUn ce0 w1] nds 
twde[tCsPeU1ts] wt_dt_eosotn_noncccee e1  n s1
1  n[CsnPU
1s] w
dtest_once 1 ns[
CPU0[]C[ CwPPUd1U]t ew2dst]tes t__owoncdne tc1 nees
s 1t [_CPnUo1s]n 
wdcteest_ on1ce  1 [nnsC
sP
U0] wdtest_once 1 ns
[CPU1] wdtest_on[cCPeU 21] 0w0 dnts
[esCtP[_CUPoU01n]]c  wedw ted1s0tt_ eonnsscet
 1_0 nosn
ce [1C[P0CPU U12n]] sw dtw
edstt_oencse t10_ nos
[ncCeP[ UC1P0U01]]  w ndtwsesd
t_tonecse t10_ nso[
nCcP[UeCP2 U]11]  0wwdd ttenestss_ot
n_ceo 1n0 cnse
 1[0C[ CPnPUsU1
] w0dte]st_ own[cedC 1P0t Un2es
]s twd_[CtPoeU1sn]t wc_dtoeesntc_ oen1 ce1 0100   nsn
ns
s
[CPU1] wdtest[_oCnPceU 210]  nsw
dtest_once 10 ns
[CPU1] wdtest_once 1000 ns
[CPU[2[]CC PPUw1U]d w0tdtee]sstt_ o_ncowe nd1c0t0e n es1
s00t [nC_PsU1o
] wndtecst_eon c[e1 C10P00U n20s]
  wnd[CtPsUe1
s] twd_teosnt_conece  110[00 Cn0s
P nUs0[
CP]U1 ] wwdtdestt_[oenCcesP 1Ut002 _]ns 
ownd[ctCePeU 1s] 1wtdt0_es0to_ nonncce se10
 0 n1s
00 ns[[C
CPU1P] Uwdt0est]_o n[cweC 1dP00t Unse
2s]t [C_wPUo1d] ntwdceteesst _ton1_ce0o 01n0 0 cnnses
 
1[C0PU01]  wndtesst
_once 100 ns
[CPU1] wdtest_once 10000 ns
[... 120 lines cut ...]
seidv et iwcdko [g1 2Ct
Pr[UiC2gPgU]e1 r]re edwc du6tr estsiitvm_eerse ,cw uderolsaigpv sete rd1i 0gt0gi0ec0rk0e 0d10 2u6
s [
tCiPmUe0]s ,w detleaspts_eredc utriscikv e1 21
0[0C0P0U020]0 uwsd
test_recursive 10000000us
[CPU1] recursive wdog triggered 6 times, elapsed tick 12
[CPU2] recursive wd[[oCCgPP UUt03r]]i  [grrCgeePeccUuru1rer]ssd ii rvve6eec   uwtwrddisooimggv e etts rrw,iidg goggegee lrrteaerddpi  gs66g e etdtri iemtemdsie ,sc6 ,k e t leial1mpae2sps
es,de  edtl iatcpiks [ce1Ckd2P  
Ut12i]2c k
r e1c2u
[rCsPiUv0e]  wrdeo[c[gCuC PrtPUsr1Uii]v3g eg]r e ewrcrdeuoedrgc s 6uit vrrtesii gimwgevdesoer,ge   dewt lrd6aio pgtgsgiee mdrte esrtd,ii  c6gek lg tae1ip2rms
eeesdd,   te6il ca[ktpC isP1me2Ued
2 s]t ,ir ceekcl ua1r2ps
[isCvePedU  0w]td oirg[ce CcktPu rUri11sg]2ig 
verere ecwdud ro6sg i tvtierm iewgsdgo,eg[r  CeetPdlrU ia36g]p g sterieremedcedu s r,t6s  iietvcliekam  pews1sde,2od
 g e tltiarcpiksg eg1d[e2 rC
tePidcU k62  1]t[2i C
mrPeeUsc0,u] r es[rliCeavPcpUeus1r e]sdw i rdvteoeic gcuwk rd sto1irg2v i
etg rwgidgeogrge ertderd i g66g  ettriiemdme es6s,  ,tei lmeaelpsas,pe dse letaidpsce kdt  i1tc2ik
c k 1122

wdog_test end...

```

hujun260, 2024-10-12:
if revert one commit (sim: fix context-switch when do wdog callback()) all looks good
git revert a3568af10551c37b839169634c33fbe78044725f

xiaoxiang781216, 2024-10-12:
look like sim_uart driver forget to protect the concurrent access:
https://github.com/apache/nuttx/blob/master/arch/sim/src/sim/sim_uart.c

hujun260, 2024-10-12:
> another problem is assertion in ostest which happens randomly:
> 
> ```
> signest_test: With task locking
>   Total signalled 2480  Odd=1240 Even=1240
>   Total handled   2480  Odd=1240 Even=1240
>   Total nested    0    Odd=0   Even=0  
> [CPU1] dump_assert_info: Current Version: NuttX  10.4.0 6e12b3d74b Oct 12 2024 10:57:16 sim
> [CPU1] dump_assert_info: Assertion failed : at file: signest.c:228 task(CPU1): ostest process: ostest 0x441c23
> ostest_main: Exiting with status 256
> stdio_test: Standard I/O Check: fprintf to stderr
> ```

if revert one commit (sim: fix context-switch when do wdog callback()) signest_test looks good
git revert a3568af10551c37b839169634c33fbe78044725f

jerpelea, 2024-10-16:
backported https://github.com/apache/nuttx/pull/13968

jerpelea, 2024-10-16:
@raiden00pl please check releases/12.7

raiden00pl, 2024-10-16:
@jerpelea I get the same result with "Segmentation fault" for release/12.7.

The branch master still has not solved the problem with missing characters on output

raiden00pl, 2024-10-16:
here backtrace from GDB for `sim/smp` crash on release/12.7:

```
pwndbg> bt
#0  0x00007ffff7d371d8 in pthread_sigmask () from /usr/lib/libc.so.6
#1  0x0000000000459a8d in up_irq_restore (flags=0) at sim/posix/sim_hostirq.c:118
#2  0x00000000004081d0 in leave_critical_section (flags=0) at irq/irq_csection.c:422
#3  0x0000000000405b2b in sched_unlock () at sched/sched_unlock.c:206
#4  0x000000000040445e in sim_unlock () at sim/sim_doirq.c:44
#5  0x000000000045a355 in sim_idle_trampoline (arg=0x7fffffffda00) at sim/posix/sim_hostsmp.c:109
#6  0x00007ffff7d2fded in ?? () from /usr/lib/libc.so.6
#7  0x00007ffff7db30dc in ?? () from /usr/lib/libc.so.6

```

hujun260, 2024-10-16:
> @hujun260 please take a look.

https://github.com/apache/nuttx/pull/1375 try add this

hujun260, 2024-10-16:
> @hujun260 please take a look.
try add this pr to release/12.7
https://github.com/apache/nuttx/pull/13753

raiden00pl, 2024-10-16:
@jerpelea damn, I accidentally did a `git push` to release/12.7 branch with this change https://github.com/apache/nuttx/pull/13753
Shouldn't `git push` to upstream repo be locked somehow in github settings?

raiden00pl, 2024-10-16:
with this change https://github.com/apache/nuttx/pull/13753 `sim/smp` boots.
But I just get crash with `ostest`:

```
user_main: pthread_rwlock_cancel test
pthread_rwlock_cancel: Starting test
[CPU1] dump_assert_info: Current Version: NuttX  12.7.0 1a3537bcab Oct 16 2024 11:26:52 sim
[CPU1] dump_assert_info: Assertion failed tcb->cpu == up_cpu_index(): at file: sched/sched_removereadytorun.c:293 task(CPU1): ostest process: ostest 0x441715
[CPU1] dump_assert_info: Current Version: NuttX  12.7.0 1a3537bcab Oct 16 2024 11:26:52 sim
[CPU1] dump_assert_info: Assertion failed tcb->cpu == up_cpu_index(): at file: sched/sched_removereadytorun.c:293 task(CPU1): ostest process: ostest 0x441715
[1]    2493755 killed     ./nuttx
```

jerpelea, 2024-10-17:
> @jerpelea damn, I accidentally did a `git push` to release/12.7 branch with this change #13753 Shouldn't `git push` to upstream repo be locked somehow in github settings?

not for commiters

jerpelea, 2024-10-17:
@raiden00pl 
backported
https://github.com/apache/nuttx/pull/14380

local test 
NuttShell (NSH) NuttX-12.7.0
nsh> ls
/:
 dev/
 proc/
nsh>

jerpelea, 2024-10-17:
please test and close the bug

raiden00pl, 2024-10-17:
The problem with strange console output is still not solved (probably random missing newline character):
```
NuttShell (NSH) NuttX-12.7.0
nsh> nsh> help
```
or:
```
NuttShell (NSH) NuttX-12.7.0
nsh> help
help usage:  help [-v] [<cmd>]

    .           cp          exit        ls          ps          true        
    [           cmp         expr        mkdir       pwd         truncate    
    ?           dirname     false       mkrd        rm          uname       
    alias       dd          fdinfo      mount       rmdir       umount      
    unalias     df          free        mv          set         unset       
    basename    dmesg       help        pidof       sleep       uptime      
    break       echo        hexdump     poweroff    source      usleep      
    cat         env         kill        quit        test        xd          
    cd          exec        pkill       printf      time        wait        

Builtin Apps:
    getprime    nsh         sh          taskset     
    hello       ostest      smp         
nsh> help
help usage:  help [-v] [<cmd>]
  help [-v] [<cmd>]

    .           cp          exit        ls          ps          true        
    [           cmp         expr        mkdir       pwd         truncate    
    ?           dirname     false       mkrd        rm          uname       
    alias       dd          fdinfo      mount       rmdir       umount      
    unalias     df          free        mv          set         unset       
    basename    dmesg       help        pidof       sleep       uptime      
    break       echo        hexdump     poweroff    source      usleep      
    cat         env         kill        quit        test        xd          
    cat         env         kill        quit       cd          exec        pkill       printf      time        wait        

Builtin Apps:

Builtin Apps:
    getprime    nsh         sh          taskset     
    hello       ostest      smp         
nsh> 
```

jerpelea, 2024-10-17:
> > git revert [a3568af](https://github.com/apache/nuttx/commit/a3568af10551c37b839169634c33fbe78044725f)
> 
> > The problem with strange console output is still not solved (probably random missing newline character):
> > ```
> > NuttShell (NSH) NuttX-12.7.0
> > nsh> nsh> help
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > or:
> > ```
> > NuttShell (NSH) NuttX-12.7.0
> > nsh> help
> > help usage:  help [-v] [<cmd>]
> > 
> > 
> >     .           cp          exit        ls          ps          true        
> >     [           cmp         expr        mkdir       pwd         truncate    
> >     ?           dirname     false       mkrd        rm          uname       
> >     alias       dd          fdinfo      mount       rmdir       umount      
> >     unalias     df          free        mv          set         unset       
> >     basename    dmesg       help        pidof       sleep       uptime      
> >     break       echo        hexdump     poweroff    source      usleep      
> >     cat         env         kill        quit        test        xd          
> >     cd          exec        pkill       printf      time        wait        
> > 
> > Builtin Apps:
> >     getprime    nsh         sh          taskset     
> >     hello       ostest      smp         
> > nsh> help
> > help usage:  help [-v] [<cmd>]
> >   help [-v] [<cmd>]
> > 
> >     .           cp          exit        ls          ps          true        
> >     [           cmp         expr        mkdir       pwd         truncate    
> >     ?           dirname     false       mkrd        rm          uname       
> >     alias       dd          fdinfo      mount       rmdir       umount      
> >     unalias     df          free        mv          set         unset       
> >     basename    dmesg       help        pidof       sleep       uptime      
> >     break       echo        hexdump     poweroff    source      usleep      
> >     cat         env         kill        quit        test        xd          
> >     cat         env         kill        quit       cd          exec        pkill       printf      time        wait        
> > 
> > Builtin Apps:
> > 
> > Builtin Apps:
> >     getprime    nsh         sh          taskset     
> >     hello       ostest      smp         
> > nsh> 
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > it only happens for `sim/smp`
> 
> git revert [a3568af](https://github.com/apache/nuttx/commit/a3568af10551c37b839169634c33fbe78044725f) can solve this

Hi 
[a3568af10551c37b839169634c33fbe78044725f ](https://github.com/apache/nuttx/commit/a3568af10551c37b839169634c33fbe78044725f) is included in the https://github.com/apache/nuttx/pull/14380

raiden00pl, 2024-10-17:
my mistake. I didn't notice that https://github.com/apache/nuttx/pull/14380 PR is not merged. With this PR sim/smp works fine

jerpelea, 2024-10-17:
after merge please close the bug

raiden00pl, 2024-10-17:
ostest is throwing garbage at wdtest, but I think we can close this issue for now:

```
wdog_test start...
[CPU1] wdtest_once 0 ns
[CPU1] wdtest_once 0 ns
[CPU1] wdtest_once 0 ns
[CPU1] wdtest_once 0 ns
[CPU2] wdtest_once 0 ns
[CPU1] wdtest_once 0 ns
[CPU1] wdtest_once 0 ns
[CPU1] wdtest_once 0 ns
[CPU1] wdtest_once 0 ns
[CPU1] wdtest_once 0 ns
[CPU1] wdtest_once 0 ns
[CPU1] wdtest_once 0 ns
[CPU1] wdtest_once 0 ns
[CPU1] wdtest_once 0 ns
[CPU1] wdtest_once 0 ns
[CPU1] wdtest_once 0 ns
[CPU2] wdtest_[onCce P1 [UnsC0
P]U 1w[]CdP Ut2w]e dwsdtttees_ts_ootnnc_ec o1e nn sc
1e  n1[s CP
Un2]s wd
test_once 1 n[s
CPU0] [wC[dPCPtUU2e]1 sw]dtt e_swto_donntccee e1 sn st
1_ onn[sCcP
Ue2]  wdt1es t_noncse 
1[ nCs
PU0] [wCd[CPPtUU2e]1 sw]dtt e_swto_donntccee 1es n ts1
_ onnsc[Ce
PU 2]1 w dtensts_o
nce [1 nCs
PU0] [wC[CdPPUt2U]e 1wsd]tte s_tw_oodnnctec e1e sn ts
1_ onnsc
e 1 ns
[[CCPUP2]U wd0t[e]sCt _PownUcd1et ]10e  snwst
d_toe[nsCcPtUe2_ ]o 1wnd0ctee s tn_1osn0c
 e n1s0 
ns
[CPU0[]C[ PCPUwU12d]] t wdewtesdsttt_o_encsoet n1_0c onesn
c e1 01[ C0PnU 2sn] 
swd
test_once 10 ns
[CPU[0CP][U C1Pw]U2 d]w twddettesestts_o_tnco_e on10n cncsee
  1100 [Cn PsUn2
] swd
test_once 1[0 CnsP
U1] wd[tCeP[sCUtPU_02o]] nw cdtewe sdt1_to0ne cens s1t0
 _nso
nce [1CP0U 1n]s w
dtest_once 10 ns
[[CCPPUU01]]  wwddtteesstt__oonnccee  110000  nnss

[[C[CPPCUPUU021] ]] wwddtte esstwt__doonntcceee  s11t00_00  onnssn

ce 1[C0PU00[]C  PwUnd1t]se sw
td_toenscte_ o1n0c0e  n1s0
0 ns
[CP[UCP2U0][]C  PwUdwt1e]ds twtd_toenecsets _1otn0c0_e  nos1
0n0 cnse
 [1CP0U00][  CwPndUt1se]s 
tw_dotnecset _1o0n0c en s1
00 ns
[CP[UCP2U0][]C  PwUdwt1e]d swttd_toeensctse_ ot1n0c_0e  on1s0n
0 cnse
 [1C0PU00]  wdtnesst_o
nce 100 ns
[CPU0] wdtest_once 1000 ns
[CP[UC2[P]CPUU 0]1w w]ddt etswte_dsontctee_ 1so00t0n _ncso
[... 119 lines cut ...]
e d12 [
CtP[UiC0]cP rkUec 2ur]1s i2vwe
 dwdt[oegC strPitgUg_e1rre]de  6c wtuidmretss,e ieslvtaep_s erd1 e0tci0cuk0 r01s20i
0[vC0PeUu0 ]s1 w
0dt0es0t0_r0ec0ur0siuves 1
0000000us
[CPU0] recursive wdog tr[[iCCPgPUgU2e]1r ]er edrc ue6rc sutirvisemi ewvsdeo, g  wetdrloiaggpg setreredid g tg6ie ctrikem de1 s26,
  teliampesse,d  etliackp s1e2d
 tick[ CP12U[
0CP]U 2r] erceucurrs[siiCvvPeeU  1w]wd odrgeo cgtu rristgirgvieerg egwded ro6ge  dtt irm6ie gst,g ieemrleeaspd,s  e6ed l tatipimcsekes d,1 2 t
eilcakp s1ed2 
[tCPiUc2k]  1r2e
c[uCrPsUi0ve]  w[rdCeoPcgUu 1t]rr sirigegvceeru erwdsd i6ov get i tmwrediso,gg g eetlrraepisdge gde6 r teitcdik m 6e1 s2t,
i meelsa,p seelda[ pCtsPiUe2cd]k   rt1ei2ccu
kr si1v2e
 wdog t[rCigPgUe0r]ed  r6e tc[iuCmrePssU,i1 v]ee l arwpedsceoudgr  sttiirvciek g wg1d2eo
rg etdr i6gg [eCrtPeiUdm2 e]6 s rt,ei cmueerlss,ai pvesel eawdpd soetgdi  cttrkii cg1gke2 r
1e2d
 6 times[, CePlUa0[pC]sPe Udr1 ]et cirucerkc su1ri2sv
iev e wwddoo[ggC Pt Urt2i]rg giregergceuedr rs6ie vdtei  m6we sd,to igem letarspi,sg egdee rlteaidpc ks6 e 1d2t 
itmiecsk,  1e2la
psed tick 12
wdog_test end...

```

closed 2024-10-17 by raiden00pl