apache/nuttx #13831: [BUG] highpri examples are broken

raiden00pl, 2024-10-05:
### Description / Steps to reproduce the issue

Tested with nrf52840-dk/highpri but probably all examples of highpri are broken.

```
./tools/configure.sh nrf52840-dk/highpri 
make -j
```

output on 12.7-RC:

```
Elapsed time: 160 seconds

  basepri[80]: 920 (1%)
  basepri[e0]: 78764 (99%)
  Handler:     79684 (100%)
  Thread:      0 (0%)
```

The correct output should be like this:

```
  basepri[80]: 920 (1%)
  basepri[e0]: 78767 (99%)
  Handler:     44 (0%)
  Thread:      79643 (100%)
```

Broken with 0561b55860f655a863dabc653e5db7b213d8abff 
After this commit we have lose the ability to distinguish in the highpri handler if we are in thread or interrupt context.

When we revert this commit, highpri examples works again:

```
Elapsed time: 160 seconds

  basepri[80]: 459 (1%)
  basepri[e0]: 39383 (99%)
  Handler:     22 (0%)
  Thread:      39820 (100%)
```

ping @hujun260 @xiao

### On which OS does this issue occur?

[OS: Linux]

### What is the version of your OS?

Arch Linux

### NuttX Version

nuttx-12.7.0-RC0, master

### Issue Architecture

[Arch: arm]

### Issue Area

[Area: Kernel]

### Verification

- [X] I have verified before submitting the report.

xiaoxiang781216, 2024-10-07:
@GUIDINGLI

xiaoxiang781216, 2024-10-07:
@raiden00pl could you try https://github.com/apache/nuttx/pull/13606?

jasonbu, 2024-10-08:
hi, please review the patch #13895 

The case want to determine if a interrupt with higher priority the interrupt preemption occurred,
but up_interrupt_context indicates that we self inside interrupt/handedr mode.

so this should be usage problem, and we should update the case, and make it less confused.

closed 2024-10-08 by xiaoxiang781216

jerpelea, 2024-10-18:
backport https://github.com/apache/nuttx/pull/14395