apache/nuttx #13839: [BUG] stm32f746g-disco/fb example show garbage

raiden00pl, 2024-10-05:
### Description / Steps to reproduce the issue

```
./tools/configure.sh stm32f746g-disco/fb
make -j
nsh> fb
<garbage on the screen>
```

It should be:

[image]

But it is:

[image]

### On which OS does this issue occur?

[OS: Linux]

### What is the version of your OS?

Arch Linux

### NuttX Version

nuttx-12.7.0-RC0

### Issue Architecture

[Arch: arm]

### Issue Area

[Area: Drivers], [Area: OS Components]

### Verification

- [X] I have verified before submitting the report.

raiden00pl, 2024-10-05:
`stm32f746g-disco/nxdemo` also broken

cederom, 2024-10-06:
We are waiting for LVGL 9.2.1 release lots of fixes incoming and if all works fine we will include it in NuttX 12.7.0 :-)

raiden00pl, 2024-10-07:
@cederom @xiaoxiang781216 This issue is not related to LVGL. `fb` and `nxlines` examples don't use LVGL, something else is broken.

cederom, 2024-10-07:
This seems the same problem as reported by @TimJTi ?

terry0012, 2024-10-08:
@raiden00pl can you provide the output log of ‘fb’ ?

we want to check FB driver info first.

terry0012, 2024-10-08:
@raiden00pl 
I found that output bpp is 24 in `boards/arm/stm32f7/stm32f746g-disco/include/board.h`
`#define BOARD_LTDC_OUTPUT_BPP           24`
and gpio settings is using 24bpp as below.

```
static const uint32_t g_ltdcpins[] =
{
  GPIO_LTDC_R4, GPIO_LTDC_R5, GPIO_LTDC_R6, GPIO_LTDC_R7,
  GPIO_LTDC_G4, GPIO_LTDC_G5, GPIO_LTDC_G6, GPIO_LTDC_G7,
  GPIO_LTDC_B4, GPIO_LTDC_B5, GPIO_LTDC_B6, GPIO_LTDC_B7,
#if BOARD_LTDC_OUTPUT_BPP > 12
  GPIO_LTDC_R3, GPIO_LTDC_G2, GPIO_LTDC_G3, GPIO_LTDC_B3,
#  if BOARD_LTDC_OUTPUT_BPP > 16
  GPIO_LTDC_R2, GPIO_LTDC_B2,
#    if BOARD_LTDC_OUTPUT_BPP > 18
  GPIO_LTDC_R0, GPIO_LTDC_R1, GPIO_LTDC_G0, GPIO_LTDC_G1,
  GPIO_LTDC_B0, GPIO_LTDC_B1,
#    endif
#  endif
#endif
  GPIO_LTDC_VSYNC, GPIO_LTDC_HSYNC, GPIO_LTDC_DE, GPIO_LTDC_CLK
};
```

But the bpp reported by framebuffer driver is 16, and framebuffer length is also in bpp 16 mode  (261120 = 480 * 272 * 2).

```CONFIG_STM32F7_LTDC_FB_BASE=0xc0000000
CONFIG_STM32F7_LTDC_FB_SIZE=261120
CONFIG_STM32F7_LTDC_L1_CHROMAKEYEN=y
CONFIG_STM32F7_LTDC_L1_CHROMAKEY=0x00000000
CONFIG_STM32F7_LTDC_L1_COLOR=0x00000000
CONFIG_STM32F7_LTDC_L1_RGB565=y
```

can you check this problem, and try to set BOARD_LTDC_OUTPUT_BPP  to 16 and test again ?

raiden00pl, 2024-10-08:
@terry0012
Thanks for help. The problem is somewhere in board configuration which was broken long time ago. I didn't find the reason yet, but I'll take care of it. Thanks for your help.

closed 2024-10-08 by xiaoxiang781216