apache/nuttx #13856: [BUG] nsh is not SMP-proof

UAV-Pilot, 2024-10-06:
### Description / Steps to reproduce the issue

1) On Pi Pico W board with Pi Pico SDK 2.0, setup a system with working flash SmartFS file system and without SMP, e.g.,
               ./tools/configure.sh -l raspberrypi-pico-w:nsh-flash

2) verify the following file creation and deletion commands work
         cd /flash
        echo  Hello_NuttX > test2.txt
        cat test2.txt
        rm test2.txt
        ls -l
3) enable SMP and run ps or other command to verify that SMP is working
   ref: https://blog.mark-stevens.co.uk/2023/06/adding-smp-to-nuttx-on-the-raspberry-pi-picow/

4) re-run commands in step 2)
   notice that when executing file creation or deletion command, nsh hangs
   
Not sure if there is any other part of NuttX OS that is also not SMP-proof.

### On which OS does this issue occur?

[OS: Linux]

### What is the version of your OS?

Windows 11 WSL2 Ubuntu 22.04.2

### NuttX Version

NuttX 12.7.0-RC0 0976258-dirty Oct 4 2024 22:58:36 arm rasp

### Issue Architecture

[Arch: arm]

### Issue Area

[Area: OS Components]

### Verification

- [X] I have verified before submitting the report.

xiaoxiang781216, 2024-10-07:
could you test on littlefs? so we can know whether is specific to smartfs.

jerpelea, 2024-10-10:
@UAV-Pilot updates?