apache/nuttx #13883: [BUG] nrf9160 (Cortex-M33) with tickless enabled randomly locks up in ostest

raiden00pl, 2024-10-07:
### Description / Steps to reproduce the issue

nrf9160 (Cortex-M33) with tickless enabled randomly locks up in ostest (in random places).

Tested with config: `./tools/configure.sh nrf9160-dk/ostest_tickless`

When tickless is disabled, the problem doesn't occur. 
CONFIG_DEBUG_ASSERTIONS=y makes the problem not occur (or its harder to trigger).

The problem doesn't occur on release/12.6 (or its harder to trigger).

### On which OS does this issue occur?

[OS: Linux]

### What is the version of your OS?

Arch Linux

### NuttX Version

nuttx-12.7.0-RC0, master

### Issue Architecture

[Arch: arm]

### Issue Area

[Area: Kernel]

### Verification

- [X] I have verified before submitting the report.

raiden00pl, 2024-10-31:
most likely broken by https://github.com/apache/nuttx/pull/13283

Fix-Point, 2024-11-01:
I don't have a board to verify this. From the code, it seems to be a problem with the RTC tickless alarm driver. The function `rtc_prepare_alarm` in `arch/arm/src/nrf91/nrf91_tickless_rtc.c` may lose the next timer setting if the current time has exceeded the next time to be set.

maxikrie, 2026-10-26:
I believe #21005 should fix this and we can close the issue @raiden00pl

closed 2026-10-27 by xiaoxiang781216