apache/nuttx #13884: [BUG] nsh free is not present even if not disabled in the config.

cederom, 2024-10-07:
### Description / Steps to reproduce the issue

1. Check the `.config` to contain `# CONFIG_NSH_DISABLE_FREE is not set`.
2. Build and flash the firmware.
3. See that `free` command is not present.
```
nsh> ?
help usage:  help [-v] [<cmd>]

    ?          exec       exit       help       hexdump    xd
```

### On which OS does this issue occur?

[OS: BSD]

### What is the version of your OS?

FreeBSD 13.3-RELEASE-p7 GENERIC amd64

### NuttX Version

12.7.0-RC0

### Issue Architecture

[Arch: arm]

### Issue Area

[Area: Applications], [Area: Configuring], [Area: Memory Management]

### Verification

- [X] I have verified before submitting the report.

raiden00pl, 2024-10-07:
`CONFIG_FS_PROCFS=y` must be set, otherwise free doesn't work:
https://github.com/apache/nuttx-apps/blob/c30583dda48a812f045961f88e0d751f06196b28/nshlib/nsh.h#L498-L501

This 'silent' disabling of commands in `nsh.h` can be really irritating and if someone doesn't know where to look, it can cause problems. But I don't know if there is a more user-friendly solution here, PROCFS must be supported for `free`.

acassis, 2024-10-08:
@raiden00pl maybe we should put "depends on FS_PROCFS" there, but this solution it bad because it hides the option from users.

A clumsy workaround it creating a menu entry depending on !FS_PROCFS just to warn the user:

```
+menu "[ ] Disable free (please enable FS_PROCFS first)"
+       depends on !FS_PROCFS
+endmenu
```

But I think many people don't like this approach (it pollutes the Kconfig). In fact Kconfig has some limitations when we think about some user friendly interface