apache/nuttx #13885: [BUG] I2C tool

linguini1, 2024-10-07:
### Description / Steps to reproduce the issue

The I2C tool on the Raspberry Pi Pico is incorrectly showing responses on every other address despite not being
connected to any I2C bus.

```console
nsh> i2c dev -b 0 00 77
NOTE: Some devices may not appear with this scan.
You may also try a scan with the -z flag to discover more devices using a zero-byte write request.
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: -- -- 02 -- 04 -- 06 -- 08 -- 0a -- 0c -- 0e --
10: 10 -- 12 -- 14 -- 16 -- 18 -- 1a -- 1c -- 1e --
20: 20 -- 22 -- 24 -- 26 -- 28 -- 2a -- 2c -- 2e --
30: 30 -- 32 -- 34 -- 36 -- 38 -- 3a -- 3c -- 3e --
40: 40 -- 42 -- 44 -- 46 -- 48 -- 4a -- 4c -- 4e --
50: 50 -- 52 -- 54 -- 56 -- 58 -- 5a -- 5c -- 5e --
60: 60 -- 62 -- 64 -- 66 -- 68 -- 6a -- 6c -- 6e --
70: 70 -- 72 -- 74 -- 76 --
```

Tested on 2 different Picos so it should not be a hardware issue.

I have configured the I2C tool, the I2C0 interface and the I2C character driver (I2C master support).

### On which OS does this issue occur?

[OS: Linux]

### What is the version of your OS?

Linux 5.15.153.1-microsoft-standard-WSL2

### NuttX Version

12.6.0-RC1

### Issue Architecture

[Arch: arm]

### Issue Area

[Area: Drivers], [Area: Sensors]

### Verification

- [X] I have verified before submitting the report.

linguini1, 2024-10-08:
I've also noticed through some further testing that I am able to read all the data in an EEPROM that the Pico is connected to on the I2C bus with the `i2c dump -a 50 -b 0 -r 0 256` command, but as soon as I then try the `i2c dev 00 77` command, I get the incorrect scan results and all subsequent dumps fail.

```console
nsh> i2c dump -a 50 -b 0 -r 0 256
nsh: i2c: too many arguments
DUMP Bus: 0 Addr: 50 Subaddr: 00
00000000  43 55 49 6e 53 70 61 63  65 0d 0a 53 45 4e 2d 42  |CUInSpace..SEN-B|
00000010  3a 20 32 30 32 34 2d 30  32 2d 30 34 0d 0a 4d 53  |: 2024-02-04..MS|
00000020  35 36 31 31 20 37 36 20  37 37 0d 0a 4d 41 58 4d  |5611 76 77..MAXM|
00000030  31 30 53 20 34 32 0d 0a  4c 53 4d 36 44 53 4f 33  |10S 42..LSM6DSO3|
00000040  32 20 36 42 0d 0a 53 48  54 34 31 20 34 34 0d 0a  |2 6B..SHT41 44..|
00000050  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000060  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000070  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |......|
00000080  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00000090  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
000000a0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |........|
000000b0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
000000c0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
000000d0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
000000e0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |............|
000000f0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
nsh> i2c dev 00 77
NOTE: Some devices may not appear with this scan.
You may also try a scan with the -z flag to discover more devices using a zero-byte write request.
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: 00 -- 02 -- -- 05 -- 07 -- 09 -- 0b -- -- 0e -- 
10: 10 -- -- 13 -- -- -- -- 18 -- 1a -- 1c -- 1e -- 
20: -- 21 -- 23 -- 25 -- 27 -- 29 -- 2b -- 2d -- 2f 
30: -- 31 -- -- 34 35 -- -- 38 -- 3a -- -- 3d -- 3f 
40: -- 41 42 -- 44 -- -- -- 48 -- -- -- 4c -- -- -- 
50: 50 -- 52 -- -- 55 -- 57 -- -- 5a -- -- 5d -- 5f 
60: -- -- 62 -- -- 65 -- -- 68 -- -- 6b -- 6d -- 6f 
70: -- 71 -- -- 74 -- -- --                         
nsh> i2c dump -a 50 -b 0 -r 0 256
nsh: i2c: too many arguments
i2ctool: dump: Transfer failed: 1
nsh> 
nsh> i2c dump-a 50 -b 0 -r 0 256
nsh: i2c: too many arguments
i2ctool: dump: Transfer failed: 1
nsh> i2c dump -a 50 -b 0 -r 0 256
nsh: i2c: too many arguments
i2ctool: dump: Transfer failed: 1
nsh> i2c dump -a 50 - 0 -r 0 256
nsh: i2c: too many arguments
i2ctool: dump: Transfer failed: 1
```

linguini1, 2024-10-08:
> @linguini1 maybe it is some recent regression because it was working few month ago

I agree, I had used this tool fairly recently after your article as well. I'm trying to pinpoint when this issue first appears in the commit history; I'm suspecting this may have something to do with the Pico 2.0.0 SDK.

I'm unfortunately quite pressed for time this month but I will see if I can make much progress!

acassis, 2024-10-08:
@xiaoxiang781216 this is why we need a hardware CI test urgently. We are breaking NuttX all the time!

linguini1, 2024-10-08:
I have tried reverting commits on `driver/i2c` back to this commit: 75fa94af023b5a01bdff3021ccfe6e41bfdb4a9b

Dated from before the embedded related article where the i2ctool was working. No change in behaviour, so I don't think any recent commits to the upper level I2C driver has caused this problem.

linguini1, 2024-10-08:
The 12.5 release works using SDK version 1.5.1. Going to continue bisecting.

xiaoxiang781216, 2024-10-08:
> I have tried reverting commits on `driver/i2c` back to this commit: [75fa94a](https://github.com/apache/nuttx/commit/75fa94af023b5a01bdff3021ccfe6e41bfdb4a9b)
> 
> Dated from before the embedded related article where the i2ctool was working. No change in behaviour, so I don't think any recent commits to the upper level I2C driver has caused this problem.

this patch only affects I2C slave. @linguini1 could you reverify again?

linguini1, 2024-10-08:
@xiaoxiang781216 @acassis 

From git bisect, it appears the first bad commit is 3b111c8b99bfcb00998355ba30e5276ffbb302be.

xiaoxiang781216, 2024-10-08:
@Fix-Point let's try repo on `Raspberry Pi Pico`.

Fix-Point, 2024-10-09:
Could this be related to a watchdog waiting issue? Would it work fine to change the I2C_TIMEOUT from `20*1000/CONFIG_USEC_PER_TICK` to `80*1000/CONFIG_USEC_PER_TICK` in `arch/arm/src/rp2040/rp2040_i2c.c`?

linguini1, 2024-10-09:
@Fix-Point
I tried both lower and faster speeds, the 80 you suggested and also 10. Neither made a difference in the output
unfortunately.

I also suspect that this is related to the watchdog, although I'm wary of fixing the issue on the rp2040 i2c side. Since
this worked before, it means the watchdog changes do not maintain the same watchdog behaviour. There is perhaps other
hardware platforms which have broken from this change, and I think the easiest way to fix them all would be to modify
the watchdog code.

Fix-Point, 2024-10-09:
Can you try this patch? [https://github.com/apache/nuttx/pull/14003](url)

xiaoxiang781216, 2024-10-09:
@linguini1

linguini1, 2024-10-09:
Thank you @Fix-Point for your work on this! That patch resolves the problem.

closed 2024-10-09 by acassis