apache/nuttx #13946: [BUG] Removing IFF_DOWN flag broke ESP32C6 wifi build

engdavidiogo, 2024-10-08:
### Description / Steps to reproduce the issue

Hi @xiaoxiang781216 , removing the **_IFF_DOWN_** flag in https://github.com/apache/nuttx/pull/13842 ended up breaking _apps/netutils/netlib/netlib_setifstatus.c_ which still uses the flag in _netlib_ifdown()_

Let me know if you need more information on this.

### On which OS does this issue occur?

[OS: Linux]

### What is the version of your OS?

Ubuntu 24.04.1 LTS

### NuttX Version

NuttX version 0.0.0 846cb7d432-dirty Oct  8 2024 12:09:40 esp32c6-devkitc:wifi

### Issue Architecture

[Arch: risc-v]

### Issue Area

[Area: File System]

### Verification

- [X] I have verified before submitting the report.

xiaoxiang781216, 2024-10-08:
@engdavidiogo do you update apps folder and ensure which contain https://github.com/apache/nuttx-apps/pull/2647.

engdavidiogo, 2024-10-08:
@xiaoxiang781216, thanks for the feedback.

For some reason my pull wasn't enough.

Closing

closed 2024-10-08 by engdavidiogo