apache/nuttx #14002: [BUG] Regression: cxxtest crash

leducp, 2024-10-09:
### Description / Steps to reproduce the issue

I tried cxxtest on stm32f411e-disco and xmc4800-relax with the C++ toolchain enabled:

NuttShell (NSH) NuttX-12.7.0-RC0
nsh> cxxtest
prindump_assert_info: Current Version: NuttX  12.7.0-RC0 8fe6c0ee8e Oct  9 2024 15:14:13 arm
dump_assert_info: Assertion failed panic: at file: :0 task: <noname> process: <noname> 0x800e905
up_dump_register: R0: 00000000 R1: 20005f20 R2: 00000005  R3: 00000100
up_dump_register: R4: 20000000 R5: 20005f20 R6: 00000000  FP: 00000000
up_dump_register: R8: 00000000 SB: 00000000 SL: 00000000 R11: 00000000
up_dump_register: IP: 20003f94 SP: 20005e80 LR: 0802c055  PC: 00000100
up_dump_register: xPSR: 60000000 PRIMASK: 00000000 CONTROL: 00000000
up_dump_register: EXC_RETURN: fffffffd
dump_stackinfo: User Stack:
dump_stackinfo:   base: 0x200058d8
dump_stackinfo:   size: 00002016
dump_stackinfo:     sp: 0x20005e80
stack_dump: 0x20005e60: 00000000 20005f20 00000005 00000100 20003f94 0802c055 00000100 60000000
stack_dump: 0x20005e80: 20005ea8 20005f20 0805a570 0802c083 0805a618 20005ea8 0805a618 0802bbb1
stack_dump: 0x20005ea0: 0805a55c 0800e58d 0805a55c 0805add0 00000000 00000000 00000000 00000000
stack_dump: 0x20005ec0: 00000000 00000000 20003830 00000000 00000000 00000001 00000000 00000000
stack_dump: 0x20005ee0: 00000000 00000000 00000000 00000000 00000000 00000000 00000400 00000000
stack_dump: 0x20005f00: 00000000 00000000 20001f00 20003434 00000000 00000000 00000000 00000000
stack_dump: 0x20005f20: 0805a64c 00000006 00000000 00001002 00000000 00000005 20000000 20000048
stack_dump: 0x20005f40: 00000028 08003e7b 00000000 20000000 0805a5a8 00000006 00000000 00001002
stack_dump: 0x20005f60: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
stack_dump: 0x20005f80: 00000000 00000000 00000000 20003830 00000000 00000000 00000000 20003468
stack_dump: 0x20005fa0: 20003440 20003448 00000000 00000000 00000000 00000008 20005f74 20003830
stack_dump: 0x20005fc0: 00000000 20000000 20005eac 20003468 20003440 20003448 0800e905 00000001
stack_dump: 0x20005fe0: 200058c8 0800e90d 20006120 08002651 20006133 08001f5d 20006048 00000001
stack_dump: 0x20006000: 200061dc 200054b8 00000001 20000e4c 00000000 00000000 00000000 00000000
stack_dump: 0x20006020: 00000000 fffffffd 00000002 20005554 20001f68 20000d34 00000003 0800c0e1
stack_dump: 0x20006040: 0800c0f4 01000000 ffffff80 00000003 200061c4 00000003 00000001 0800c02f
stack_dump: 0x20006060: 00000000 00000003 200060b8 00000000 00000000 00000000 00000000 00000000
stack_dump: 0x20006080: 00000000 00000000 00000000 0800e905 00000001 200058c8 00000000 08003f97
stack_dump: 0x200060a0: 0800e905 08001fbd 00000000 00000000 00000000 00000000 00000000 00000000
dump_tasks:    PID GROUP PRI POLICY   TYPE    NPX STATE   EVENT      SIGMASK          STACKBASE  STACKSIZE   COMMAND
dump_tasks:   ----   --- --- -------- ------- --- ------- ---------- ---------------- 0x20000530      2048   irq
dump_task:       0     0   0 FIFO     Kthread -   Ready              0000000000000000 0x20003ac0      1008   <noname>
dump_task:       1     1 100 RR       Task    -   Waiting Semaphore  0000000000000000 0x20004590      2008   <noname>
dump_task:       2     2 100 RR       Task    -   Running            0000000000000000 0x200058d8      2016   <noname>

I do not know how to investigate further.
In general, NuttX is really unstable since mid August and it is starting to be blocking on our side...

### On which OS does this issue occur?

[OS: Linux]

### What is the version of your OS?

Debian testing

### NuttX Version

master

### Issue Architecture

[Arch: arm]

### Issue Area

[Area: Applications]

### Verification

- [X] I have verified before submitting the report.

xiaoxiang781216, 2024-10-09:
does libcxx generate the similar panic?

leducp, 2024-10-10:
I'll try this. In the mid time, I tried the same configuration on 12.6.0 and same result so I wonder if it even work one day.

leducp, 2024-10-10:
OK this is better but it still crash:

```
nsh> cxxtest
task_spawn: name=cxxtest entry=0xc0481a9 file_actions=0x200141d0 attr=0x200141d4 argv=0x200142cc
spawn_execattrs: Setting policy=2 priority=100 for pid=2
nxtask_activate: cxxtest pid=2,TCB=0x20014758
Test ofstream ================================
printf: Starting test_ostream
printf: Successfully opened /dev/console
cout: Successfully opened /dev/console
Writing this to /dev/console
Test iostream ================================
Hello, this is only a test
Print an int: 190
Print a char: d
Test std::vector =============================
v1=1 2 3
Hello World Good Luck 
Test std::map ================================
Test C++17 features ==========================
pthread_mutex_timedlock: mutex=0x200001a0
pthread_mutex_timedlock: Returning 0
dump_assert_info: Current Version: NuttX  12.7.0-RC0 1c2856dfcb-dirty Oct 10 2024 11:13:46 arm
dump_assert_info: Assertion failed panic: at file: :0 task: `2 �2 `2 �2 `2 �2 �2 7 �2 �2 7 �2 �2 �2 �2 �2 Ձ
                                                                                                           `( process: Kernel 0xc0481a9
up_dump_register: R0: 20014758 R1: 20016d68 R2: 00000005  R3: 4870e2bb
up_dump_register: R4: 200169e4 R5: 200029ec R6: 0c0e2278  FP: 2000299c
up_dump_register: R8: 00000020 SB: 00000000 SL: 00000000 R11: 00000000
up_dump_register: IP: a0000000 SP: 20016200 LR: 0c017bc9  PC: 0c006144
up_dump_register: xPSR: 21000000 PRIMASK: 00000001 CONTROL: 00000000
up_dump_register: EXC_RETURN: ffffffed
dump_stackinfo: User Stack:
dump_stackinfo:   base: 0x20014bb8
dump_stackinfo:   size: 00008144
dump_stackinfo:     sp: 0x20016200
stack_dump: 0x200161e0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 20001558
stack_dump: 0x20016200: 20016d68 20014758 00000000 20014758 00640000 4870e2bb 200162a8 20016d64
stack_dump: 0x20016220: 200001a8 200162a0 0c0e58d0 00000000 20014758 0c008b1b 20016254 0c008b7d
stack_dump: 0x20016240: 200169e4 20016d64 a0000000 20016d64 0c008b6e 0c008d2d 20014b88 20016d64
stack_dump: 0x20016260: 00000000 00000000 00000000 0c00aba1 00000000 20016d60 00004c08 0c04eea5
stack_dump: 0x20016280: 20016d60 0000000b 00000001 0c0e5030 00000000 00000002 00000000 0c0d88c1
stack_dump: 0x200162a0: 0c0daccd 20003334 00000000 0c0dad41 0c0e5278 20016274 00004c08 00000001
stack_dump: 0x200162c0: 0c0daccd 20003334 00000001 0c0db945 0c0dad15 2000019c 0c0e2278 0c0dadd5
stack_dump: 0x200162e0: 0000000e 00000000 0c0dad15 2000019c 2000429c 0c0d9ce9 00000001 00000000
stack_dump: 0x20016300: 20002bb0 0c0daac7 00002000 0c0daab1 00002000 0c050f01 2000429c 20016410
stack_dump: 0x20016320: 00000001 0c0e2278 20003270 0000000e 00000002 0c0534f9 00000002 2000299c
stack_dump: 0x20016340: 2000299c 2000299c 2000299c 2000299c 200169e4 0c04a70f 0c0e2286 20014aa4
stack_dump: 0x20016360: 200169d4 200169d0 20012cd4 20012ccc 20016b90 20016ba0 00000004 20016ba0
stack_dump: 0x20016380: 20016ba0 20016b90 20016ba0 20016b90 20016b90 20016ba0 200132c0 20016b90
stack_dump: 0x200163a0: 20016b90 20016b90 20016ba0 ffffffff 20012c18 ffffffff ffffffff ffffffff
stack_dump: 0x200163c0: 00000001 20012cdc 20012cdc 20012cb4 200132c0 200132c0 20016ba4 20016b90
stack_dump: 0x200163e0: 20016b94 200132c0 200132c0 200132b8 20016b94 20016b94 20016b94 20016b90
stack_dump: 0x20016400: 200132b8 200132c0 200132c0 200029ec 20013201 20002998 20013200 20013200
stack_dump: 0x20016420: 20016b90 200132b8 200132c0 200132c0 200132b8 200132b8 20016b90 200132b8
stack_dump: 0x20016440: 20016b90 200132b8 200132b8 200132b8 200132c8 20016ba4 00000004 20016ba4
stack_dump: 0x20016460: 20016ba4 200132c8 20016ba4 200132c8 200029ec 200029ec 20003268 200132c8
stack_dump: 0x20016480: 20016ba4 20012d04 20012d04 20012cf8 00000008 00000000 00000000 2000299c
stack_dump: 0x200164a0: 0000000e 0c0e2278 200029ec 200132c0 00000002 200132c0 00000002 200132c0
stack_dump: 0x200164c0: 20012d08 20012d04 20012cf8 200132c0 200132c0 20012d08 200132c0 200133bc
stack_dump: 0x200164e0: 200133bc 20012d04 200133bc 200132c0 20012d04 200132c0 00000001 00000002
stack_dump: 0x20016500: 200132b8 20012bf0 200133b8 20012bf0 200133b8 00000001 200132c0 0000000e
stack_dump: 0x20016520: 00000000 2000299c 0000000e 20adbeef 2000299c 0c0e2286 0c0e2278 0c0e2278
stack_dump: 0x20016540: 200133b4 20003130 200133b0 2000299c ffffffff ffffffff 2000299c 2000299c
stack_dump: 0x20016560: 2000299c 2000299c 20002998 0000000e 0c0e2278 20002998 200132c8 0c0e2278
stack_dump: 0x20016580: 0c0e2278 0c0e2278 deadbeef deadbeef 200028b4 deadbe01 20002860 200028b4
stack_dump: 0x200165a0: 200028b4 deadbeef deadbeef deadbeef 200028b4 200028b4 deadbeef deadbeef
stack_dump: 0x200165c0: deadbeef 200028b4 200028b4 deadbeef deadbeef deadbeef 200132c0 200132c0
stack_dump: 0x200165e0: 200132b8 deadbeef deadbeef 200028b4 200028b4 deadbeef deadbeef deadbeef
stack_dump: 0x20016600: deadbeef deadbeef deadbeef deadbeef 00000000 00000000 20002864 0000002e
[... 26 lines cut ...]
stack_dump: 0x20016960: 00000002 20002864 20002864 20002860 20002864 20002864 20002864 20002864
stack_dump: 0x20016980: 0000000a 0c048253 200132c0 20002860 20002860 20016418 200169e4 0c0e21d8
stack_dump: 0x200169a0: 20002864 0c0e219a 00000000 0c0465f9 200132c4 200132b8 200132c4 0000000e
stack_dump: 0x200169c0: 00000000 200028b4 00000001 20002860 20002864 0c0e219c 0000000d 0c0e21ac
stack_dump: 0x200169e0: 00000008 0c0e21b8 0000000d deadbeef 200028b4 200028b4 20002864 20002864
[... cut ...]
dump_tasks:   ----   --- --- -------- ------- --- ------- ---------- ---------------- 0x200004d8      4096      1192    29.1%    irq
dump_task:       0     0   0 FIFO     Kthread -   Running            0000000000000000 0x20003588      2016       936    46.4%    Idle_Task
dump_task:       1     1 100 RR       Task    -   Waiting Semaphore  0000000000000000 0x20004488     65472      7668    11.7%    nsh_main
```

Note: I had to increase drastically the default stack value to have something working, so maybe it is also the problem for the C++ toolchain version. I'll give it a try

leducp, 2024-10-10:
OK! I have something working! When using the C++ toolchain, one have to initialize std::cout or you will crash the application. 
Something like that do the trick:
```cpp
  std::ofstream ttyOut;
  ttyOut.open ("/dev/console");
  std::cout.rdbuf(ttyOut.rdbuf());
```

One probably need to to it also to std::cerr and std::clog. I can do a patch to initialize properly this part of the C++ library in NuttX bootup but I don't knwo where I can put it. Any clue?

In summary:
 * C++ toolchain library crash because std::cout is not initialized, if init cxx work perfectly
 * C++ libcxx crash while testing the C++17 features (regression?)

xiaoxiang781216, 2024-10-10:
do you have the similar snippet in your ld script:
```
    .init_section : {
        _sinit = ABSOLUTE(.);
        KEEP(*(SORT_BY_INIT_PRIORITY(.init_array.*) SORT_BY_INIT_PRIORITY(.ctors.*)))
        KEEP(*(.init_array EXCLUDE_FILE(*crtbegin.o *crtbegin?.o *crtend.o *crtend?.o) .ctors))
        _einit = ABSOLUTE(.);
    } > flash
```

leducp, 2024-10-10:
I didn't have this snippet but this one:
```
    .init_section :
    {
        _sinit = ABSOLUTE(.);
        KEEP(*(SORT_BY_INIT_PRIORITY(.init_array.*) SORT_BY_INIT_PRIORITY(.ctors.*)))
        KEEP(*(.init_array .ctors))
        _einit = ABSOLUTE(.);
    } > flash
```

I tried yours but it is worse libcxx with no impact on libstdc++?

libcxx:
```
nsh> cxxtest
task_spawn: name=cxxtest entry=0xc0481a9 file_actions=0x200141d0 attr=0x200141d4 argv=0x200142cc
spawn_execattrs: Setting policy=2 priority=100 for pid=2
nxtask_activate: cxxtest pid=2,TCB=0x20014758
Test ofstream ================================
printf: Starting test_ostream
printf: Successfully opened /dev/console
cout: Succe
```

libstdc++ (with cout patch, without it crash immediatly): no impact (working)