apache/nuttx #14005: [BUG] Regression: ostest crash (on xmc4800-relax at least)

leducp, 2024-10-09:
### Description / Steps to reproduce the issue

Enable ostest on xmc4800-relax
start it

```
nsh> ostest
stdio_test: write fd=1
stdio_test: Standard I/O Check: printf
stdio_test: write fd=2
ostest_main: putenv(Variable1=BadValue3)
ostest_main: setenv(Variable1, GoodValue1, TRUE)
ostest_main: setenv(Variable2, BadValue1, FALSE)
ostest_main: setenv(Variable2, GoodValue2, TRUE)
ostest_main: setenv(Variable3, GoodValue3, FALSE)
ostest_main: setenv(Variable3, BadValue2, FALSE)
show_variable: Variable=Variable1 has value=GoodValue1
show_variable: Variable=Variable2 has value=GoodValue2
show_variable: Variable=Variable3 has value=GoodValue3
ostest_main: Started user_main at PID=4

user_main: Begin argument test
user_main: Started with argc=5
user_main: argv[0]="ostest"
user_main: argv[1]="Arg1"
user_main: argv[2]="Arg2"
user_main: argv[3]="Arg3"
user_main: argv[4]="Arg4"

End of test memory usage:
VARIABLE  BEFORE   AFTER
======== ======== ========
arena        bba0     bba0
ordblks         1        1
mxordblk     6b90     6b90
uordblks     5010     5010
fordblks     6b90     6b90

user_main: getopt() test
getopt():  Simple test
getopt():  Invalid argument
getopt():  Missing optional argument
getopt_long():  Simple test
getopt_long():  No short options
getopt_long():  Argument for --option=argument
getopt_long():  Invalid long option
getopt_long():  Mixed long and short options
getopt_long():  Invalid short option
getopt_long():  Missing optional arguments
getopt_long_only():  Mixed long and short options
getopt_long_only():  Single hyphen long options

End of test memory usage:
VARIABLE  BEFORE   AFTER
======== ======== ========
arena        bba0     bba0
ordblks         1        1
mxordblk     6b90     6b90
uordblks     5010     5010
fordblks     6b90     6b90

user_main: libc tests

End of test memory usage:
VARIABLE  BEFORE   AFTER
======== ======== ========
arena        bba0     bba0
ordblks         1        1
mxordblk     6b90     6b90
uordblks     5010     5010
fordblks     6b90     6b90
show_variable: Variable=Variable1 has value=GoodValue1
show_variable: Variable=Variable2 has value=GoodValue2
show_variable: Variable=Variable3 has value=GoodValue3
show_variable: Variable=Variable1 has no value
show_variable: Variable=Variable2 has value=GoodValue2
show_variable: Variable=Variable3 has value=GoodValue3

End of test memory usage:
VARIABLE  BEFORE   AFTER
======== ======== ========
arena        bba0     bba0
ordblks         1        2
mxordblk     6b90     6b90
uordblks     5010     4ff0
fordblks     6b90     6bb0
show_variable: Variable=Variable1 has no value
show_variable: Variable=Variable2 has no value
show_variable: Variable=Variable3 has no value

End of test memory usage:
VARIABLE  BEFORE   AFTER
======== ======== ========
arena        bba0     bba0
ordblks         2        2
mxordblk     6b90     6b90
uordblks     4ff0     4f78
fordblks     6bb0     6c28

user_main: setvbuf test
setvbuf_test: Test NO buffering
setvbuf_test: Using NO buffering
setvbuf_test: Test default FULL buffering
setvbuf_test: Using default FULL buffering
setvbuf_test: Test FULL buffering, buffer size 64
setvbuf_test: Using FULL buffering, buffer size 64
setvbuf_test: Test FULL buffering, pre-allocated buffer
setvbuf_test: Using FULL buffering, pre-allocated buffer
setvbuf_test: Test LINE buffering, buffer size 64
setvbuf_test: Using LINE buffering, buffer size 64
setvbuf_test: Test FULL buffering, pre-allocated buffer
setvbuf_test: Using FULL buffering, pre-allocated buffer

End of test memory usage:
VARIABLE  BEFORE   AFTER
======== ======== ========
arena        bba0     bba0
ordblks         2        2
mxordblk     6b90     6b90
uordblks     4f78     4f78
fordblks     6c28     6c28

user_main: /dev/null test
dev_null: Read 0 bytes from /dev/null
dev_null: Wrote 1024 bytes to /dev/null

End of test memory usage:
VARIABLE  BEFORE   AFTER
======== ======== ========
arena        bba0     bba0
ordblks         2        2
mxordblk     6b90     6b90
uordblks     4f78     4f78
fordblks     6c28     6c28

user_main: FPU test
Starting task FPU#1
fpu_test: Started task FPU#1 at PID=5
FPU#1: pass 1
Starting task FPU#2
fpu_test: Started task FPU#2 at PID=6
FPU#2: pass 1
FPU#1: pass 2
FPU#2: pass 2
FPU#1: pass 3
FPU#2: pass 3
FPU#1: pass 4
FPU#2: pass 4
FPU#1: pass 5
FPU#2: pass 5
FPU#1: pass 6
FPU#2: pass 6
FPU#1: pass 7
FPU#2: pass 7
FPU#1: pass 8
FPU#2: pass 8
FPU#1: pass 9
FPU#2: pass 9
FPU#1: pass 10
FPU#2: pass 10
FPU#1: pass 11
FPU#2: pass 11
FPU#1: pass 12
FPU#2: pass 12
FPU#1: pass 13
FPU#2: pass 13
FPU#1: pass 14
FPU#2: pass 14
FPU#1: pass 15
FPU#2: pass 15
FPU#1: pass 16
FPU#2: pass 16
FPU#1: Succeeded
FPU#2: Succeeded
fpu_test: Returning

End of test memory usage:
VARIABLE  BEFORE   AFTER
======== ======== ========
arena        bba0     bba0
ordblks         2        4
mxordblk     6b90     53a0
uordblks     4f78     5f88
fordblks     6c28     5c18

user_main: task_restart test

Test task_restart()
restart_main: setenv(VarName, VarValue, TRUE)
restart_main: Started restart_main at PID=7
restart_main: Started with argc=4
restart_main: argv[0]="ostest"
restart_main: argv[1]="This is argument 1"
restart_main: argv[2]="Argument 2 here"
restart_main: argv[3]="Lastly, the 3rd argument"
restart_main: Variable=VarName has value=VarValue
restart_main: I am still here
restart_main: I am still here
restart_main: Started restart_main at PID=7
restart_main: Started with argc=4
restart_main: argv[0]="ostest"
restart_main: argv[1]="This is argument 1"
restart_main: argv[2]="Argument 2 here"
restart_main: argv[3]="Lastly, the 3rd argument"
restart_main: Variable=VarName has value=VarValue
restart_main: Started with argc=4
restart_main: argv[0]="ostest"
restart_main: argv[1]="This is argument 1"
restart_main: argv[2]="Argument 2 here"
restart_main: argv[3]="Lastly, the 3rd argument"
restart_main: Variable=VarName has value=VarValue
restart_main: Exiting

End of test memory usage:
VARIABLE  BEFORE   AFTER
======== ======== ========
arena        bba0     bba0
ordblks         4        2
mxordblk     53a0     4798
uordblks     5f88     73a8
fordblks     5c18     47f8

user_madump_assert_info: Current Version: NuttX  12.7.0-RC0 47fc3a67f7 Oct  9 2024 15:54:44 arm
dump_assert_info: Assertion failed : at file: :0 task: ostest process: ostest 0xc01c5c9
up_dump_register: R0: 200012e0 R1: 00000000 R2: 00000000  R3: 00000000
up_dump_register: R4: 20001fec R5: 20001fd0 R6: 00000000  FP: 00000000
up_dump_register: R8: 00000000 SB: 00000000 SL: 00000000 R11: 00000000
up_dump_register: IP: a0000000 SP: 20009358 LR: 0c002c6f  PC: 0c002c6f
up_dump_register: xPSR: 60000000 PRIMASK: 00000001 CONTROL: 00000006
up_dump_register: EXC_RETURN: 00000000
dump_stackinfo: User Stack:
dump_stackinfo:   base: 0x200074b0
dump_stackinfo:   size: 00008120
dump_stackinfo:     sp: 0x20009358
stack_dump: 0x20009338: 20006f00 00000000 00017204 0c002197 0c066060 20006fe0 20001fec 0c002cc9
stack_dump: 0x20009358: 200012e0 200071cc 200012e0 00000000 00000000 00000000 00000000 20006fe0
stack_dump: 0x20009378: 200012e0 00000000 00000000 00000000 00000038 00000000 000071c8 00000000
stack_dump: 0x20009398: 20006fe0 00000140 0c020a5d 0c00877b 00000000 00000000 00000000 00000000
stack_dump: 0x200093b8: 00000002 0c020ac9 00000000 0000000f ffffffff 00000002 000073a8 0c020c13
stack_dump: 0x200093d8: 00000000 00000018 200071c8 00000019 20001fd0 0c01c741 0000bba0 00000001
stack_dump: 0x200093f8: 00000020 00006b90 00005010 00006b90 00005168 00000000 20007478 00000005
stack_dump: 0x20009418: 0c004e04 00000005 00000000 00000000 00000000 0c008c83 00000000 20007478
stack_dump: 0x20009438: 00000005 0c01c5c9 00000000 0c004e6d 00000000 20007478 00000000 20006fe0
stack_dump: 0x20009458: 00000005 00000001 00000000 00000000 00000000 00000000 00000000 00000000
in: waitpid test

Test waitpid()
waitpid_start_child: Started waitpid_main at PID=8
waitpid_start_child: Started waitpid_main at PID=9
waitpid_main: PID 8 Started
waitpid_main: PID 9 Started
waitpid_start_child: ERROR Failed to start waitpid_main
ostest_main: Exiting with status 256
stdio_test: Standard I/O Check: fprintf to stderr
nsh> waitpid_main: PID 8 exitting with result=14
waitpid_main: PID 9 exitting with result=14
```

### On which OS does this issue occur?

[OS: Linux]

### What is the version of your OS?

Debian testing

### NuttX Version

master

### Issue Architecture

[Arch: arm]

### Issue Area

[Area: Kernel]

### Verification

- [X] I have verified before submitting the report.

xiaoxiang781216, 2024-10-09:
@leducp could you git bisect the bad commit id?

leducp, 2024-10-10:
I'll do that when I'll manage to get a bit of time. In the mid time, I managed to workaround the problem by remobing the "CONFIG_SCHED_HPWORK" option. I do not know what's the impact on the kernel but at least it let ostest finish properly without it :)